atar-axis/xpadneo #99: Use Gamepad as Mouse on GUI

FreakyAlleyCat, 2019-05-05:
**Summary:**
I'd love having the ability to control Linux GUI using the XBox One Controller and xpadneo driver as a mouse substitute.

**Specifics:**
I'm currently running Raspbian OS on a Raspberry Pi 3 B+. 
Using Parsec i created a streaming box to stream games from my computer to my TV via LAN.
My XBox One Controller is hooked up to the Raspberry Pi via bluetooth and works like a charm in game with xpadneo.

Now i'd love to use the Gamepad as a mouse on Raspbian so i don't need a physical mouse hooked up to the Raspberry Pi anymore to control the Raspbian GUI.

**Here's a suggestion for a possible key assignment:**

Left Joystick = cursor movement
right joystick = scroll wheel
button "A" = left click
button "B" = enter 
button "X" = right click
"menu" button = maybe a shortcut to the linux shutdown/restart window?

(Scrolling could also be done using the left and right shoulder buttons.)

atar-axis, 2019-05-05:
Hey there Lukas ;)
I would suggest to use the menu Button to Switch between the Mouse and gamepad behaviour. 

There are already some Projects around (QJoyPad and Antimicro e.g.). I am not sure therefore If it is a good Idea to so this in kernelspace.  I will give it a try and think about it - stay tuned :)

atar-axis, 2019-05-05:
Okay, it is quite easy to do it using antimicro :)

[image]

atar-axis, 2019-05-05:
It is quite easy to do this using antimicro :)
I wasn't able to record my screen for some reason, but I think the following screenshot may explain it too:
[image]

1. Connect your gamepad
2. Launch AntiMicro
3. Click on the `L-Stick Up` Button
4. Select the `Mouse` Tab
5. Press the big `Up` button
6. Do the same for the other directions and buttons/keys

FreakyAlleyCat, 2019-05-06:
That looks awesome, will give it a shot and report back! :)

atar-axis, 2019-05-06:
I thought a bit about it and I think I will not implement this in this driver, the reason is redundancy in essence. I would bloat up the kernel if every driver translates the gamepad events into mouse events itself.
I think it is better if we have one central module/application which is able to translate every connected gamepad. Any other thought?

atar-axis, 2019-05-09:
I created a new branch, called `vmouse`. This version does use another kernel module which is called `vmouse` too, you can get it here: https://github.com/atar-axis/vmouse

You have to install the `vmouse` module first, then - afterwards - the `xpadneo` vmouse branch. It is still under development but the mouse should at least already move :)

atar-axis, 2019-05-10:
It is useable now, but I will have to change the behaviour of reports.
Currently the mouse does only move as long as you keep moving the stick - when you stop somewhere, even if it is not in the middle, then the mouse will stop moving. That's because normally the gamepad does not report any new events when nothing happens. I will fix this within the next days.

To install the current version, do the following:

1. Clone `vmouse` by running `git clone https://github.com/atar-axis/vmouse.git vmouse_module`
2. Switch into the repository (`cd vmouse_module`) and run the install script: `sudo ./install.sh`
3. Move to another directory
4. Clone the vmouse branch of `xpadneo` by: `git clone https://github.com/atar-axis/xpadneo -b vmouse xpadneo_vmouse`
5. Switch to the repository `cd xpadneo_vmouse`
6. Uninstall all installed versions of xpadneo: `sudo ./uninstall.sh`
7. Install the cloned vmouse branch: `sudo ./install`

To use the mouse, press and hold the XBOX button for around 2 seconds until the gamepad confirms the mode change by a short rumble. You can now use the left stick to move the mouse and simulate a mouseclick by the (A)-button. Press and hold the XBOX button again to switch back to the gamepad mode.

germangoergs, 2019-05-10:
> It is useable now, but I will have to change the behaviour of reports.
> Currently the mouse does only move as long as you keep moving the stick - when you stop somewhere, even if it is not in the middle, then the mouse will stop too. That's because normally the gamepad does not report any new events when nothing happens. So that's a problem which is caused by the translation of the `absolute` stick values into `relative` mouse values. I will fix this within the next days.
> 
> To install the current version, do the following:
> 
>     1. Clone `vmouse` by running `git clone https://github.com/atar-axis/vmouse.git vmouse_module`
> 
>     2. Switch into the repository (`cd vmouse_module`) and run the install script: `sudo ./install.sh`
> 
>     3. Move to another directory (like `cd ..`)
> 
>     4. Clone the vmouse branch of `xpadneo` by: `git clone https://github.com/atar-axis/xpadneo -b vmouse xpadneo_vmouse`
> 
>     5. Switch to the repository `cd xpadneo_vmouse`
> 
>     6. Uninstall all installed versions of xpadneo: `sudo ./uninstall.sh`
> 
>     7. Install the cloned vmouse branch: `sudo ./install`
> 
> 
> To use the mouse, press and hold the XBOX button for around 2 seconds until the gamepad confirms the mode change by a short rumble. You can now use the left stick to move the mouse and simulate a mouseclick by the (A)-button. Press and hold the XBOX button again to switch back to the gamepad mode.

That's kinda cool, will test it out. :)

atar-axis, 2019-05-10:
> I will fix this within the next days.

Done (Please update vmouse). It should work now smoothly :grin:
Have to do some smaller fixes, but should be to test

atar-axis, 2019-05-15:
tested under:
*** 4.14.98 (arm, raspbian)
* 5.0.10 (arch, x86_64)**

KocproZ, 2019-05-22:
This works great!
Here is an idea: would it be possible to make right stick work like a mouse scroll?

atar-axis, 2019-05-22:
No problemo, I will add it to the TODO list :)

atar-axis, 2019-05-22:
@KocproZ Done, please get the latest version of  the vmouse branch of xpadneo

KocproZ, 2019-05-22:
It doesn't work :/
Here is the dmesg log:
```

[  650.848270] hid-generic 0005:045E:02FD.000C: unknown main item tag 0x0
[  650.848329] input: Xbox Wireless Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-14/1-14:1.0/bluetooth/hci0/hci0:256/0005:045E:02FD.000C/input/input43
[  650.848415] input: Xbox Wireless Controller Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-14/1-14:1.0/bluetooth/hci0/hci0:256/0005:045E:02FD.000C/input/input44
[  650.848467] hid-generic 0005:045E:02FD.000C: input,hidraw6: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 0c:54:15:89:06:0b
[  650.850458] xpadneo udev: 0005:045E:02FD.000C
[  650.852062] vmouse: hi there!
[  650.852084] input: Mouse Emulation as /devices/virtual/input/input45
[  650.852427] hid_xpadneo: Unknown symbol vmouse_wheel (err -2)
[  650.998452] xpadneo udev: ok

```
If you need any additional logs let me know

atar-axis, 2019-05-22:
Ah sorry, haven't uploaded the new [vmouse](https://github.com/atar-axis/vmouse) module, please update this one too ;)

KocproZ, 2019-05-22:
Installing vmouse fails on build
```
$ sudo ./install.sh  
* copying module into /usr/src
* adding module to DKMS

Creating symlink /var/lib/dkms/vmouse/0.1/source ->
                 /usr/src/vmouse-0.1

DKMS: add completed.
* installing module (using DKMS)

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.0.13-arch1-1-ARCH -C /usr/lib/modules/5.0.13-arch1-1-ARCH/build M=/var/lib/dkms/vmouse/0.1/build/src modules...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.0.13-arch1-1-ARCH (x86_64)
Consult /var/lib/dkms/vmouse/0.1/build/make.log for more information.

```
Here is the `make.log`: https://pastebin.com/VEj0i5Hg

atar-axis, 2019-05-22:
Thanks, and sorry again... fixed (hopefully)

KocproZ, 2019-05-22:
Thanks, it works now!
One more thing: can you make it a little slower?

atar-axis, 2019-05-22:
Yes for sure, I will make that configurable in the future. I slowed it down for now - just update xpadneo again

KocproZ, 2019-05-22:
It's faster now :sweat_smile:

atar-axis, 2019-05-22:
sorry, I am quite busy right now therefore I did not tested it. try again ;)

KocproZ, 2019-05-22:
It's perfect, thanks!

atar-axis, 2019-05-22:
No problem, thanks for reporting and testing!

atar-axis, 2019-05-22:
Hum, isn't it still a bit fast? I tried it myself and it's still hard to navigate

KocproZ, 2019-05-22:
I mean, a little, but I didn't want to take up more of your time

atar-axis, 2019-05-22:
You are welcome to do so ;) I slowed it down again, its now 50% of what it was at the beginning. Feels good now.

KocproZ, 2019-05-22:
I agree, scrolling feels nice now.

atar-axis, 2019-05-22:
Okay, perfect! Next step is to implement a deadband so that the mouse does not move anymore if the sticks are in idle position. Cheers!

atar-axis, 2019-05-23:
It is nearly ready to merge now, xpadneo will now only use the vmouse module if it is really available (installed) on the system. Last thing: deadband implementation

Lucict, 2020-06-11:
Hi there--

I'm having a little trouble getting the vmouse version of this working. I installed vmouse and then installed the vmouse branch of xpadneo (after I uninstalled the master branch). I am using the Xbox One S controller--when I hold down the Xbox button, it doesn't seem to switch modes.

Any ideas or points in the right direction?  Thanks!

kakra, 2020-06-11:
The vmouse branch is vastly outdated, it should still work, tho. If it doesn't, it may need to be updated to current master... @atar-axis is working on this, not sure how much time he can dedicate to the project currently.

eiswuerfel, 2020-09-22:
Any updates here? Do I still need the vmouse branch, will it be updated to master?

kakra, 2020-09-22:
I'm planning to schedule this for v0.10

kakra, 2021-04-01:
@cybergurke If you'd like to try there's a very early work-in-progress PR #282 that implements mouse mode. Feel free to continue discussion over there. To toggle mouse mode, press and hold the guide button, the tap the select button (left below guide), and release the guide button.

Left stick is for the pointer, right stick for the wheel. The triggers are the mouse buttons, the shoulder buttons are for back and forward in a web browser. Most other buttons are undecided yet, tho this is planned: dpad for cursor movement, A = enter, B = escape. In theory, the start button (right below guide) should show an on-screen keyboard if your desktop environment understands the KEY_KEYBOARD event but this assignment is not final.

closed 2026-03-01 by kakra