atar-axis/xpadneo #101: Controller Connects with No Vibration

Dhamma-valence, 2019-05-08:
Controller connects, but doesn't rumble and input isn't working anywhere (tried Cemu through Playonlinux and multiple Steam games) after previously working fine.

Steps to reproduce the behavior:
1. Install Xpadneo
2.Wait for it to (seemingly randomly) happen

Expected the controller to rumble when connected and for input to work normally.

`dmesg -wH`
https://pastebin.com/0xfDtikQ

I think this may be related to the Xpadneo driver being unsigned, but Secure Boot is turned off so I didn't think it would matter anyways. I tried running `sudo /root/module-signing/one-time-setup` to sign the module, but I got the error: `command not found`.

atar-axis, 2019-05-08:
> [  +0.000301] hid_xpadneo: `n' invalid for parameter `disable_ff'

have you changed the disable_ff parameter?

Dhamma-valence, 2019-05-08:
No I haven't, I've done nothing in the way of customization so far. What's strange is that xpadneo worked fine last night, and overnight (never rebooting the pc), it seems to have gotten this bug. 

`cat /etc/modprobe.d/99-xpadneo-bluetooth.conf`
https://pastebin.com/458KbMkF

`ls /etc/modprobe.d/`
https://pastebin.com/88D9jBG5

atar-axis, 2019-05-09:
Hum, could you scan your system with `sudo grep -r "disable_ff" /` please?

Dhamma-valence, 2019-05-09:
`sudo grep -r "disable_ff" /`
https://pastebin.com/Km6JzMfM

atar-axis, 2019-05-09:
thanks, can you please `sudo ./uninstall.sh`, get the new version (master) and `sudo ./install.sh` it again? :)

Dhamma-valence, 2019-05-09:
Wow, it works beautifully now! I feel like an idiot for it having such a simple solution :P 
What did you realize was wrong with it? I'd like to understand if you have the time.

atar-axis, 2019-05-09:
We changed the behaviour of `disable_ff`, therefore we switched from a boolean type (`y` / `n`) to a numeric one. Your kernel complained about the `n`, which is not longer accepted since it has to be numeric now, so it was clear that there must be the `n` somewhere - and as you can see I have forgotten to change the default value in the configure script. :)

> Wow, it works beautifully now!

Thank you for reporting this issue!

closed 2019-05-09 by atar-axis