atar-axis/xpadneo #102: Provide example script in Python to start a rumble effect on the xbox controller

gordongnarf, 2019-05-08:
**Is your feature request related to a problem? Please describe.**
I don't get a rumble effect when trying to make the xbox controller rumble using the example python script provided on the evdev documentation site:
[https://python-evdev.readthedocs.io/en/latest/tutorial.html#injecting-an-ff-event-into-first-ff-capable-device-found](url)

I am testing on a raspberry py raspbian stretch, kernel 4.14.98-v7+, python3.6 and the xpadneo driver installed. The controller is wireless using bluetooth. Rumble should in principle be possible since the tool fftest works fine and the controller rumbles, when xpadneo takes over control :-)

**Describe the solution you'd like**
A python example script using **evdev** would be perfect, since i am using evdev together with **asyncio** to read out the controller buttons and analog input.

**Describe alternatives you've considered**
An alternative would be a solution, where one uses e.g. **fcntl.ioctl** to write directly to the /dev/input/eventx file.
An example for that is given in 
[https://github.com/chrippa/ds4drv/issues/91](url)
In my case i get en error and i am currently unable to understand/remove the error.

atar-axis, 2019-05-09:
I am not a trained python programmer, so it doesn't work yet but I fixed at least the import errors
to start somewhere:

```
from evdev import ecodes, InputDevice, ff, util

dev = None

# Find first EV_FF capable event device (that we have permissions to use).
for name in util.list_devices():
    dev = InputDevice(name)
    if ecodes.EV_FF in dev.capabilities():
        break

if dev is None:
    
    print("Sorry, no FF capable device found")
    
else:
    
    print("Preparing FF effect...")

    rumble = ff.Rumble(strong_magnitude=0xc000, weak_magnitude=0xc000)
    effect_type = ff.EffectType(ff_rumble_effect=rumble)
    duration_ms = 1000

    effect = ff.Effect(
        ecodes.FF_RUMBLE, # type
        -1, # id (set by ioctl)
        0,  # direction
        ff.Trigger(0, 0), # no triggers
        ff.Replay(duration_ms, 0), # length and delay
        ff.EffectType(ff_rumble_effect=rumble)
    )
    
    print("Uploading FF effect...")
    
    effect_id = dev.upload_effect(effect)
    
    
    print("Playing FF effect...")
    
    repeat_count = 1
    dev.write(ecodes.EV_FF, effect_id, repeat_count)
    
    dev.erase_effect(effect_id) 
```

atar-axis, 2019-05-09:
Here you go:

```
from evdev import ecodes, InputDevice, ff, util
import time

dev = None

# Find first EV_FF capable event device (that we have permissions to use).
for name in util.list_devices():
    dev = InputDevice(name)
    if ecodes.EV_FF in dev.capabilities():
        break

if dev is None:
    
    print("Sorry, no FF capable device found")
    
else:
    print("found " + dev.name + " at " + dev.path)
    print("Preparing FF effect...")

    rumble = ff.Rumble(strong_magnitude=0xc000, weak_magnitude=0xc000)
    effect_type = ff.EffectType(ff_rumble_effect=rumble)
    duration_ms = 1000

    effect = ff.Effect(
        ecodes.FF_RUMBLE, # type
        -1, # id (set by ioctl)
        0,  # direction
        ff.Trigger(0, 0), # no triggers
        ff.Replay(duration_ms, 0), # length and delay
        ff.EffectType(ff_rumble_effect=rumble)
    )
    
    print("Uploading FF effect...")
    
    effect_id = dev.upload_effect(effect)
    
    
    print("Playing FF effect...")
    
    repeat_count = 1
    
    for i in range(0, 100):
        dev.write(ecodes.EV_FF, effect_id, repeat_count)
        time.sleep(1)
    
    dev.erase_effect(effect_id) 
```

atar-axis, 2019-05-09:
Would be nice, if you could in return create a **PR** to add a example to the repository ;)

gordongnarf, 2019-05-09:
> https://github.com/atar-axis/xpadneo/tree/pydev-example/misc/examples/python_evdev_rumble
> Give it a try and tell me if it works for you please, then I will merge it into master

You are simply too fast :-)
Script works well, Thank you!
Yesterday i took the fftest.c code from
[https://github.com/flosse/linuxconsole/blob/master/utils/fftest.c](url), 
learned some c-coding and the basics how to compile.
Then i stripped the code until i had only the very necessary parts in it:
open the device, 
generate the effect data, 
upload the effect data, 
start the effect, 
stop the effect.
Doing so, i learned, that i have to wait after starting the effect before stopping it :-)
Today i wanted to insert the wait into the python code and see if it works then ... :-)

Below the c-code with the basic steps to generate a rumble (to compile, you have to download also bitmaskros.h from the site given above).
This example might help people coding in c.
/*
 * $id$
 *
 * Tests the force feedback driver
 * Copyright 2001-2002 Johann Deneux <deneux@ifrance.com>
 */

#include <stdio.h>
#include <sys/types.h>
#include <sys/stat.h>
#include <fcntl.h>
#include <unistd.h>
#include <stdlib.h>
#include <string.h>
#include <sys/ioctl.h>
#include <linux/input.h>
#include "bitmaskros.h"

int main(int argc, char** argv)
{
	struct ff_effect effect;
	struct input_event play, stop, gain;
	int fd;
	const char * device_file_name = "/dev/input/event0";
	int i;

	printf("Force feedback test program.\n");
	printf("HOLD FIRMLY YOUR WHEEL OR JOYSTICK TO PREVENT DAMAGES\n\n");

	for (i=1; i<argc; ++i) {
		if (strncmp(argv[i], "--help", 64) == 0) {
			printf("Usage: %s /dev/input/eventXX\n", argv[0]);
			printf("Tests the force feedback driver\n");
			exit(1);
		}
		else {
			device_file_name = argv[i];
		}
	}
	// Open device 
	fd = open(device_file_name, O_RDWR);
	if (fd == -1) {
		perror("Open device file");
		exit(1);
	}
	printf("Device %s opened\n", device_file_name);
	// generate Data for a weak rumbling effect 
	effect.type = FF_RUMBLE;
	effect.id = -1;
	effect.u.rumble.strong_magnitude = 0;
	effect.u.rumble.weak_magnitude = 0xc000;
	effect.replay.length = 2000;
	effect.replay.delay = 0;
	// upload effect
	printf("Uploading effect #5 (Weak rumble, with light motor) ... ");
	if (ioctl(fd, EVIOCSFF, &effect) == -1) {
		perror("Error");
	} else {
		printf("OK (id %d)\n", effect.id);
	}
	fflush(stdout);
	// play effect
	memset(&play,0,sizeof(play));
	play.type = EV_FF;
	play.code = effect.id;
	play.value = 1;
	if (write(fd, (const void*) &play, sizeof(play)) == -1) {
		perror("Play effect");
		exit(1);
	}
    // wait until the effect is over
	sleep(2);
	// Stop the effect 
	printf("Stopping effect\n");
	memset(&stop,0,sizeof(stop));
	stop.type = EV_FF;
	stop.code =  effect.id;
	stop.value = 0;

	exit(0);
}

gordongnarf, 2019-05-09:
Hi Florian,
I can open a pull request. Haven't done so before... But if it helps you...
I read that pr are used to collaborate, help fixing bugs...
What would be my contribution in the pr?
Thomas

gordongnarf, 2019-05-09:
Ah, I guess I understand now. You mean I could add the example where I use
the controller together with the FF effect.
In my case it's a remote control for a car controlled by a raspberry pi.
Ok, I can do that.
Sorry for being so slow :-)
First time I work with github and open source stuff. Don't really know the
rules.
It's fun, by the way 😁

atar-axis, 2019-05-09:
I am happy If you are, thank you for getting back and confirming that it works 😊

atar-axis, 2019-05-09:
merged into master :grin:

commit 2cae03bc (2019-05-09): added pydev example, closes #102 (#103)
closed 2019-05-09 by atar-axis