atar-axis/xpadneo #107: Controller stopped working with Kernel 5.1

germangoergs, 2019-05-23:
**Describe the bug**
A clear and concise description of what the bug is.
I can connect a controller, but it doesn't rumble and also isn't detected as xpadneo device since I use the 5.1 kernel series.

**To Reproduce**
Steps to reproduce the behavior:
install xpadneo driver on 5.1 kernel series
connect controller

**Expected behavior**
A clear and concise description of what you expected to happen.
controller rumbles and is detected correctly

**Screenshots/Gifs**
If applicable, add screenshots or animated gifs to help explain your problem.

**System information**
Please enable debugging output <https://atar-axis.github.io/xpadneo/index.html#debugging>
and add at least the following outputs
-   `uname -a`
Linux Tobias-Linux 5.1.3-1-default #1 SMP Fri May 17 04:54:29 UTC 2019 (07d2e25) x86_64 x86_64 x86_64 GNU/Linux

-   `dmesg`
https://pastebin.com/sdiHBecB

-   `sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;`
https://pastebin.com/6cKtnq2d

entry in /sys/bus/hid/...
/sys/bus/hid/devices/0005:045E:02FD.0006> ls -la
insgesamt 0
drwxr-xr-x 3 root root    0 23. Mai 18:36 .
drwxr-xr-x 4 root root    0 23. Mai 18:36 ..
-r--r--r-- 1 root root 4096 23. Mai 18:37 modalias
drwxr-xr-x 2 root root    0 23. Mai 18:36 power
-r--r--r-- 1 root root 4096 23. Mai 18:37 report_descriptor
lrwxrwxrwx 1 root root    0 23. Mai 18:36 subsystem -> ../../../../../../../../../../../bus/hid
-rw-r--r-- 1 root root 4096 23. Mai 18:36 uevent

**Additional context**  
Add any other context about the problem here.
If I can help, let me know. :)

atar-axis, 2019-05-23:
Hey there!
Thank you for reporting that issue.

What's the output of `modinfo hid_xpadneo` and which distribution are you running (i.e. what package manager do you use)?

germangoergs, 2019-05-24:
```
modinfo hid_xpadneo
Absolute path to 'modinfo' is '/usr/sbin/modinfo', so running it may require superuser privileges (eg. root).
tobias@Tobias-Linux:~> sudo modinfo hid_xpadneo
filename:       /lib/modules/5.1.3-1-default/updates/hid-xpadneo.ko
version:        0.6.0
description:    Linux kernel driver for Xbox ONE S+ gamepads (BT), incl. FF
author:         Florian Dollinger <dollinger.florian@gmx.de>
license:        GPL
suserelease:    openSUSE Tumbleweed
srcversion:     644F5CE4A16728D1F4BBB39
alias:          hid:b0005g*v0000045Ep000002E0
alias:          hid:b0005g*v0000045Ep000002FD
depends:        ff-memless
retpoline:      Y
name:           hid_xpadneo
vermagic:       5.1.3-1-default SMP mod_unload modversions 
parm:           debug_level:(u8) Debug information level: 0 (none) to 3+ (most verbose). (byte)
parm:           disable_ff:(u8) Disable FF: 0 (all enabled), 1 (disable main), 2 (disable triggers), 3 (disable all). (byte)
parm:           combined_z_axis:(bool) Combine the triggers to form a single axis. 1: combine, 0: do not combine (bool)
parm:           trigger_rumble_damping:(u8) Damp the trigger: 1 (none) to 2^8+ (max) (byte)
parm:           fake_dev_version:(u16) Fake device version # to hide from SDL's mappings. 0x0001-0xFFFF: fake version, others: keep original (ushort)
```

```
sudo dkms status
hid-xpadneo, 0.6.0, 5.1.3-1-default, x86_64: installed
```

I'm currently running OpenSUSE Tumbleweed on kernel 5.1.3-1. :)

That's interesting, it literally stopped working after I updated to Kernel 5.1, I had the vmouse version installed, tried to reinstall it, still didn't work, uninstalled everything and reinstalled it, but it was still not working, uninstalled it again, installed normal version, but still no success.
Thought it could be something else, so I waited until the next update, but still not recognized properly. :(

atar-axis, 2019-05-24:
Hum, looks like it is installed correctly - so maybe it is the udev rule then which does not work as desired...

Could you give me the last ~20 lines of `dmesg` after running `sudo modprobe hid_xpadneo` please :)

germangoergs, 2019-05-24:
```
dmesg:

[   13.823138] VBoxPciLinuxInit
[   13.823141] vboxpci: IOMMU found
[   13.923569] bpfilter: Loaded bpfilter_umh pid 1776
[   13.938856] No iBFT detected.
[   13.950557] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[   14.600316] br0: port 1(enp0s31f6) entered blocking state
[   14.600321] br0: port 1(enp0s31f6) entered disabled state
[   14.600447] device enp0s31f6 entered promiscuous mode
[   17.478348] e1000e: enp0s31f6 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
[   17.478435] br0: port 1(enp0s31f6) entered blocking state
[   17.478438] br0: port 1(enp0s31f6) entered forwarding state
[   17.478595] IPv6: ADDRCONF(NETDEV_CHANGE): br0: link becomes ready
[   17.569573] NET: Registered protocol family 17
[   20.561148] Loading iSCSI transport class v2.0-870.
[   20.669534] nf_conntrack: default automatic helper assignment has been turned off for security reasons and CT-based  firewall rule not found. Use the iptables CT target to attach helpers instead.
[   26.571894] snd_hda_codec_hdmi hdaudioC1D0: HDMI: invalid ELD data byte 88
[   31.404954] Bluetooth: RFCOMM TTY layer initialized
[   31.404961] Bluetooth: RFCOMM socket layer initialized
[   31.404964] Bluetooth: RFCOMM ver 1.11
[   79.443087] BTRFS warning (device sda1): cannot snapshot subvolume with active swapfile
[  125.855429] BTRFS warning (device sda1): cannot snapshot subvolume with active swapfile
[  622.372615] BTRFS info (device sda1): qgroup scan completed (inconsistency flag cleared)
[ 5869.417812] device-mapper: crypt: xts(aes) using implementation "xts-aes-aesni"
[12544.040444] xpadneo: hello there!
```

```
lsmod | grep xpadneo
hid_xpadneo            24576  0
ff_memless             20480  1 hid_xpadneo
```

```
udevadm --version
242
```

germangoergs, 2019-05-24:
```
ls /etc/udev/rules.d/99-xpadneo.rules

/etc/udev/rules.d/99-xpadneo.rules
```

atar-axis, 2019-05-24:
sorry, I meant `cat` not `ls`

germangoergs, 2019-05-24:
```
cat /etc/udev/rules.d/99-xpadneo.rules
# XPADNEO UDEV BINDING
# by atar-axis (dollinger.florian@gmx.de)

# in general those files with a higher leading number are processed
# and executed _later_, that means assignment in those files can overwrite
# assignments in other rules which are parsed earlier

# every block is a rule. rules are collected from the .rules files
# according to their name. on every kernel uevent, all rules are
# compared to that event - one by one.

# kernel versions        manual binding (bind/unbind)   bind uevent             hid_generic greedy      hid_microsoft support   gamepad     method
# ---------------        ----------------------------   --------------          ------------------      ---------------------   -------     ------
# [    ... - 2.6.13 [    no                             no                      yes                     no                      *           rmmod (not supported in xpadneo)
# [ 2.6.13 - 4.14   [    yes                            no                      yes                     no                      *           1: add, hid_generic
# [   4.14 - 4.16   [    yes                            yes                     yes                     no                      *           2: bind, hid_generc
# [   4.16 - 4.20   [    yes                            yes                     no                      no                      *           automatic
# [   4.20 - ...    ]    yes                            yes                     no                      yes                     02FD        3: bind, microsoft
#                                                                                                                               02E0        automatic

ACTION=="add", \
KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*", \
SUBSYSTEM=="hid", \
RUN:="/bin/sh -c 'echo xpadneo udev: $kernel > /dev/kmsg; modprobe hid_xpadneo; echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind; echo $kernel > /sys/bus/hid/drivers/microsoft/unbind; echo $kernel > /sys/bus/hid/drivers/xpadneo/bind; echo xpadneo udev: ok > /dev/kmsg'"
```

atar-axis, 2019-05-24:
Hum, everything is good up to here, ...

Please run `modprobe hid_xpadneo` again and, afterwards show me the output of `ls /sys/bus/hid/drivers/`

germangoergs, 2019-05-24:
```
ls /sys/bus/hid/drivers/
hid-generic  xpadneo

```

atar-axis, 2019-05-24:
Okay...
So lets summarize:

1. the udev rule is executed, I know that because in your dmesg appears a "xpadneo udev: ok", which is the last command in the udev rule
2. your module is at least loadable by hand, which we know by running `lsmod` after modprobe
3. xpadneo is **not** loaded by the udev rule, because we are missing the `hello there` message
4. the problem is not the bind command, at least the path is correct

so in result it looks like the `modprobe xpadneo` command in your udev rule does not work on your system anymore. let's proof that:

1. Change the last line in 99-xpadneo.rules into:
`RUN:="/bin/sh -c 'echo xpadneo udev: $kernel > /dev/kmsg; modprobe hid_xpadneo && echo /dev/kmsg; echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind; echo $kernel > /sys/bus/hid/drivers/microsoft/unbind; echo $kernel > /sys/bus/hid/drivers/xpadneo/bind; echo xpadneo udev: ok > /dev/kmsg'"`
2. reboot your system
3. connect the gamepad and show me your `dmesg`

germangoergs, 2019-05-26:
The dmesg
https://pastebin.com/UWiMrbA8

atar-axis, 2019-05-26:
Have you rebooted? Looks like the old udec rule in dmesg

germangoergs, 2019-05-26:
Yep, and I replaced it.

That's very strange. O.o

atar-axis, 2019-05-26:
Can you please give `sudo udevam --reload` a try? Reboot afterwards and show me the log ;)

germangoergs, 2019-05-26:
I'd like to do so, but it couldn't find the command. :(

atar-axis, 2019-05-26:
Well... Because it should be `sudo udevadm control --reload`, sorry

germangoergs, 2019-05-26:
No problem. ;)
You brought us this driver and you try to do your best on enhancing the whole user experience, so I'd be really stupid upset. :)

the dmesg:
https://pastebin.com/JD2HCnCt

atar-axis, 2019-05-26:
Haha :)
But,... it's still the same, something is really going wrong there. It is still the old rule which is executed...

Let's have a look at the udev rule again, maybe something went wrong there while copy&paste:
`cat /etc/udev/rules.d/99-xpadneo.rules`
Or maybe there is another rule which overwrites the other one?
`ls /usr/lib/udev/rules.d`

germangoergs, 2019-05-26:
```
cat /etc/udev/rules.d/99-xpadneo.rules
# XPADNEO UDEV BINDING
# by atar-axis (dollinger.florian@gmx.de)

# in general those files with a higher leading number are processed
# and executed _later_, that means assignment in those files can overwrite
# assignments in other rules which are parsed earlier

# every block is a rule. rules are collected from the .rules files
# according to their name. on every kernel uevent, all rules are
# compared to that event - one by one.

# kernel versions        manual binding (bind/unbind)   bind uevent             hid_generic greedy      hid_microsoft support   gamepad     method
# ---------------        ----------------------------   --------------          ------------------      ---------------------   -------     ------
# [    ... - 2.6.13 [    no                             no                      yes                     no                      *           rmmod (not supported in xpadneo)
# [ 2.6.13 - 4.14   [    yes                            no                      yes                     no                      *           1: add, hid_generic
# [   4.14 - 4.16   [    yes                            yes                     yes                     no                      *           2: bind, hid_generc
# [   4.16 - 4.20   [    yes                            yes                     no                      no                      *           automatic
# [   4.20 - ...    ]    yes                            yes                     no                      yes                     02FD        3: bind, microsoft
#                                                                                                                               02E0        automatic

ACTION=="add", \
KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*", \
SUBSYSTEM=="hid", \
RUN:="/bin/sh -c 'echo xpadneo udev: $kernel > /dev/kmsg; modprobe hid_xpadneo; echo modprobe $? > /dev/kmsg; echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind; echo unbind_generic $? > /dev/kmsg; echo $kernel > /sys/bus/hid/drivers/microsoft/unbind; echo unbind_ms $? > /dev/kmsg; echo $kernel > /sys/bus/hid/drivers/xpadneo/bind; echo bind_xpadneo $? > /dev/kmsg; echo xpadneo udev: ok > /dev/kmsg'"
```
```

ls /usr/lib/udev/rules.d
00-dont-del-part-nodes.rules   60-persistent-input.rules             70-touchpad.rules                         80-libinput-device-groups.rules
01-md-raid-creating.rules      60-persistent-storage.rules           70-uaccess.rules                          80-mm-candidate.rules
10-dm.rules                    60-persistent-storage-tape.rules      71-seat.rules                             80-net-setup-link.rules
11-dm-lvm.rules                60-persistent-v4l.rules               73-seat-late.rules                        80-stelladaptor-joystick.rules
11-dm-mpath.rules              60-scdaemon.rules                     75-net-description.rules                  80-udisks2.rules
11-dm-parts.rules              60-sensor.rules                       75-probe_mtd.rules                        84-nm-drivers.rules
13-dm-disk.rules               60-serial.rules                       75-rdma-description.rules                 85-nm-unmanaged.rules
39-usbmuxd.rules               60-steam-input.rules                  77-mm-cinterion-port-types.rules          85-regulatory.rules
40-libgphoto2.rules            60-steam-vr.rules                     77-mm-dell-port-types.rules               89-alsa-ucm.rules
40-usb-blacklist.rules         60-vboxdrv.rules                      77-mm-ericsson-mbm.rules                  90-alsa-restore.rules
40-usb-media-players.rules     61-kde-bluetooth-rfkill.rules         77-mm-fibocom-port-types.rules            90-haveged.rules
40-usb_modeswitch.rules        61-persistent-storage-compat.rules    77-mm-haier-port-types.rules              90-issue-generator.rules
42-hd-audio-pm.rules           63-md-raid-arrays.rules               77-mm-huawei-net-port-types.rules         90-libgpod.rules
42-usb-hid-pm.rules            64-btrfs-dm.rules                     77-mm-longcheer-port-types.rules          90-libinput-model-quirks.rules
50-iscsi-firmware-login.rules  64-btrfs.rules                        77-mm-mtk-port-types.rules                90-nm-thunderbolt.rules
50-udev-default.rules          64-md-raid-assembly.rules             77-mm-nokia-port-types.rules              90-pulseaudio.rules
51-android.rules               65-libwacom.rules                     77-mm-pcmcia-device-blacklist.rules       90-rdma-hw-modules.rules
54-before-scsi-sg3_id.rules    65-wacom.rules                        77-mm-sierra.rules                        90-rdma-ulp-modules.rules
55-scsi-sg3_id.rules           66-kpartx.rules                       77-mm-simtech-port-types.rules            90-rdma-umad.rules
56-hpmud.rules                 68-del-part-nodes.rules               77-mm-telit-port-types.rules              90-vconsole.rules
56-multipath.rules             69-brltty.rules                       77-mm-ublox-port-types.rules              95-dm-notify.rules
58-scsi-sg3_symlink.rules      69-dm-lvm-metad.rules                 77-mm-usb-device-blacklist.rules          95-upower-csr.rules
59-fc-wwpn-id.rules            69-libmtp.rules                       77-mm-usb-serial-adapters-greylist.rules  95-upower-hid.rules
60-block.rules                 69-md-clustered-confirm-device.rules  77-mm-x22x-port-types.rules               95-upower-wup.rules
[... cut ...]
60-drm.rules                   70-joystick.rules                     78-sound-card.rules                       98-rdma-sriov.rules
60-evdev.rules                 70-kdump.rules                        80-drivers.rules                          99-iwlwifi-led.rules
60-input-id.rules              70-mouse.rules                        80-g810-led.rules                         99-systemd.rules
60-io-scheduler.rules          70-power-switch.rules                 80-hotplug-cpu-mem.rules                  99-wakeup-from-idle.rules
```

atar-axis, 2019-05-27:
I am really sorry, I am a bit clueless right now... no Idea what is going wrong there on your system.
What happens if you run `/bin/sh -c 'echo xpadneo udev: $kernel > /dev/kmsg; modprobe hid_xpadneo; echo modprobe $? > /dev/kmsg; echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind; echo unbind_generic $? > /dev/kmsg; echo $kernel > /sys/bus/hid/drivers/microsoft/unbind; echo unbind_ms $? > /dev/kmsg; echo $kernel > /sys/bus/hid/drivers/xpadneo/bind; echo bind_xpadneo $? > /dev/kmsg; echo xpadneo udev: ok > /dev/kmsg'` when your gamepad is connected?

germangoergs, 2019-05-27:
That way it actually works, but I get the following message:
```
/bin/sh: Zeile 0: echo: Schreibfehler: Kein passendes Gerät gefunden.
/bin/sh: /sys/bus/hid/drivers/microsoft/unbind: Datei oder Verzeichnis nicht gefunden
/bin/sh: Zeile 0: echo: Schreibfehler: Kein passendes Gerät gefunden.
```

atar-axis, 2019-05-27:
I know, it is going to be annoying... but could you please delete the whole content of the udev rule
and replace it by:

```
ACTION=="add", \
KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*", \
SUBSYSTEM=="hid", \
RUN:="/bin/sh -c 'echo xpadneo udev: debug > /dev/kmsg'"
```

reboot afterwards, connect your gamepad and show me the dmesg.

germangoergs, 2019-05-29:
Sorry for delay, here you go:
https://pastebin.com/wDmEVwd3

atar-axis, 2019-06-02:
No problem and sorry too, I am quite busy at the moment.
We still have the old `xpadneo udev: ok` in dmesg (last line)... that's so weird!
We have to find where this does come from:

`grep -r " ok " /etc/udev/rules.d/; grep -r " ok " /usr/lib/udev/rules.d/`

germangoergs, 2019-06-04:
dmesg with ok at the end:
https://pastebin.com/tC3TK7Ve

the other command didn't output nothing.

germangoergs, 2019-06-04:
Could it be that it shows "ok" at the end because that's my manual try to connect it and that works? O.o

atar-axis, 2019-06-04:
yes :) that's the reason! do not bind it to xpadneo manually, just connect it via bluetoothctl :)

atar-axis, 2019-06-04:
Okay, lets give it another try...

1. Remove xpadneo by running `./uninstall.sh`
2. Reboot
3. Install xpadneo again
4. Replace the last line in `/etc/udev/rules.d/99-xpadneo.rules` with the following:
`RUN:="/bin/sh -c 'echo xpadneo udev: $kernel > /dev/kmsg; modprobe hid_xpadneo && { echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind; echo $kernel > /sys/bus/hid/drivers/microsoft/unbind; echo $kernel > /sys/bus/hid/drivers/xpadneo/bind; }; echo xpadneo udev: ok > /dev/kmsg'"`
5. Run `sudo udevadm control --reload-rules && sudo udevadm` trigger afterwards
6. Start your gamepad - if it does not connect (solid white light), then please connect it via `bluetoothctl`
7. Show me your `dmesg`

Please do **not** run any other commands :grin:

germangoergs, 2019-06-04:
I've done everything as you said, uninstalled it, rebooted, installed it again, replaced the line, rebooted, run that command, removed my gamepad via bluetoothctl, paired it, connected to it and that's the dmesg:
https://pastebin.com/5miR3wq1

atar-axis, 2019-06-04:
write me a email please, I think we will have to solve this via whatsapp or sth like that...

atar-axis, 2019-06-04:
the issue is in essence solved now, somehow the modprobe command failed - I don't know why and it is the first time I saw that. As a workaround you can replace the `RUN` line in 99-xpadneo.rules by `RUN+="/bin/sh -c 'echo xpadneo udev: $kernel > /dev/kmsg && { echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind; echo $kernel > /sys/bus/hid/drivers/microsoft/unbind; echo $kernel > /sys/bus/hid/drivers/xpadneo/bind; }; '"`

germangoergs, 2019-06-05:
So...
I made a fresh installation of OpenSUSE Tumbleweed and still, I can just connect the controller properly with the manual command, otherwise it won't work.
So it's not a problem of a screwed up system. :D

atar-axis, 2019-06-05:
Thanks! Therefore it looks like I have to find an alternative to the modprobe command withing the udev rule... neverending story. Stay tunes.

germangoergs, 2019-06-07:
I've updated my kernel a few moments ago again, here's the whole update log of my system.
I hope it helps you. :)
[history.txt](https://github.com/atar-axis/xpadneo/files/3267888/history.txt)

davw4453, 2019-07-05:
Hi I am having a similar problem on my fedora 30 kernel 5.1.15 installation. xpadneo installed successfully but when I connect my controller it does not vibrate and I believe hid-generic is still being used. I've tried unbinding hid-generic with "echo 0005:045E:02FD > /sys/bus/hid/drivers/hid-generic/unbind" but the output shows error: no such device. I've also noticed that in /sys/bus/hid/drivers there is no xpadneo file or folder. dmesg output is as follows:
 

dmesg | grep hid-generic
[    2.358889] hid-generic 0003:0C40:8000.0001: input,hidraw0: USB HID v1.11 Keyboard [ELMCU iPazzPort] on usb-0000:00:12.0-2/input0
[    2.416881] hid-generic 0003:0C40:8000.0002: input,hidraw1: USB HID v1.11 Mouse [ELMCU iPazzPort] on usb-0000:00:12.0-2/input1
[    2.586019] hid-generic 0003:1BCF:08A0.0003: input,hiddev96,hidraw2: USB HID v1.10 Mouse [HID 1bcf:08a0] on usb-0000:00:13.1-2/input0
[    3.271966] hid-generic 0003:046D:C31C.0004: input,hidraw3: USB HID v1.10 Keyboard [Logitech USB Keyboard] on usb-0000:00:13.1-3/input0
[    3.333930] hid-generic 0003:046D:C31C.0005: input,hidraw4: USB HID v1.10 Device [Logitech USB Keyboard] on usb-0000:00:13.1-3/input1
[    9.262868] hid-generic 0005:045E:02FD.0006: unknown main item tag 0x0
[    9.263268] hid-generic 0005:045E:02FD.0006: input,hidraw5: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 00:19:86:00:25:f7
[   43.328784] hid-generic 0005:045E:02FD.0007: unknown main item tag 0x0
[   43.329464] hid-generic 0005:045E:02FD.0007: input,hidraw5: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 00:19:86:00:25:f7
[  410.272559] hid-generic 0005:045E:02FD.0008: unknown main item tag 0x0
[  410.273294] hid-generic 0005:045E:02FD.0008: input,hidraw5: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 00:19:86:00:25:f7
[ 1120.098630] hid-generic 0005:045E:02FD.0009: unknown main item tag 0x0
[ 1120.099347] hid-generic 0005:045E:02FD.0009: input,hidraw5: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 00:19:86:00:25:f7

dmesg | grep xpadneo
[    9.268324] xpadneo udev: 0005:045E:02FD.0006
[    9.296054] xpadneo udev: ok
[   43.333814] xpadneo udev: 0005:045E:02FD.0007
[   43.387884] xpadneo udev: ok
[  410.277800] xpadneo udev: 0005:045E:02FD.0008
[  410.326551] xpadneo udev: ok
[ 1120.103544] xpadneo udev: 0005:045E:02FD.0009
[ 1120.146693] xpadneo udev: ok

From this I think maybe its trying to use both hid-generic and xpadneo at once?

nekocentral, 2019-10-12:
hey atar-axis, when you have some udev rules you want to test please send them and i will test them

sinkillerj, 2020-01-06:
Can confirm that after an update to 5.3.16-200.fc30.x86_64 the dirty fix is still needed, but rumble support is seemingly not detected by any app including fftest.

currybullen, 2020-03-08:
Did some experimenting with this on Fedora 31, I think the problem is not that modprobe fails, but that modprobe is not on the PATH when the udev rule runs. Simply switching out modprobe for /usr/sbin/modprobe works for me. I submitted a somewhat ugly fix for this in pull request #167

jefersonnavarro, 2020-05-05:
https://github.com/atar-axis/xpadneo/issues/179#issuecomment-623792824

kakra, 2020-05-09:
Closing as duplicate of #167

closed 2020-05-09 by kakra