atar-axis/xpadneo #110: No input & Connection drops

vahtos, 2019-05-26:
Hello,

I have two computers, one running Kubuntu 18.04, and one running Manjaro 18.0.4. I was able to connect my Xbox Wireless Controller to the Kubuntu computer, but cannot get it to work for the Manjaro computer.

There are no errors when installing, but when I use sudo bluetoothctl, I will randomly get errors connecting to the controller. If I keep trying eventually I can get it to connect, the controller rumbles and the light stays solid.

When it connects, Dmesg shows:

https://ghostbin.com/paste/efanc

However, there is no input and after about a minute the light goes back to flashing (and will eventually shut off), and the controller no longer shows as connected on the computer.

I searched the issues here and made sure disable_ertm was off and it is:

> ~ >>> cat /sys/module/bluetooth/parameters/disable_ertm                                                                                                                                                                                                                                  
> Y

Any help would be greatly appreciated!

atar-axis, 2019-05-26:
What kind of Bluetooth device are you using on this PC?

vahtos, 2019-05-26:
Internal. I have an Asus Maximus Hero XI motherboard.

atar-axis, 2019-05-26:
Bluetooth connection problems are never related to the gamepad driver because the connection is made by the kernel and the bluetooth driver. If you have connection problems it is most probably a bluetooth driver problem.

Can you give a bluetooth dongle a try?

vahtos, 2019-05-26:
Sure, is there a preferred dongle I should get?

vahtos, 2019-05-27:
Turned off Bluetooth for the motherboard in the bios settings, plugged in the dongle and everything works perfectly out of the box. Thanks!

atar-axis, 2019-05-28:
Perfect! I'm glad it works now :)

closed 2019-05-28 by atar-axis