atar-axis/xpadneo #111: No Keyboard and Mouse after installing xpadneo

Frankstar, 2019-05-29:
installed xpadneo on Fedora 30.
it worked great for Fedora 29 - so i assumed there are no problem on Fedora 30.

Anyway i have the strange problem that i cant login on my Fedora since i dont have a keyboard (or mouse) since the install.

same problem when i try to boot into runlevel 3.
no keyboard for login.

i removed the `quite` flag from the kernel line to see whats going on.
Everytime i hotlog the keyboard, xpadneo is doing something.

if i press the Button `a` repeatedly and hotplug the keyboard, i will get the `a` on the login. so keyboard is working, but not after xpadneo is doing something with it.

i tryed to remove it manualy (via live cd).
removed it under `/usr/src/hid-xpadneo` and even remove it manual from the dkms dir `/etc/modprobe.d/xpadneo`

somehow it still gets loaded.
how can i remove it manually from the kernel ??
or disable/blacklist it ?

Frankstar, 2019-05-29:
dang it - i forgot to remove `/etc/udev/rules.d/99-xpadneo.rules`
now its working again (i have keyboard!!!)

now the other thing is, i deleted manualy alot of things... how to clean up now ? (dkms)

Frankstar, 2019-05-29:
well, first i try to remove everything.
after that i will try install it again and provide more logs.

as expected `./uninstall.sh` cant find anything.
```
 sudo ./uninstall.sh 
* unloading current driver module
* looking for registered instances
found 0 registered instance(s) on your system
```

atar-axis, 2019-05-29:
yeah for sure, no problem.

take  a look into `/usr/src/` - is there any xpadneo directory in there? if yes, remove it

Frankstar, 2019-05-29:
> Overall it looks like the udev rule is messing up your system, but we will see after reinstalling it.

yes, also my thought
BUT right now its working.
`/etc/udev/rules.d/99-xpadneo.rules` :

```
# kernel versions        manual binding (bind/unbind)   bind uevent             hid_generic greedy      hid_microsoft support   gamepad     method
# ---------------        ----------------------------   --------------          ------------------      ---------------------   -------     ------
# [    ... - 2.6.13 [    no                             no                      yes                     no                      *           rmmod (not supported in xpadneo)
# [ 2.6.13 - 4.14   [    yes                            no                      yes                     no                      *           1: add, hid_generic
# [   4.14 - 4.16   [    yes                            yes                     yes                     no                      *           2: bind, hid_generc
# [   4.16 - 4.20   [    yes                            yes                     no                      no                      *           automatic
# [   4.20 - ...    ]    yes                            yes                     no                      yes                     02FD        3: bind, microsoft
#                                                                                                                               02E0        automatic

ACTION=="add", \
KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*", \
SUBSYSTEM=="hid", \
RUN:="/bin/sh -c 'echo xpadneo udev: $kernel > /dev/kmsg; modprobe hid_xpadneo; echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind; echo $kernel > /sys/bus/hid/drivers/microsoft/unbind; echo $kernel > /sys/bus/hid/drivers/xpadneo/bind; echo xpadneo udev: ok > /dev/kmsg'"
~                                                                                                                           
```

Frankstar, 2019-05-29:
i will observe this, if i cant reproduce, i will close the task.

atar-axis, 2019-06-04:
Reopen it if necessary ;)

closed 2019-06-04 by atar-axis

Frankstar, 2019-06-04:
well, i couldnt reproduce it.
it seems working now.
i also dont understand WHAT happend - looked over the install script, nothing there could create that issue ...