atar-axis/xpadneo #112: Install fails on updated clean install of Fedora 30

dumpsterfireplace, 2019-06-02:
**Describe the bug**
Install failing on newly installed and updated Fedora 30 system.

**To Reproduce**
1. Install Fedora 30
Clean install.  Updated everything using Software app.

2. Install Prereqs
sudo dnf install dkms bluez bluez-tools kernel-devel-`uname -r` kernel-headers-`uname -r`

3. Install xpadneo
I downloaded repo and ran install.sh.

**Expected behavior**
The installer to successfully complete.  I've setup xpadneo a couple times successfully in the past when running Fedora 28 and Fedora 29.  It also installed successfully on my prior Fedora 30 install, but eventually did a clean install after being unable to troubleshoot issues which I believe may have been caused by testing ds4drv first. The first possible cause that comes to mind is a problem with a new kernel version, since it looks like the current kernel is version 5.1.5 and my previous install was running 5.0.17.

**Screenshots/Gifs**
None

**System information**
Since the driver won't install, not sure this will provide much.

- `echo 3 | sudo tee /sys/module/hid_xpadneo/parameters/debug_level`
tee: /sys/module/hid_xpadneo/parameters/debug_level: No such file or directory
3

-   `uname -a`
Linux localhost.localdomain 5.1.5-300.fc30.x86_64 #1 SMP Sat May 25 18:00:11 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

-   `dmesg`
Since xpadneo won't complete installation, I have not connected my xbox controller to keep things as clean as possible.  This output is massive and irrelevant I believe, if the controller isnt connected. Maybe add this when I find how to add this content to a smaller scroll box.

-   `sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;`
No output

**Additional context**  
This is my second attempt at setup on this hardware and Fedora 30, and experienced something different the first time.  The first attempt was after I had setup ds4drv to see how a ps4 controller works.  I didn't like some limitations of the driver, but the controller worked properly.  I uninstalled it and installed xpadneo to use my xbox controller.  Xpadneo installed successfully, but I experienced bluetooth connectivity issues (even after disabling ertm) which I have not experienced before with this controller.  Also most of the time it wouldnt rumble, which means the correct driver wasnt being loaded, I believe.  I could not resolve this, and figured ds4drv messed something up, so I reformatted and did a clean install of Fedora.

Now all I've done to the system after the clean install was update it via Software, then install xpadneo following the readme page.  So there is little room for error or conflicts.  The xpadneo install fails with the following messages:
`[mediashare@localhost xpadneo]$ sudo ./install.sh
* replacing version string if necessary
* copying module into /usr/src
* adding module to DKMS

Creating symlink /var/lib/dkms/hid-xpadneo/0.6.0/source ->
                 /usr/src/hid-xpadneo-0.6.0

DKMS: add completed.
* installing module (using DKMS)

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...(bad exit status: 127)
make -j6 KERNELRELEASE=5.1.5-300.fc30.x86_64 -C /lib/modules/5.1.5-300.fc30.x86_64/build M=/var/lib/dkms/hid-xpadneo/0.6.0/build/src modules...(bad exit status: 127)
Error! Bad return status for module build on kernel: 5.1.5-300.fc30.x86_64 (x86_64)
Consult /var/lib/dkms/hid-xpadneo/0.6.0/build/make.log for more information.'

Contents of /var/lib/dkms/hid-xpadneo/0.6.0/build/make.log:
DKMS make.log for hid-xpadneo-0.6.0 for kernel 5.1.5-300.fc30.x86_64 (x86_64)
Sun 02 Jun 2019 10:12:53 AM EDT
/usr/sbin/dkms: line 92: make: command not found

I verified dkms was installed by running the command to install all prereqs again and it shows all installed:
Last metadata expiration check: 0:14:58 ago on Sun 02 Jun 2019 12:23:47 PM EDT.
Package dkms-2.6.1-3.fc30.noarch is already installed.
Package bluez-5.50-6.fc30.x86_64 is already installed.
Package bluez-tools-0.2.0-0.10.git20170912.7cb788c.fc30.x86_64 is already installed.
Package kernel-devel-5.1.5-300.fc30.x86_64 is already installed.
Package kernel-headers-5.1.5-300.fc30.x86_64 is already installed.
Dependencies resolved.
Nothing to do.
Complete!

I am newish to Linux, but willing to help.  Just let me know what I need to do to help.
Thanks

dumpsterfireplace, 2019-06-02:
I tried running Make on its own and apparently it wasn't even installed.  After installing it, the xpadneo install completed.  However, now I think Im back in the same boat which I originally blamed on ds4drv mucking up the system.  The controller pairs, trusts, connects successfully, but does not rumble after connection and is not recognized by Kodi (with joystick peripheral addon installed).  My ps4 controller worked without issues via blueooth before reinstalling the system and I've used this same xbox controller successfully with Fedora 28 and 29 on very similar hardware (just a model# bump.  I'll need to find a way to find exact bluetooth chipset info and compare them). There is an active issue open (#107 ) which may be similar.  

`dmesg`
https://pastebin.com/9shH8J7v

`sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;`
/sys/kernel/debug/hid/0005:045E:02FD.0008
05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 32 09 35 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 02 09 c5 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 02 09 c4 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0f 15 00 25 01 75 01 95 0f 81 02 15 00 25 00 75 01 95 01 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75 08 95 01 81 02 c0 00 
2579218694 1006

`modinfo hid_xpadneo`
filename:       /lib/modules/5.1.5-300.fc30.x86_64/extra/hid-xpadneo.ko.xz
version:        0.6.0
description:    Linux kernel driver for Xbox ONE S+ gamepads (BT), incl. FF
author:         Florian Dollinger <dollinger.florian@gmx.de>
license:        GPL
srcversion:     644F5CE4A16728D1F4BBB39
alias:          hid:b0005g*v0000045Ep000002E0
alias:          hid:b0005g*v0000045Ep000002FD
depends:        ff-memless
retpoline:      Y
name:           hid_xpadneo
vermagic:       5.1.5-300.fc30.x86_64 SMP mod_unload 
parm:           debug_level:(u8) Debug information level: 0 (none) to 3+ (most verbose). (byte)
parm:           disable_ff:(u8) Disable FF: 0 (all enabled), 1 (disable main), 2 (disable triggers), 3 (disable all). (byte)
parm:           combined_z_axis:(bool) Combine the triggers to form a single axis. 1: combine, 0: do not combine (bool)
parm:           trigger_rumble_damping:(u8) Damp the trigger: 1 (none) to 2^8+ (max) (byte)
parm:           fake_dev_version:(u16) Fake device version # to hide from SDL's mappings. 0x0001-0xFFFF: fake version, others: keep original (ushort)

`dkms status`
hid-xpadneo, 0.6.0, 5.1.5-300.fc30.x86_64, x86_64: installed

atar-axis, 2019-06-02:
Thank you for the detailed bug report!
The issue looks in fact very similar to #107, which is still unsolved unfortunately.

Can you please replace the last line in `/etc/udev/rules.d/99-xpadneo.rules` with the following:
`RUN:="/bin/sh -c 'echo xpadneo udev: $kernel > /dev/kmsg; modprobe hid_xpadneo && { echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind; echo $kernel > /sys/bus/hid/drivers/microsoft/unbind; echo $kernel > /sys/bus/hid/drivers/xpadneo/bind; }; echo xpadneo udev: ok > /dev/kmsg'"`

Run `sudo udevadm control --reload-rules && sudo udevadm trigger` afterwards, reboot and show me your dmesg after connecting the gamepad

dumpsterfireplace, 2019-06-03:
I updated the RUN line in that file to what you posted and ran the udevadm command and rebooted.  However, now the controller just disconnects/reconnects repeatedly, so I cant get the next step for you.

I hope this isn't a problem with the bluetooth chipset.  It's an internal bluetooth and I expected to use it.  Not sure why a brand new Intel bluetooth 5.0 chipset would not work properly. Unless it's too new for linux and not supported well yet lol.  The specs page lists "Connectivity: Intel 9560 802.11AC (2 x 2) with vPro® + Bluetooth 5.0".

atar-axis, 2019-06-03:
You can try to `remove` the gamepad, reboot and reconnect it. Otherwise I would recommend to try a dongle, some chipsets have a timing problem, which I described in another issue a bit more detailed. Furthermore, it could be a ERTM problem, but I don't think so.

atar-axis, 2019-06-04:
the issue is in essence solved now, somehow the modprobe command failed - I don't know why and it is the first time I saw that. As a workaround you can replace the `RUN` line in 99-xpadneo.rules by `RUN+="/bin/sh -c 'echo xpadneo udev: $kernel > /dev/kmsg && { echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind; echo $kernel > /sys/bus/hid/drivers/microsoft/unbind; echo $kernel > /sys/bus/hid/drivers/xpadneo/bind; }; '"`

dumpsterfireplace, 2019-06-28:
Thanks for keeping this open.  I've kept my eye out for a firmware update for the bluetooth and there was one.  Pretty sure it was package "linux-firmware-20190618-97.fc30.src.rpm" which the changelog specifically mentioned this Intel 9560 chipset.  

I installed that update, updated cloned xpadneo repo (already up to date), uninstalled xpadneo, reinstalled it, restarted, and tried setting it up again.  I have no problem connecting it with bluetoothctl now.  The controller stays connected now.  But when it connects, it does not rumble, which you state means the wrong driver is loaded, but Im not finding how to troubleshoot that issue. How could I continue from here?

atar-axis, 2019-06-28:
https://github.com/atar-axis/xpadneo/issues/107#issuecomment-498858788

Maybe the same issue?

dumpsterfireplace, 2019-06-28:
I replaced the RUN command in the rule file like that comment said, restarted, and yes, the controller rumbles when connecting now and now it works properly! Thank you for all your help!

So to summarize this issue, I'd say we worked through three layers.  
1. Fedora 30 needed Make installed.
2. Intel 9560 bluetooth chipset needed the linux-firmware-20190618 update.
3. xpadneo modprobe command within the udev rule.

nekocentral, 2019-09-28:
Joining in, as i'm running fedora 30 with the latest kernel,
my Xbox one S bluetooth controller does connect but does not get picked up by the drivers.
When connecting an USB cable the controller does rumble and work.

If you want to to send some debug information then please tell me and i will send you as much as i can.

atar-axis, 2019-10-03:
This is a replicate of #107 in my eyes, please use the other issue for any further discussion

nekocentral, 2019-10-03:
Should I open a separate issue for being able to connect using cable but not Bluetooth,(it does connect but no ruble and no game detect it it on fedora 30 with 5.1 kernel)

atar-axis, 2019-10-03:
Have you tried the solution in the other thread?

nekocentral, 2019-10-03:
About to chance the use rule when I find my charger will update you on it tomorrow as it's almost 12am for me

nekocentral, 2019-10-11:
not yet, but had a kernel update today so going to reinstall xpadneo later today

atar-axis, 2019-10-11:
perfect, let me know if something changes :)

atar-axis, 2019-10-11:
so let me close this one again since it is a duplicate of #107 then :)

closed 2019-10-11 by atar-axis

nekocentral, 2019-10-11:
yea but i would change #107 to kernel 5.X as its also on 5.2