atar-axis/xpadneo #116: Add a browser app

MeetPing, 2019-06-10:
**Is your feature request related to a problem? Please describe.**NO, well kinda  it is that i want to be able to configure this FAST

**Describe the solution you'd like** Make the app thing have a browser app i have the terminal thingi say [Xbox Wireless Controller]#  and i would like so i could say localhost or configure so it makes a localhost:1234  so i can go there to configure the controller  like ok so this yoysticks does this and that button left clicks and that right clcks that pressed button a like that

**Describe alternatives you've considered**
Using another driver

**Additional context**
[image]
I can make a design it but you will have to code it

MeetPing, 2019-06-10:
if you want me to design it i can

MeetPing, 2019-06-10:
@atar-axis this is a great way to configure if so i say it

MeetPing, 2019-06-10:
hey man if you have discord we could talk business there im
Friend me
`Mr.Mask#9047`

atar-axis, 2019-06-10:
feel free to work on it and start a pull request :) I like the idea but I don't think that there are many ppl around really needing it since configuration is already quite easy using the configure script

MeetPing, 2019-06-10:
but its hard for me to configure

MeetPing, 2019-06-10:
also i dont know how the configure stuff

MeetPing, 2019-06-10:
also i cant code this because i dont know how to configure this
and also the configuring browser app should be able like `A` is a `space` idk how to do that or anything

atar-axis, 2019-06-10:
Run the configure script, everything else is explained there

atar-axis, 2019-07-03:
I will close this because I do not have time to work on that and I don't see any need for it

closed 2019-07-03 by atar-axis