atar-axis/xpadneo #117: Xpadneo doesn't work

DirkMeister79, 2019-06-22:
**Describe the bug**
I've followed the instructions to install xpadneo on a raspberry pi (running Rasbian Stretch, version April 2019, kernel 4.14) and connect the xbox wireless controller. The controller is paired and does rumble. The controller doesn't do anything in games. I've installed the vmouse branch. The mouse doesn't respond to any of my input on the controller. I'm however able to switch between game/mouse mode. 

Thanks. I hope you can help me. Please let me know if more info is needed.

**System information**
Please enable debugging output <https://atar-axis.github.io/xpadneo/index.html#debugging>
and add at least the following outputs

-   `dmesg`
[jun22 20:28] xpadneo 0005:045E:02FD.0003: probing device: Xbox Wireless Controller
[  +0,000027] xpadneo 0005:045E:02FD.0003: REPORT (DESCRIPTOR) FIXUP HOOK
[  +0,000012] xpadneo: report-descr: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01
[  +0,000008] xpadneo: report-descr: a1 00 09 32 09 35 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 02 09 c5 15 00 26 ff 03 95 01 75
[  +0,000009] xpadneo: report-descr: 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 02 09 c4 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00
[  +0,000008] xpadneo: report-descr: 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01
[  +0,000007] xpadneo: report-descr: 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0f 15 00 25 01 75 01 95 0f 81 02 15 00 25 00
[  +0,000009] xpadneo: report-descr: 75 01 95 01 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 05
[  +0,000008] xpadneo: report-descr: 0c 09 01 85 02 a1 01 05 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03
[  +0,000007] xpadneo: report-descr: c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09
[  +0,000008] xpadneo: report-descr: 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15
[  +0,000007] xpadneo: report-descr: 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 85 04 05 06
[  +0,000007] xpadneo: report-descr: 09 20 15 00 26 ff 00 75 08 95 01 81 02 c0 00
[  +0,001761] xpadneo 0005:045E:02FD.0003: unknown main item tag 0x0
[  +0,000087] xpadneo 0005:045E:02FD.0003: driver:
[  +0,000009] xpadneo 0005:045E:02FD.0003: * version: 0.7.0
[  +0,000008] xpadneo 0005:045E:02FD.0003: hdev:
[  +0,000009] xpadneo 0005:045E:02FD.0003: * raw rdesc: (unfixed, see above)
[  +0,000010] xpadneo 0005:045E:02FD.0003: * raw rsize: 335
[  +0,000009] xpadneo 0005:045E:02FD.0003: * bus: 0x0005
[  +0,000009] xpadneo 0005:045E:02FD.0003: * report group: 1
[  +0,000009] xpadneo 0005:045E:02FD.0003: * vendor: 0x0000045E
[  +0,000009] xpadneo 0005:045E:02FD.0003: * version: 0x00000903
[  +0,000009] xpadneo 0005:045E:02FD.0003: * product: 0x000002FD
[  +0,000011] xpadneo 0005:045E:02FD.0003: * country: 33
[  +0,000009] xpadneo 0005:045E:02FD.0003: * driverdata: 0
[  +0,000011] xpadneo 0005:045E:02FD.0003: * serial: 3a:61:61:3a:63:39
[  +0,000009] xpadneo 0005:045E:02FD.0003: * physical location: 3a:37:32:3a:38:62
[  +0,000031] xpadneo 0005:045E:02FD.0003: UP: 0x10000, USG: 0x0030 -> EV: 0x003, INP: 0x000
[  +0,000020] xpadneo 0005:045E:02FD.0003: UP: 0x10000, USG: 0x0031 -> EV: 0x003, INP: 0x001
[  +0,000012] xpadneo 0005:045E:02FD.0003: UP: 0x10000, USG: 0x0032 -> EV: 0x003, INP: 0x003
[  +0,000013] xpadneo 0005:045E:02FD.0003: UP: 0x10000, USG: 0x0035 -> EV: 0x003, INP: 0x004
[  +0,000012] xpadneo 0005:045E:02FD.0003: UP: 0x20000, USG: 0x00C5 -> EV: 0x003, INP: 0x002
[  +0,000016] xpadneo 0005:045E:02FD.0003: UP: 0x20000, USG: 0x00C4 -> EV: 0x003, INP: 0x005
[  +0,000013] xpadneo 0005:045E:02FD.0003: UP: 0x10000, USG: 0x0039 -> automatically
[  +0,000015] xpadneo 0005:045E:02FD.0003: UP: 0x90000, USG: 0x0001 -> EV: 0x001, INP: 0x130
[  +0,000013] xpadneo 0005:045E:02FD.0003: UP: 0x90000, USG: 0x0002 -> EV: 0x001, INP: 0x131
[  +0,000011] xpadneo 0005:045E:02FD.0003: UP: 0x90000, USG: 0x0003 -> ignored
[  +0,000012] xpadneo 0005:045E:02FD.0003: UP: 0x90000, USG: 0x0004 -> EV: 0x001, INP: 0x133
[  +0,000013] xpadneo 0005:045E:02FD.0003: UP: 0x90000, USG: 0x0005 -> EV: 0x001, INP: 0x134
[  +0,000011] xpadneo 0005:045E:02FD.0003: UP: 0x90000, USG: 0x0006 -> ignored
[  +0,000011] xpadneo 0005:045E:02FD.0003: UP: 0x90000, USG: 0x0007 -> EV: 0x001, INP: 0x136
[  +0,000015] xpadneo 0005:045E:02FD.0003: UP: 0x90000, USG: 0x0008 -> EV: 0x001, INP: 0x137
[  +0,000010] xpadneo 0005:045E:02FD.0003: UP: 0x90000, USG: 0x0009 -> ignored
[  +0,000010] xpadneo 0005:045E:02FD.0003: UP: 0x90000, USG: 0x000A -> ignored
[  +0,000010] xpadneo 0005:045E:02FD.0003: UP: 0x90000, USG: 0x000B -> ignored
[  +0,000011] xpadneo 0005:045E:02FD.0003: UP: 0x90000, USG: 0x000C -> EV: 0x001, INP: 0x13B
[  +0,000011] xpadneo 0005:045E:02FD.0003: UP: 0x90000, USG: 0x000D -> ignored
[  +0,000012] xpadneo 0005:045E:02FD.0003: UP: 0x90000, USG: 0x000E -> EV: 0x001, INP: 0x13D
[  +0,000011] xpadneo 0005:045E:02FD.0003: UP: 0x90000, USG: 0x000F -> EV: 0x001, INP: 0x13E
[  +0,000012] xpadneo 0005:045E:02FD.0003: UP: 0xC0000, USG: 0x0224 -> EV: 0x001, INP: 0x13A
[  +0,000013] xpadneo 0005:045E:02FD.0003: UP: 0xC0000, USG: 0x0223 -> EV: 0x001, INP: 0x13C
[  +0,000010] xpadneo 0005:045E:02FD.0003: UP: 0x60000, USG: 0x0020 -> ignored
[  +0,000014] xpadneo 0005:045E:02FD.0003: INPUT CONFIGURED HOOK
[  +0,000010] xpadneo 0005:045E:02FD.0003: Fake device version: 0x1130
[  +0,000277] input: Xbox Wireless Controller as /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11/0005:045E:02FD.0003/input/input4
[  +0,009540] xpadneo 0005:045E:02FD.0003: input,hidraw0: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on b8:27:eb:2a:98:58
[  +0,006173] xpadneo udev: 0005:045E:02FD.0003
[  +0,021357] xpadneo udev: ok
[  +0,963131] xpadneo 0005:045E:02FD.0003: battery registered
[  +2,375432] xpadneo: raw_event: 04 87
[  +0,000050] xpadneo 0005:045E:02FD.0003: hid-up: 60000, hid-usg: 20, input-code: 00, value: 87
[  +0,000012] xpadneo 0005:045E:02FD.0003: REPORT HOOK

sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;
/sys/kernel/debug/hid/0005:045E:02FD.0003
05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 32 09 35 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 02 09 c5 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 02 09 c4 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0f 15 00 25 01 75 01 95 0f 81 02 15 00 25 00 75 01 95 01 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75 08 95 01 81 02 c0 00 
2579218694 1006

atar-axis, 2019-06-23:
sorry guys, I haven't had any spare time since weeks - I will get back to you as soon as possible

DirkMeister79, 2019-06-25:
No problem. I'll just wait for your next message.

atar-axis, 2019-06-25:
btw, does the non-vmouse branch work?
furthermore, what does `evtest /dev/input/event<#>` say if you press buttons? (replace # by the highest number if just connected the gp)

DirkMeister79, 2019-06-26:
The non-vmouse branch doesn't work either.

evtest gives the following output:

Event: time 1561579906.594420, type 3 (EV_ABS), code 3 (ABS_RX), value 2377
Event: time 1561579906.594420, -------------- SYN_REPORT ------------
Event: time 1561579906.594434, type 3 (EV_ABS), code 4 (ABS_RY), value 2305
Event: time 1561579906.594434, -------------- SYN_REPORT ------------
Event: time 1561579906.601886, type 3 (EV_ABS), code 1 (ABS_Y), value -1301
Event: time 1561579906.601886, -------------- SYN_REPORT ------------
Event: time 1561579906.601912, type 3 (EV_ABS), code 3 (ABS_RX), value 2320
Event: time 1561579906.601912, -------------- SYN_REPORT ------------
Event: time 1561579906.601927, type 3 (EV_ABS), code 4 (ABS_RY), value 2273
Event: time 1561579906.601927, -------------- SYN_REPORT ------------
Event: time 1561579906.609406, type 3 (EV_ABS), code 3 (ABS_RX), value 2278
Event: time 1561579906.609406, -------------- SYN_REPORT ------------
Event: time 1561579906.609432, type 3 (EV_ABS), code 4 (ABS_RY), value 2224
Event: time 1561579906.609432, -------------- SYN_REPORT ------------
Event: time 1561579906.624416, type 3 (EV_ABS), code 3 (ABS_RX), value 2228
Event: time 1561579906.624416, -------------- SYN_REPORT ------------
Event: time 1561579906.624439, type 3 (EV_ABS), code 4 (ABS_RY), value 2187
Event: time 1561579906.624439, -------------- SYN_REPORT ------------
Event: time 1561579906.631889, type 3 (EV_ABS), code 3 (ABS_RX), value 2191
Event: time 1561579906.631889, -------------- SYN_REPORT ------------
Event: time 1561579906.631912, type 3 (EV_ABS), code 4 (ABS_RY), value 2137
Event: time 1561579906.631912, -------------- SYN_REPORT ------------
Event: time 1561579906.669416, type 3 (EV_ABS), code 3 (ABS_RX), value 2147
Event: time 1561579906.669416, -------------- SYN_REPORT ------------
Event: time 1561579906.669442, type 3 (EV_ABS), code 4 (ABS_RY), value 2100
Event: time 1561579906.669442, -------------- SYN_REPORT ------------
Event: time 1561579906.691897, type 3 (EV_ABS), code 1 (ABS_Y), value -1260
Event: time 1561579906.691897, -------------- SYN_REPORT ------------
Event: time 1561579906.691922, type 3 (EV_ABS), code 3 (ABS_RX), value 2114
Event: time 1561579906.691922, -------------- SYN_REPORT ------------
Event: time 1561579906.736928, type 3 (EV_ABS), code 4 (ABS_RY), value 2050
Event: time 1561579906.736928, -------------- SYN_REPORT ------------
Event: time 1561579907.869433, type 3 (EV_ABS), code 4 (ABS_RY), value 2012
Event: time 1561579907.869433, -------------- SYN_REPORT ------------
Event: time 1561579907.899409, type 3 (EV_ABS), code 0 (ABS_X), value 1817
Event: time 1561579907.899409, -------------- SYN_REPORT ------------
Event: time 1561579907.929383, type 3 (EV_ABS), code 0 (ABS_X), value 1769
Event: time 1561579907.929383, -------------- SYN_REPORT ------------
Event: time 1561579908.986977, type 3 (EV_ABS), code 0 (ABS_X), value 1733
Event: time 1561579908.986977, -------------- SYN_REPORT ------------
Event: time 1561579908.987321, type 3 (EV_ABS), code 16 (ABS_HAT0X), value -1
Event: time 1561579908.987321, -------------- SYN_REPORT ------------
Event: time 1561579909.152070, type 3 (EV_ABS), code 16 (ABS_HAT0X), value 0
Event: time 1561579909.152070, -------------- SYN_REPORT ------------
Event: time 1561579909.377068, type 3 (EV_ABS), code 17 (ABS_HAT0Y), value 1
Event: time 1561579909.377068, -------------- SYN_REPORT ------------
Event: time 1561579909.557061, type 3 (EV_ABS), code 17 (ABS_HAT0Y), value 0
Event: time 1561579909.557061, -------------- SYN_REPORT ------------
Event: time 1561579909.782068, type 3 (EV_ABS), code 16 (ABS_HAT0X), value 1
Event: time 1561579909.782068, -------------- SYN_REPORT ------------
Event: time 1561579909.962069, type 3 (EV_ABS), code 16 (ABS_HAT0X), value 0
Event: time 1561579909.962069, -------------- SYN_REPORT ------------
Event: time 1561579910.292078, type 3 (EV_ABS), code 17 (ABS_HAT0Y), value -1
Event: time 1561579910.292078, -------------- SYN_REPORT ------------
Event: time 1561579910.562068, type 3 (EV_ABS), code 17 (ABS_HAT0Y), value 0
Event: time 1561579910.562068, -------------- SYN_REPORT ------------

So there is a response on the screen for every button. It somehow isn't connected to mouse control or control in games.

atar-axis, 2019-06-26:
Hum, so the driver does what it is supposed to do. Doesn't sound like a driver problem. Which games have you tried?

DirkMeister79, 2019-06-28:
Only Minecraft. I figured this would definitely work since it is in the Stretch image. Mouse support is actually the most important for me. 

Are there other games I should try for testing?

atar-axis, 2019-06-28:
Not every game has gamepad support, I fear Minecraft is one of those. You could at least give an "html5 gamepad" website a try - just Google for it and report back.

I think what we should then focus on is the vmouse thingy.

atar-axis, 2019-06-28:
E.g. https://gamepadviewer.com/

DirkMeister79, 2019-07-03:
Sorry for the late response. I was out of town for a few days.

gamepadviewer.com does work. It has a few buttons mixed up (shoulder buttons, triggers, y-button). But it does see the controller so that's great and most input is correct. What games work well with xpadneo? I'll try them too.

atar-axis, 2019-07-03:
That's good, therefore the problem is not xpadneo ;) The wrong mapping is a bug in the browser application, I already made a workaround here: #113

Since the problem is not xpadneo, I will close this issue. If you need any other help, feel free to write here.

Furthermore I am not a gamer, therefore I have absolutely no clue - I heard a lot of e.g. Steam ;)

closed 2019-07-03 by atar-axis