atar-axis/xpadneo #118: stops functioning after reboot

pookymaster, 2019-06-29:
**Describe the bug**
On Manjaro KDE kernel 4.19.49-1. After installation, controller connects and functions as expected. After reboot, the controller behaves the same before xpadneo install where it will cycle the connection, steam does not recognize controller connection. I did install the pre-req for Arch.

**To Reproduce**
Steps to reproduce the behavior:
uninstalled xpadneo, deleted folder downloaded, installed from step one described in code page. Verify proper bluetooth connection, test in steam, and is successful. Reboot, and attempt to reconnect which is unsuccessful.

**Expected behavior**
Stable, consistent connection, even after rebooting computer.

**Screenshots/Gifs**
If applicable, add screenshots or animated gifs to help explain your problem.

**System information**
Please enable debugging output <https://atar-axis.github.io/xpadneo/index.html#debugging>
and add at least the following outputs
-   `uname -a`
-   `dmesg`
-   `sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;`

I am new to linux, and I'm unsure about how to fulfill this step. 

**Additional context**  
Add any other context about the problem here.

atar-axis, 2019-06-29:
Sounds like one of those reconnection problems, the problem is most probably your BT chipset. Do you use an external dongle?

pookymaster, 2019-06-29:
Thanks for the prompt response! The BT chipset is not an external dongle. The drivers worked perfectly while I was running Pop!_os on the same laptop. It wasn't until after re-osing to Manjaro that I ran into this issue.

atar-axis, 2019-06-29:
Could be the chipset (more precisely the driver for it) anyway! Do you have an external dongle around? If yes, give it a try pls.

BTW, can you reconnect it by hand?

pookymaster, 2019-06-29:
Okay, my USB adapter doesn't work either, but nothing can connect to it. It's one of those super cheap no-name adapters so I don't know how well it's supported on Manjaro. If by reconnecting by hand, you mean to manually connect by adding a new device, then I have tried that, and it does not work. It behaves like it does without xpadneo, in that it cycles between connected, and not connected, no vibration when it connects.

atar-axis, 2019-07-03:
> Okay, my USB adapter doesn't work either, but nothing can connect to it.

Then please try another one ;)

> in that it cycles between connected, and not connected,

That happens only if ERTM is enabled *or* the dongle does not work corretly... in my experience it is the latter, therefore please try another dongle (lend one if necessary).

Furthermore connection problems are not really xpadneo issues since the connection is done by the linux kernel, not xpadneo. I will close this issue therefore but we can discuss here as long it lasts ;)

closed 2019-07-03 by atar-axis