atar-axis/xpadneo #119: Unable to install Prerequisites Manjaro

RoyvanEmpel, 2019-06-30:
**Describe the bug**
While trying to install the Prerequisites for Manjaro (Arch). It tells me that its unable to install a module. When this happend I still tried installing it and that of course doesn't work.

**To Reproduce**
Somewhat clean Manjaro install and the linux 5.1.8-1 kernel running

**Expected behavior**
Basically that it would install without any problems like it did on pop os when i tried it, but not everything goes to plan :)

**Screenshots/Gifs**
Unable to install module:
[image]

Trying to run the install.sh:
[image]

**System information**
Manjaro running on linux 5.1.8-1 kernel with a intel i5 6600k, nvidia gtx 1060 6gb videocard and 24 gb of ram.

**Additional context**  
I am still new to linux and just starting out with it so don't go to hard on me please. I also have dyslexia and English is not my main language sorry for any spelling mistakes.

atar-axis, 2019-07-03:
Looks like the wrong kernel headers are installed...
Run `sudo pacman -S linux-headers` and give me the result please

atar-axis, 2019-07-03:
I will close this issue since I don't think that this is a bug with xpadneo, but we can continue discussing it here. Cheers

closed 2019-07-03 by atar-axis

RoyvanEmpel, 2019-07-04:
I just read it since i forgot to read the email which my github is linked to and i tried running  sudo pacman -S linux-headers but it gives me the same result so:

[image]

RoyvanEmpel, 2019-07-05:
Hello @atar-axis 
I have been searching on google but i don't know how to get it to work. I hope you are willing to help me since i really wanna get my controller working. :)

atar-axis, 2019-07-06:
You are on `linux 5.1.8-1` but you are installing the kernel headers for 3.16 - that's the problem I think. You should get in touch with someone from the Manjaro community to get some help installing the right headers.

RoyvanEmpel, 2019-07-07:
Well i looked around and tried just installing some of the latest headers and it automatically chose the right one for me so i tried installing xpadneo again and now its working. Thank you for the suggestion. :)

atar-axis, 2019-07-07:
perfect! do you remember the right command ? in order to update the README

RoyvanEmpel, 2019-07-07:
This should be the whole command for manjaro then :):

sudo pacman -S dkms linux-latest-headers bluez bluez-utils