atar-axis/xpadneo #121: Controller freezes after long vibration

thankjura, 2019-07-17:
**Describe the bug**
After long time vibration in game, controller freezes and always send last pressed keys
Then I disconnect and connect gamepad and after next long vibration bug repeat

**To Reproduce**
Connect xbox one gamepad over bluetooth.
Run run game, ex. Dirt Rally or Burnout Paradise(proton)
Wait for long time vibration(crash car for example)

**System information**
-   Linux UltraHost 5.2.1-gentoo #2 SMP PREEMPT Mon Jul 15 01:39:26 MSK 2019 x86_64 Intel(R) Core(TM) i7-6700K CPU @ 4.00GHz GenuineIntel GNU/Linux
-  [dmesg_xpadneo.log](https://github.com/atar-axis/xpadneo/files/3399585/dmesg_xpadneo.log)
-  /sys/kernel/debug/hid/0005:045E:02FD.0014
05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 32 09 35 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 02 09 c5 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 02 09 c4 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0f 15 00 25 01 75 01 95 0f 81 02 15 00 25 00 75 01 95 01 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75 08 95 01 81 02 c0 00 
2579218694 1006

**Additional context**  
Controller work fine if I connect gamepad over wire

ugly95, 2019-07-17:
Are you using a USB Bluetooth dongle?  This sort of sounds similar to what I'd get when I used a USB Bluetooth dongle that has a Broadcom chip.

thankjura, 2019-07-17:
Yes.
Bus 001 Device 004: ID 0b05:17cb ASUSTek Computer, Inc. Broadcom BCM20702A0 Bluetooth

thankjura, 2019-07-17:
Is it fixable?

atar-axis, 2019-07-17:
Please have a look at the search function, we had this issue already a couple of times and it has nothing to do with xpadneo ;) It is fixable in a dirty way, but I would simply suggest to buy another dongle.

closed 2019-07-17 by atar-axis