atar-axis/xpadneo #122: Controller Input Stops when rumbling occurs randomly

robzombie91, 2019-07-21:
**Describe the bug**
Sometimes when in a game, if the game starts a rumble action any other input will stop and the controller will continue to rumble until either it is disconnected by holding the guide button or the batteries are pulled out

**To Reproduce**
This error is random, sometimes it will stop working while rumbling continuously, sometimes it will continue to work regardless

**Expected behavior**
Controller works regardless of rumble or not

**System information**

Linux rob 5.2.1-arch1-1-ARCH #1 SMP PREEMPT Sun Jul 14 14:52:52 UTC 2019 x86_64 GNU/Linux

`[ 5782.215747] xpadneo 0005:045E:02E0.000D: input,hidraw9: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 00:1a:7d:da:71:13
[ 5782.217324] xpadneo udev: 0005:045E:02E0.000D
[ 5782.218659] xpadneo udev: ok
[ 5825.765459] xpadneo: raw_event: 04 86
[ 5825.765477] xpadneo 0005:045E:02E0.000D: hid-up: 60000, hid-usg: 20, input-code: 00, value: 86
[ 5825.765479] xpadneo 0005:045E:02E0.000D: REPORT HOOK
[ 5845.785457] xpadneo: raw_event: 04 86
[ 5845.785475] xpadneo 0005:045E:02E0.000D: hid-up: 60000, hid-usg: 20, input-code: 00, value: 86
[ 5845.785476] xpadneo 0005:045E:02E0.000D: REPORT HOOK
[ 5865.805126] xpadneo: raw_event: 04 86
[ 5865.805144] xpadneo 0005:045E:02E0.000D: hid-up: 60000, hid-usg: 20, input-code: 00, value: 86
[ 5865.805145] xpadneo 0005:045E:02E0.000D: REPORT HOOK
[ 5885.825155] xpadneo: raw_event: 04 86
[ 5885.825161] xpadneo 0005:045E:02E0.000D: hid-up: 60000, hid-usg: 20, input-code: 00, value: 86
[ 5885.825162] xpadneo 0005:045E:02E0.000D: REPORT HOOK
[ 5905.848532] xpadneo: raw_event: 04 86
[ 5905.848551] xpadneo 0005:045E:02E0.000D: hid-up: 60000, hid-usg: 20, input-code: 00, value: 86
[ 5905.848552] xpadneo 0005:045E:02E0.000D: REPORT HOOK`

`05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0a 15 00 25 01 75 01 95 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 80 85 02 a1 00 09 85 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75 08 95 01 81 02 c0 00 
437417229 922
`

robzombie91, 2019-07-21:
Will update with pertinent information if this issue happens to me

ugly95, 2019-07-21:
This sounds like the Bluetooth dongle issue again.  You are most likely using a bluetooth controller with a Broadcom chipset.

See [this issue](https://github.com/atar-axis/xpadneo/issues/121) and [this issue](https://github.com/atar-axis/xpadneo/issues/76)

And this list of [Bluetooth dongles](https://atar-axis.github.io/xpadneo/#bt-dongles)

robzombie91, 2019-07-21:
this is the bt dongle that im using

https://www.amazon.com/Wavlink-Wireless-Bluetooth-Dual-Mode-BTMpblie/dp/B01H8NXDXC

robzombie91, 2019-07-21:
Just looked it up, Its a qualcom chipset

robzombie91, 2019-07-21:
CSR 8510 A10

ugly95, 2019-07-21:
Interesting that you're having the issue with a CSR chip.  The CSR-based chips have been solid for me (although problematic to initially connect).  The Broadcom chips have the rumble issue for me.

Either way, this is likely a Bluetooth issue and not an issue with the xpadneo driver.  The only real solution is to try a different dongle.  And, honestly, I would have said try a CSR chip, so I can't recommend anything.

robzombie91, 2019-07-21:
is this due to force feedback in anyway? and is there a way to disable it? I can live without rumble if it causes this issue

thankjura, 2019-07-21:
Maybe add option to set max rumble force?
I don't have problem if not reach maximum rumble force in game

Zeioth, 2019-07-22:
I can confirm this issue. I had it on "Tree of savior" and "Devil may cry 5". The workaround by now is disabling "vibration" inside of the game.

atar-axis, 2019-07-22:
Hey there, the `./configure.sh` script does offer the `disable_ff` option if you really think it is related to the vibration. Especially this comment is interesting:

> I don't have problem if not reach maximum rumble force in game

It would be great if you could enable the debug output (using the `./configure.sh` script once again) and send me the output (in `dmesg`) when it occurs - maybe the driver is sending something that the gamepad cannot handle?

thankjura, 2019-07-22:
https://github.com/atar-axis/xpadneo/issues/121
see dmesg_xpadneo.log

>>Is the driver still sending commands afterwards (while the gamepad does not respond anymore)?
Yes. For ex. if button "A" has been pressed before freeze, after freeze this button wiil be pressed until controller will be turn off

P.S. sorry for bad english

atar-axis, 2019-07-22:
Can you explain that a little bit more? It's hard to understand what is happening by just reading the log, where did it hung up (at which line)?

An how exactly does it behave? What do you mean by "if button A has been pressed before freeze, after freeze..." I thought there is nothing *after* since it you have to turn it off

atar-axis, 2019-07-22:
Futhermore the Log is incomplete, i am missing the rumble  message

thankjura, 2019-07-22:
In game(DIRT Rally) I press "RT" for gas. After achieve maximum vibration power, controller no response to press any button. I released all button, but car still going and controller vibrates.
I create new log from journalctl by `journalctl -f -n 0 > /tmp/journalctl`

[journalctl.zip](https://github.com/atar-axis/xpadneo/files/3417748/journalctl.zip)

atar-axis, 2019-07-22:
Have you manually removed the gamepad at those last couple of lines?

thankjura, 2019-07-22:
By long press to "xbox" button

atar-axis, 2019-07-22:
The question is if you have manually removed the gamepad or not?

> июл 22 16:53:32 UltraHost kernel: xpadneo 0005:045E:02FD.0011: playing effect: strong: 0x9112, weak: 0x4042, direction: 0x00
> июл 22 16:53:32 UltraHost kernel: xpadneo 0005:045E:02FD.0011: active: 0x0f, max: 0x91, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00
> июл 22 16:53:32 UltraHost kernel: xpadneo 0005:045E:02FD.0011: playing effect: strong: 0xa797, weak: 0x48cc, direction: 0x00
> июл 22 16:53:32 UltraHost kernel: xpadneo 0005:045E:02FD.0011: active: 0x0f, max: 0xa7, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00
> июл 22 16:53:32 UltraHost kernel: xpadneo 0005:045E:02FD.0011: playing effect: strong: 0x00, weak: 0x00, direction: 0x00
> июл 22 16:53:32 UltraHost kernel: xpadneo 0005:045E:02FD.0011: active: 0x0f, max: 0x00, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00

HERE

> июл 22 16:53:32 UltraHost steam.desktop[801]: Controller device closed after hid_read failure
> июл 22 16:53:32 UltraHost kernel: xpadneo 0005:045E:02FD.0011: Goodbye Xbox Wireless Controller!
> июл 22 16:53:32 UltraHost steam.desktop[801]: Controller 0 disconnected
> июл 22 16:53:32 UltraHost steam.desktop[801]: saving roaming config store to 'sharedconfig.vdf'
> июл 22 16:53:32 UltraHost steam.desktop[801]: roaming config store 2 saved successfully
> июл 22 16:53:34 UltraHost guake.desktop[1063]: Showing the terminal

there is a `hid_read` error and I would need to know if this happened because you removed the gamepad of if this error occured before you removed it

thankjura, 2019-07-22:
I try check it later

atar-axis, 2019-07-22:
Perfect, please install the `ff_debug` branch before trying it (remove the currently installed one).
Tell me if you don't know how to do that.

atar-axis, 2019-07-22:
**note to myself:**
maybe we will have to implement a service worker due to timing issues
- https://lwn.net/Articles/11360/
- https://elixir.bootlin.com/linux/latest/source/drivers/hid/hid-microsoft.c#L305
- https://elixir.bootlin.com/linux/latest/source/drivers/hid/hid-wiimote-core.c#L35

robzombie91, 2019-07-22:
I can reliably replicate this error in american truck simulator, ill post my dmesg in a bit..

Edit: 

First DMESG LOG: 

`[ 6660.543159] xpadneo 0005:045E:02E0.000E: probing device: Xbox Wireless Controller
[ 6660.543162] xpadneo: report-descr: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01
[ 6660.543162] xpadneo: report-descr: a1 00 09 33 09 34 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff 03 95 01 75
[ 6660.543163] xpadneo: report-descr: 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00
[ 6660.543163] xpadneo: report-descr: 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01
[ 6660.543164] xpadneo: report-descr: 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0a 15 00 25 01 75 01 95 0a 81 02 15 00 25 00
[ 6660.543164] xpadneo: report-descr: 75 06 95 01 81 03 05 01 09 80 85 02 a1 00 09 85 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07
[ 6660.543165] xpadneo: report-descr: 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95
[ 6660.543165] xpadneo: report-descr: 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91
[ 6660.543165] xpadneo: report-descr: 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0
[ 6660.543166] xpadneo: report-descr: 85 04 05 06 09 20 15 00 26 ff 00 75 08 95 01 81 02 c0 00
[ 6660.543245] xpadneo 0005:045E:02E0.000E: unknown main item tag 0x0
[ 6660.543250] xpadneo 0005:045E:02E0.000E: driver:
[ 6660.543251] xpadneo 0005:045E:02E0.000E: * version: 0.6.0
[ 6660.543251] xpadneo 0005:045E:02E0.000E: hdev:
[ 6660.543252] xpadneo 0005:045E:02E0.000E: * raw rdesc: (unfixed, see above)
[ 6660.543252] xpadneo 0005:045E:02E0.000E: * raw rsize: 307
[ 6660.543253] xpadneo 0005:045E:02E0.000E: * bus: 0x0005
[ 6660.543253] xpadneo 0005:045E:02E0.000E: * report group: 1
[ 6660.543254] xpadneo 0005:045E:02E0.000E: * vendor: 0x0000045E
[ 6660.543255] xpadneo 0005:045E:02E0.000E: * version: 0x00000903
[ 6660.543255] xpadneo 0005:045E:02E0.000E: * product: 0x000002E0
[ 6660.543256] xpadneo 0005:045E:02E0.000E: * country: 33
[ 6660.543256] xpadneo 0005:045E:02E0.000E: * driverdata: 0
[ 6660.543257] xpadneo 0005:045E:02E0.000E: * serial: 3a:33:38:3a:63:65
[ 6660.543258] xpadneo 0005:045E:02E0.000E: * physical location: 3a:61:31:3a:30:30
[ 6660.543260] xpadneo 0005:045E:02E0.000E: UP: 0x10000, USG: 0x0030 -> EV: 0x003, INP: 0x000
[ 6660.543261] xpadneo 0005:045E:02E0.000E: UP: 0x10000, USG: 0x0031 -> EV: 0x003, INP: 0x001
[ 6660.543262] xpadneo 0005:045E:02E0.000E: UP: 0x10000, USG: 0x0033 -> EV: 0x003, INP: 0x003
[ 6660.543263] xpadneo 0005:045E:02E0.000E: UP: 0x10000, USG: 0x0034 -> EV: 0x003, INP: 0x004
[ 6660.543264] xpadneo 0005:045E:02E0.000E: UP: 0x10000, USG: 0x0032 -> EV: 0x003, INP: 0x002
[ 6660.543264] xpadneo 0005:045E:02E0.000E: UP: 0x10000, USG: 0x0035 -> EV: 0x003, INP: 0x005
[ 6660.543265] xpadneo 0005:045E:02E0.000E: UP: 0x10000, USG: 0x0039 -> automatically
[ 6660.543266] xpadneo 0005:045E:02E0.000E: UP: 0x90000, USG: 0x0001 -> EV: 0x001, INP: 0x130
[ 6660.543267] xpadneo 0005:045E:02E0.000E: UP: 0x90000, USG: 0x0002 -> EV: 0x001, INP: 0x131
[ 6660.543267] xpadneo 0005:045E:02E0.000E: UP: 0x90000, USG: 0x0003 -> EV: 0x001, INP: 0x133
[ 6660.543268] xpadneo 0005:045E:02E0.000E: UP: 0x90000, USG: 0x0004 -> EV: 0x001, INP: 0x134
[ 6660.543269] xpadneo 0005:045E:02E0.000E: UP: 0x90000, USG: 0x0005 -> EV: 0x001, INP: 0x136
[ 6660.543269] xpadneo 0005:045E:02E0.000E: UP: 0x90000, USG: 0x0006 -> EV: 0x001, INP: 0x137
[ 6660.543270] xpadneo 0005:045E:02E0.000E: UP: 0x90000, USG: 0x0007 -> EV: 0x001, INP: 0x13A
[ 6660.543271] xpadneo 0005:045E:02E0.000E: UP: 0x90000, USG: 0x0008 -> EV: 0x001, INP: 0x13B
[ 6660.543271] xpadneo 0005:045E:02E0.000E: UP: 0x90000, USG: 0x0009 -> EV: 0x001, INP: 0x13D
[ 6660.543272] xpadneo 0005:045E:02E0.000E: UP: 0x90000, USG: 0x000A -> EV: 0x001, INP: 0x13E
[ 6660.543273] xpadneo 0005:045E:02E0.000E: UP: 0x10000, USG: 0x0085 -> EV: 0x001, INP: 0x13C
[ 6660.543274] xpadneo 0005:045E:02E0.000E: UP: 0x60000, USG: 0x0020 -> ignored
[ 6660.543275] xpadneo 0005:045E:02E0.000E: Fake device version: 0x1130
[ 6660.543300] input: Xbox Wireless Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/bluetooth/hci0/hci0:71/0005:045E:02E0.000E/input/input35
[ 6660.543395] xpadneo 0005:045E:02E0.000E: input,hidraw9: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 00:1a:7d:da:71:13
[ 6660.545503] xpadneo udev: 0005:045E:02E0.000E
[ 6660.546685] xpadneo udev: ok
`

AFTER DMSEG LOG: 
Soon

atar-axis, 2019-07-22:
> I can reliably replicate this error in american truck simulator, ill post my dmesg in a bit..

Sounds good! Please install the `ff_debug` branch first:
0) `./uninstall.sh`
1) `git fetch`
2) `git checkout ff_debug`
3) `./install.sh`

robzombie91, 2019-07-22:
Sorry, its going to be a little bit, The game is being a bit snobby and not starting after first launch, im going to try starting its linux variant

atar-axis, 2019-07-22:
no hurry

robzombie91, 2019-07-22:
After: (No goodbye xbox message after disconnecting with guide button)

`[ 2021.321577] xpadneo 0005:045E:02E0.000F: playing effect: strong: 0x1998, weak: 0x1998, direction: 0x00
[ 2021.321579] xpadneo 0005:045E:02E0.000F: active: 0x0f, max: 0x19, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00
[ 2021.321588] xpadneo 0005:045E:02E0.000F: hid_hw_output_report returned with 0
[ 2021.321593] xpadneo 0005:045E:02E0.000F: playing effect: strong: 0x206a, weak: 0x206a, direction: 0x00
[ 2021.321594] xpadneo 0005:045E:02E0.000F: active: 0x0f, max: 0x20, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00
[ 2021.321596] xpadneo 0005:045E:02E0.000F: hid_hw_output_report returned with 0
[ 2021.338506] xpadneo 0005:045E:02E0.000F: playing effect: strong: 0x1998, weak: 0x1998, direction: 0x00
[ 2021.338508] xpadneo 0005:045E:02E0.000F: active: 0x0f, max: 0x19, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00
[ 2021.338516] xpadneo 0005:045E:02E0.000F: hid_hw_output_report returned with 0
[ 2021.338520] xpadneo 0005:045E:02E0.000F: playing effect: strong: 0x20e6, weak: 0x20e6, direction: 0x00
[ 2021.338521] xpadneo 0005:045E:02E0.000F: active: 0x0f, max: 0x20, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00
[ 2021.338522] xpadneo 0005:045E:02E0.000F: hid_hw_output_report returned with 0
[ 2021.354688] xpadneo 0005:045E:02E0.000F: playing effect: strong: 0x1998, weak: 0x1998, direction: 0x00
[ 2021.354690] xpadneo 0005:045E:02E0.000F: active: 0x0f, max: 0x19, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00
[ 2021.354696] xpadneo 0005:045E:02E0.000F: hid_hw_output_report returned with 0
[ 2021.354701] xpadneo 0005:045E:02E0.000F: playing effect: strong: 0x2164, weak: 0x2164, direction: 0x00
[ 2021.354701] xpadneo 0005:045E:02E0.000F: active: 0x0f, max: 0x21, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00
[ 2021.354702] xpadneo 0005:045E:02E0.000F: hid_hw_output_report returned with 0
[ 2021.371596] xpadneo 0005:045E:02E0.000F: playing effect: strong: 0x1998, weak: 0x1998, direction: 0x00
[ 2021.371598] xpadneo 0005:045E:02E0.000F: active: 0x0f, max: 0x19, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00
[ 2021.371605] xpadneo 0005:045E:02E0.000F: hid_hw_output_report returned with 0
[ 2021.371609] xpadneo 0005:045E:02E0.000F: playing effect: strong: 0x21e0, weak: 0x21e0, direction: 0x00
[ 2021.371610] xpadneo 0005:045E:02E0.000F: active: 0x0f, max: 0x21, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00
[ 2021.371611] xpadneo 0005:045E:02E0.000F: hid_hw_output_report returned with 0
[ 2021.387808] xpadneo 0005:045E:02E0.000F: playing effect: strong: 0x1998, weak: 0x1998, direction: 0x00
[ 2021.387810] xpadneo 0005:045E:02E0.000F: active: 0x0f, max: 0x19, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00
[ 2021.387816] xpadneo 0005:045E:02E0.000F: hid_hw_output_report returned with 0
[ 2021.387822] xpadneo 0005:045E:02E0.000F: playing effect: strong: 0x2256, weak: 0x2256, direction: 0x00
[ 2021.387823] xpadneo 0005:045E:02E0.000F: active: 0x0f, max: 0x22, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00
[ 2021.387824] xpadneo 0005:045E:02E0.000F: hid_hw_output_report returned with 0
[ 2021.404801] xpadneo 0005:045E:02E0.000F: playing effect: strong: 0x1998, weak: 0x1998, direction: 0x00
[ 2021.404803] xpadneo 0005:045E:02E0.000F: active: 0x0f, max: 0x19, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00
[ 2021.404810] xpadneo 0005:045E:02E0.000F: hid_hw_output_report returned with 0
[ 2021.404813] xpadneo 0005:045E:02E0.000F: playing effect: strong: 0x22c4, weak: 0x22c4, direction: 0x00
[ 2021.404814] xpadneo 0005:045E:02E0.000F: active: 0x0f, max: 0x22, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00
[ 2021.404815] xpadneo 0005:045E:02E0.000F: hid_hw_output_report returned with 0
[ 2021.405011] xpadneo 0005:045E:02E0.000F: playing effect: strong: 0x22c4, weak: 0x22c4, direction: 0x00
[ 2021.405012] xpadneo 0005:045E:02E0.000F: active: 0x0f, max: 0x22, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00
[ 2021.405016] xpadneo 0005:045E:02E0.000F: hid_hw_output_report returned with 0
[ 2021.421626] xpadneo 0005:045E:02E0.000F: playing effect: strong: 0x1998, weak: 0x1998, direction: 0x00
[ 2021.421628] xpadneo 0005:045E:02E0.000F: active: 0x0f, max: 0x19, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00
[ 2021.421634] xpadneo 0005:045E:02E0.000F: hid_hw_output_report returned with 0
[ 2021.421638] xpadneo 0005:045E:02E0.000F: playing effect: strong: 0x232c, weak: 0x232c, direction: 0x00
[ 2021.421639] xpadneo 0005:045E:02E0.000F: active: 0x0f, max: 0x23, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00
[ 2021.421640] xpadneo 0005:045E:02E0.000F: hid_hw_output_report returned with 0
[ 2021.438840] xpadneo 0005:045E:02E0.000F: playing effect: strong: 0x1998, weak: 0x1998, direction: 0x00
[ 2021.438842] xpadneo 0005:045E:02E0.000F: active: 0x0f, max: 0x19, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00
[ 2021.438848] xpadneo 0005:045E:02E0.000F: hid_hw_output_report returned with 0
[ 2021.438853] xpadneo 0005:045E:02E0.000F: playing effect: strong: 0x2396, weak: 0x2396, direction: 0x00
[ 2021.438854] xpadneo 0005:045E:02E0.000F: active: 0x0f, max: 0x23, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00
[... cut ...]
[ 2022.898202] xpadneo 0005:045E:02E0.000F: active: 0x0f, max: 0x16, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00
[ 2022.898203] xpadneo 0005:045E:02E0.000F: hid_hw_output_report returned with 0
[ 2022.898204] xpadneo 0005:045E:02E0.000F: playing effect: strong: 0x00, weak: 0x00, direction: 0x00
[ 2022.898205] xpadneo 0005:045E:02E0.000F: active: 0x0f, max: 0x00, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00
[ 2022.898206] xpadneo 0005:045E:02E0.000F: hid_hw_output_report returned with 0
`

atar-axis, 2019-07-22:
After means "after the gamepad hung up", right? so the last messages before the gamepad freezed was

> [ 2022.898203] xpadneo 0005:045E:02E0.000F: hid_hw_output_report returned with 0
> [ 2022.898204] xpadneo 0005:045E:02E0.000F: playing effect: strong: 0x00, weak: 0x00, direction: 0x00
> [ 2022.898205] [...]

robzombie91, 2019-07-22:
Journalctl dmesg pastebin

https://pastebin.com/A9ph5RZa

robzombie91, 2019-07-22:
yes

robzombie91, 2019-07-22:
the game keeps sending the same inputs after the freeze, so fe am truck sim the steering wheel keeps turning the same direction as you last pressed and the controller keeps vibrating until you turn off the controller

atar-axis, 2019-07-22:
It isn't the game which is sending that input, neither is it the driver - it is the gamepad which simply does not get a "stop" message. The question is why, unfortunately there is no obvious problem at the log.

Anyway, I guess that it is some sort of timing issue, therefore I will implement a force-feedback worker. stay tuned

robzombie91, 2019-07-22:
Dont worry on it being a fast fix, ATM i just have ff disabled, it seems to at least alleviate the issue with games that wont let you disable rumble, or having force feedback enabled in-game but ff disabled via your driver. It works regardless.

atar-axis, 2019-07-22:
@robzombie91 how exactly do you replicate this issue? has it to do with the strength of the ff-effect as @thankjura said?

robzombie91, 2019-07-22:
@atar-axis the Game "ATS" constantly rumbles when force feedback is enabled (im guessing that is a default feature of the game), and when you start moving the force feedback fluctuates depending on your speed so im guessing maybe it has to do with the strength (though i highly think it has to due with the length/timing of the rumble/force feedback)

thankjura, 2019-07-22:
> июл 22 22:03:13 UltraHost kernel: xpadneo 0005:045E:02FD.0014: active: 0x0f, max: 0x01, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00
июл 22 22:03:13 UltraHost kernel: xpadneo 0005:045E:02FD.0014: hid_hw_output_report returned with 0
июл 22 22:03:13 UltraHost kernel: xpadneo 0005:045E:02FD.0014: playing effect: strong: 0x00, weak: 0x46, direction: 0x00
июл 22 22:03:13 UltraHost kernel: xpadneo 0005:045E:02FD.0014: active: 0x0f, max: 0x00, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00
июл 22 22:03:13 UltraHost kernel: xpadneo 0005:045E:02FD.0014: hid_hw_output_report returned with 0
июл 22 22:03:13 UltraHost kernel: xpadneo 0005:045E:02FD.0014: playing effect: strong: 0x00, weak: 0x00, direction: 0x00
июл 22 22:03:13 UltraHost kernel: xpadneo 0005:045E:02FD.0014: active: 0x0f, max: 0x00, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00
июл 22 22:03:13 UltraHost kernel: xpadneo 0005:045E:02FD.0014: hid_hw_output_report returned with 0
июл 22 22:03:19 UltraHost steam.desktop[801]: Controller device closed after hid_read failure
июл 22 22:03:19 UltraHost kernel: xpadneo 0005:045E:02FD.0014: Goodbye Xbox Wireless Controller!
июл 22 22:03:19 UltraHost steam.desktop[801]: Controller 0 disconnected
июл 22 22:03:20 UltraHost steam.desktop[801]: saving roaming config store to 'sharedconfig.vdf'
июл 22 22:03:20 UltraHost steam.desktop[801]: roaming config store 2 saved successfully

>> июл 22 22:03:19 UltraHost steam.desktop[801]: Controller device closed after hid_read failure

this error happened after I turned off controller

P.S. I can't reproduce second time this bug. I played about 10 min, then I played to "dirt 4" and "burnout paradise", reached maximum rumble force, but controller is still work.
P.S. I reverted to master branch, and can't reproduce bug

michaudg, 2019-07-24:
Same problem for me in _Metal Gear Solid 5_ with a CSR dongle. Tell me if I can help.

atar-axis, 2019-07-24:
> Journalctl dmesg pastebin
> 
> https://pastebin.com/A9ph5RZa

Strange thing, the game is requesting the same effects all the time... search for `playing effect: strong: 0x1998, weak: 0x1998, direction: 0x00` and you will see what I mean.

ugly95, 2019-07-24:
I'm surprised to see people with the CSR chipset adapters having this issue.  I've played numerous games with rumble using my CSR-based adapter and have not had an issue.

Using a Broadcom-based adapter I did have this issue.  I remember Dirt Showdown causing this problem a lot.  From what I remember, there was nothing in particular that could reliably reproduce the problem.  But usually within 20 minutes of playing, the controller would lock up on a rumble.  Switching to the CSR adapter completely resolved the issue.

michaudg, 2019-07-25:
I will made more tests. I have both adapters (CSR and Broadcom) and I switched it often today. And I'm not sure if I disabled the Xbox controller option in steam before the problem occured in MGSV.

robzombie91, 2019-07-25:
@atar-axis 

1. Install xpadneo regularly with force feedback enabled
2. connect controller
3. install american truck simulator (wine or native "doesnt matter which")
4. start playing american truck simulator 
5. create a new game
6. set up controls (game does his naturally)
7. load into game (automatic)
8. start truck
9. start driving and turning
10. glitch appears until you turn off the controller or remove the batteries

michaudg, 2019-07-25:
I confirm the problem with a [Sabrent CSR dongle](https://www.sabrent.com/product/BT-UB40/usb-bluetooth-4-0-micro-adapter-pc-v4-0-class-2-low-energy-technology/#description). I had the problem with MGS5 and Helldivers with Steam Proton. All Steam Xbox controller options are disabled.

Here is my logs: https://pastebin.com/sacyVRT8

ugly95, 2019-07-25:
@michaudg I have the same Sabrent dongle (although it's a backup and I usually use another CSR dongle).  I don't have MGS5 or Helldivers.  Any other games where you can reproduce the issue so I can try and confirm?

michaudg, 2019-07-27:
After several tests today, I think that all my problems comes from Steam and not from xpadneo. xpadneo works wonderfully with Witcher 3-GOG (Lutris+Wine) but as soon as I launch games from steam (Lutris+Steam or Steam directly), I lost connection with the controller or I have the "always rumble" bug and this, with CSR and Broadcom dongle.

@Zeioth @robzombie91 Do you use Steam to play your games?

robzombie91, 2019-07-27:
@michaudg  some of my games are drm free so no its not just steam

michaudg, 2019-07-27:
@ugly95 I just had the same problem with Shadow Warrior 2-GOG. Can you tell me what game works correctly on your side so I can try to test it?

atar-axis, 2019-07-27:
@all Please use the `configure.sh` script to disable rumble/force feedback (ff). Test it for a while, does this solve the problem for anyone?

robzombie91, 2019-07-27:
@atar-axis i did this already when testing it on ATS and it functioned perfectly, just without force feedback. I even had force feedback enabled in the games options, when normally i would have to disable force feedback for it to work normally

robzombie91, 2019-07-27:
@ugly95 Dead Cells(gog version) had the same issue with force feedback and that uses libinput.so "not sure if thats the correct name for it" and xpadneo would stop working occasionally when an intensive rumble happened

ugly95, 2019-07-27:
@michaudg I do have Shadow Warrior 2 on GOG.  I don't have it installed, so give me some time to test it.

I'm currently playing Bloodstained: Ritual of the Night on Steam.  There's some light rumbling, but I'm not sure if it gets intense.  It's never had issues.

In DiRT Showdown I did get the rumbling issue you describe, but only with the Broadcom dongle.  The CSR dongle works perfectly in game.

As I mentioned, the only time I had the 'rumble lock' issue is when I used the Broadcom dongle.  When I used the dongle, I noticed it was generally a bit laggy.  And it had a short connection range.  So, my suspicion was that it had to do with the connection being poor and unreliable.  Have you tried moving the dongle to a better location (like using an extension cable to have the dongle on your desk instead of plugged in to the back of your case)?

michaudg, 2019-07-27:
I continue to do several tests. I just had this problem with FEZ-Native **without using xpadneo**. When I do small jumps, I can feel the rumble but if I jump from very high, the controller vibrate very strong and never stop. I have to power off the controller and pair it again. I don't know where to look anymore. I'm wondering if the problem could comes from the Linux kernel or modules. I'm with Arch 5.2.3, KDE.

robzombie91, 2019-07-27:
@ugly95 i dont think its a range issue since my bt dongle is in my keyboard via usb passthrough (corsair k70)

michaudg, 2019-07-27:
@ugly95 Can you share which Linux distro and kernel you are using please?

ugly95, 2019-07-27:
Kernel: 5.0.0-20-generic x86_64 bits: 64 
Desktop: Cinnamon 4.0.10 Distro: Linux Mint 19.1 Tessa

robzombie91, 2019-07-27:
My Kernel/De

Kernel: 5.2.3-arch
Desktop: KDE 5.60.0 Plasma 5.16.3

michaudg, 2019-07-28:
I downgraded to a 4.19 kernel and the IDENTIFY button from the Steam controller options do nothing. No rumble at all. As soon as I return to 5.2.3, the IDENTIFY button works and the controller vibrates.

@ugly95 Thanks for the info about your problem with the Broadcom dongle. My CSR dongle is just in front of me, at 12 inches, but I suspect a bit of lag. I lost some button input sometimes. I will look at it.

michaudg, 2019-07-28:
I just tried with a USB extension, the dongle was on my legs. Same problem. It surely a problem between the kernel and the Xbox One S controller... Something to do with [this feature](https://www.phoronix.com/scan.php?page=news_item&px=Linux-HID-4.20-Xbox-Logitech)?

michaudg, 2019-07-28:
I ordered [this adapter](http://www.8bitdo.com/wireless-usb-adapter/) and I will see if it's better.

atar-axis, 2019-07-28:
My personal guess is that the rumble routine is simply too long (too much calculations). You could try to simplify it (removing the trigger FF e.g), if that works then it is modt probably a timing issue.

atar-axis, 2019-07-29:
I removed all trigger FF parts to shorten the playback routine, give the latest `ff_debug` branch a try and report back please

robzombie91, 2019-07-29:
@atar-axis no build errors so far

robzombie91, 2019-07-29:
@atar-axis with about 5 minutes of testing (finishing my delivery with ff on in ats) no ff disconnects happened even with it completely enabled (engine makes it rumble alot at varying strengths)

[40 more replies]

closed 2020-05-08 by kakra