atar-axis/xpadneo #123: Gamepad does not actively connect to wifi ap created by esp8266 (2.4g)

sj15712795029, 2019-08-09:
I see your description at wiki:
Once a Dot11Beacon is found, the GP is comparing the MAC of the Dot11Beacons origin to '62:45:B4:xx:yy:zz'
And the Gamepad will scan 1, 6, 11 (2.4 GHz) beside the following channels 36, 40, 44, 48, 153, 157, 161 and 165 (5 GHz)
But I'm creating an AP, beacon as shown here:
[image]

I found that the handle was not connected, so I grabbed the bag and found that my handle would only connect to the usb adapter on channel40 and 44 (wifi chip is MT7612). Now I want to use esp8266, because this module only supports 2.4g.
Excuse me, is there something wrong with my beacon?

sj15712795029, 2019-08-09:
If anyone can answer this question, I appreciate it very much

sj15712795029, 2019-08-09:
I see the github https://github.com/nathanRamaNoodles/Tinker-Controller
He used esp8266 to connect the xbox, but he only used an esp-test ssid(open ap), and I didn't even see him modify the MAC, but why did his xbox wireless wifi gamepad connect to esp8266

sj15712795029, 2019-08-10:
I made a 5G AP again today and set the address to 62:45:b4:ff:05:13, but the gamepad still did not initiate pairing connection, beacon is as follows
[image]

atar-axis, 2019-08-29:
Sorry, I do not have time to develop into this direction at the moment, therefore I cannot answer your questions too. I am sorry.

closed 2019-08-29 by atar-axis

sj15712795029, 2019-08-30:
Thanks for your reply. I have made xbox connect to the AP created by esp8266, but esp8266 does not reply to the connection request, but only replies ack, but does not reply to the connection request. I have also temporarily pending, and I will share it here if there is any message