atar-axis/xpadneo #125: Gamepad accessory support: chatpad as keyboard input

PixelatedHD, 2019-08-14:
Thank you for this project!  Your driver is my go-to for any machine I play games on.  Xpadneo plus qjoypad plus bluetooth trackball mouse is ALMOST the perfect living room setup for Minecraft (and my new living room retropie setup as soon as the RPi4B is better supported).

It would be awesome if you could add some basic support for accessories, the chatpad seems like the easiest and most useful accessory to start with.  I vaguely recall seeing some chatpad related code in older xbox related drivers, but it has been a long time since I looked at it.

I'll pick up the tab on the hardware for development if you can put a chatpad on your Amazon wish list (and Amazon lets me make the purchase from the US).  Specifically the official Microsoft chatpad that has the green, orange, X1, and X2 modifier keys: "Xbox One Chatpad + Chat Headset (plugs directly into Chatpad)".  I already have one, so I can help beta test.

Bonus points if you can configure the mapping between the chatpad and the keyboard output via a .conf file or something similar (think macros, varied chatpad button layouts, etc.).  Or if there is a good 3rd party utility for keyboard macros that would work cleanly with however the chatpad input device is exposed...

[image]

atar-axis, 2019-08-14:
Hey there!
I am sorry to tell you that I am not able to support any accessory because the bluetooth interface does not support them. In order to connect those devices we would need to communicate via WiFi - unfortunately I wasn't yet successful to do that.

Sorry for the bad news, thank you for your offer anyway.

Cheers

closed 2019-08-14 by atar-axis

PixelatedHD, 2019-08-14:
Ack, that sucks.

I'm newb-ish on device drivers and related stuff.  I (probably incorrectly)
think of bluetooth as a generic data transport layer that doesn't really
care too much what's connected on either end, at least after
pairing/connecting the device to the associated driver.

So a bluetooth hybrid keyboard/mouse combo device is no problem, and if you
wanted to use your own driver for a specific manufacturer/product ID, also
no problem.  But a bluetooth joypad/anything combo device would be an
unknown device type category, or something like that?

Just so I understand, where is the limitation?  The linux (and/or Windows)
Bluetooth interface driver?  Some other part of the system?

atar-axis, 2019-08-14:
The communication over bluetooth is based on the HID protocol, in which a new device has to inform the other side "what it is able to understand" and "what it is going to send". Unfortunately microsoft just hasn't implement anything but the basic "button/axis/rumble/battery" reports.

So the limitation is the gamepads interface over bluetooth.

PixelatedHD, 2019-08-15:
Ohhhhh.  So the limitation is the gamepad firmware then.  The device
firmware supports the extra stuff plus accessories via the normal xbox
wireless plus usb, just not when using bluetooth for communication?

Thanks for the info!  If you get the non-bluetooth wireless working
reliably and want a new challenge, my offer on the chatpad still stands...
;)