atar-axis/xpadneo #126: Random input

BluePlastics, 2019-08-16:
**Describe the bug**
Random input during normal use.

**To Reproduce**
Running latest retropie with xpadneo

**Expected behavior**
No random input

**Screenshots/Gifs**
If applicable, add screenshots or animated gifs to help explain your problem.

**System information**
Please enable debugging output <https://atar-axis.github.io/xpadneo/index.html#debugging>
and add at least the following outputs
-   `uname -a`:
Linux retropie 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
-   `dmesg`

-   `sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;`:
/sys/kernel/debug/hid/0005:045E:02FD.0003
05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 32 09 35 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 02 09 c5 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 02 09 c4 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0f 15 00 25 01 75 01 95 0f 81 02 15 00 25 00 75 01 95 01 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75 08 95 01 81 02 c0 00
2579218694 1006

Log:
https://pastebin.com/anvULmgr

**Additional context**  
I've removed Xpad and Xboxdrv to see if those were conflicting so only xpadneo exists on this install right now.

atar-axis, 2019-08-16:
Random input? Thats a bit vague, don't you think so?
I cannot fix something you cannot describe properly 😁

BluePlastics, 2019-08-16:
I apologize for not being clear, but by random input I mean randomly there will be a button press that I don't trigger, for example in zelda I'll be pressing a and the start button will trigger bringing up the menu or I'll be moving and the select button will be pressed and bring up the save and quit menu, this is just an example of whats happening, does that clear up my issue?

atar-axis, 2019-08-16:
I suggest to do the following:
* connect your gamepad as the last device (after keyboard, mouse, ...)
* run `evtest /dev/input/event<HIGHEST NUMBER>`

You will see some output when you press a button or move an axis,
please see if those random inputs as you call them, do appear there too :)

BluePlastics, 2019-08-17:
Just got home and tested this atar, and they do not show up in the EVTEST, I was moving the left analog stick and suddenly in game the map opened (the X button or BTN_NORTH as I have it bound). Is there anywhere else to check this out to see whats going on?

atar-axis, 2019-08-29:
If they do not show up in evtest then the problem is not caused by the driver.
I am too busy atm to offer third-party support, I am sorry.

closed 2019-08-29 by atar-axis

BluePlastics, 2019-08-29:
Uhhhh so... Its not caused by the driver... Even though when I use other drivers this issue doesn't occur, is that what you're saying...?

BluePlastics, 2019-08-29:
So what you're saying is, that even though this issue doesn't occur with any other driver that supports this controller, this issue isn't caused by your driver...?

I mean I guess that makes sense, you're the pro dev here, sorry for wasting my time reporting a bug.

atar-axis, 2019-08-29:
That's in essence what I am saying yes, the device files are the only way into the user space. If the problem does not occur there, then it is caused by a higher layer.

atar-axis, 2019-08-29:
Well, very mature @BluePlastics... 😀