atar-axis/xpadneo #128: Controller does not connect.

DeAbuyin, 2019-08-17:
**Describe the bug**

Controller does not connect neither via bluetoothctl nor GUI settings. They sometimes require PIN code for XBO controller or just rejects it.

**Screenshots/Gifs**

[image]

[image]

**System information**
Please enable debugging output <https://atar-axis.github.io/xpadneo/index.html#debugging>
and add at least the following outputs
-   `uname -a` - Linux 5.0.0-26-generic #27-Ubuntu SMP Tue Aug 13 17:47:39 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

-   `dmesg`
[image]

atar-axis, 2019-08-29:
please update the firmware using a Windows system, update it even if it is the newest one - this will reset the internal state of the gamepad.

DeAbuyin, 2019-09-05:
unfortunately, nothing changed

atar-axis, 2019-10-04:
The connection itself is not part of this driver since this is handled by bluetooth and the kernel itself.
I will therefore close this issue, but feel free to ask questions here,

My advice would be: remove all devices from the bluez, reset the gamepad, reboot, try again

closed 2019-10-04 by atar-axis