atar-axis/xpadneo #134: disable_ff doesn't work?

ipkpjersi, 2019-09-22:
**Describe the bug**
When I change disable_ff to 0, or any number from 0 to 3, when I connect the controller it still vibrates. It didn't used to do this, I think this is a regression.

**To Reproduce**
Steps to reproduce the behavior:
Change disable_ff to 0, or any number from 0 to 3

**Expected behavior**
I expect it to connect without the startup vibration.,

**System information**
Please enable debugging output <https://atar-axis.github.io/xpadneo/index.html#debugging>
and add at least the following outputs
-   Linux mypc 4.15.0-64-generic #73-Ubuntu SMP Thu Sep 12 13:16:13 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
-  
```
[ 1023.314359] hid-generic 0005:045E:02FD.000A: unknown main item tag 0x0
[ 1023.314447] input: Xbox Wireless Controller as /devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.1/3-9.1:1.0/bluetooth/hci1/hci1:12/0005:045E:02FD.000A/input/input23
[ 1023.314558] hid-generic 0005:045E:02FD.000A: input,hidraw7: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 00:19:86:00:0d:f7
[ 1023.316330] xpadneo udev: 0005:045E:02FD.000A
[ 1023.424651] xpadneo 0005:045E:02FD.000A: unknown main item tag 0x0
[ 1023.424686] input: Xbox Wireless Controller as /devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.1/3-9.1:1.0/bluetooth/hci1/hci1:12/0005:045E:02FD.000A/input/input24
[ 1023.424772] xpadneo 0005:045E:02FD.000A: input,hidraw7: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 00:19:86:00:0d:f7
[ 1024.407120] xpadneo udev: ok
[ 1038.443355] hid-generic 0005:045E:02FD.000B: unknown main item tag 0x0
[ 1038.443487] input: Xbox Wireless Controller as /devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.1/3-9.1:1.0/bluetooth/hci1/hci1:12/0005:045E:02FD.000B/input/input25
[ 1038.443558] hid-generic 0005:045E:02FD.000B: input,hidraw7: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 00:19:86:00:0d:f7
[ 1038.445251] xpadneo udev: 0005:045E:02FD.000B
[ 1038.532583] xpadneo 0005:045E:02FD.000B: unknown main item tag 0x0
[ 1038.532617] input: Xbox Wireless Controller as /devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.1/3-9.1:1.0/bluetooth/hci1/hci1:12/0005:045E:02FD.000B/input/input26
[ 1038.532693] xpadneo 0005:045E:02FD.000B: input,hidraw7: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 00:19:86:00:0d:f7
[ 1039.515066] xpadneo udev: ok
[ 1046.327967] kauditd_printk_skb: 16 callbacks suppressed
```

atar-axis, 2019-10-03:
Sorry for the late reply: Have you solved the problem in the meanwhile?
How do you change the `disable_ff` parameter?

ipkpjersi, 2019-10-03:
Hi,

I will be able to test this weekend but I think I figured it out on one PC so it should be good on my second PC. Basically, I think changing the value in the configure script doesn't work whereas I actually created my own enable and disable scripts that set the value differently, and that seemed to do the trick on at least one PC. 

I will do some more testing this weekend and I will get back to you on that.

Thanks.

atar-axis, 2019-10-03:
Perfect, I will close this issue then. Feel free to reopen it if the problem persists.

ipkpjersi, 2019-10-06:
Hello @atar-axis 

So, I guess, bad news all around. 

I'm still experiencing the issue on one PC but not on the other.

I will only be on this PC for the next day, then I won't be near this PC again until next weekend (long weekend) and then after that, I won't have access to this PC for about 6 months, so unfortunately not a lot of time to troubleshoot this.

I also can't re-open this issue it seems.

Here's the scripts I tried: 

xpadneo_enable_ff.sh:
```bash
echo "options hid_xpadneo disable_ff=0" > /etc/modprobe.d/99-xpadneo-bluetooth.conf
echo "options hid_xpadneo disable_ff=0" > /etc/modprobe.d/xpadneo.conf
```
xpadneo_disable_ff.sh:
```bash
echo "options hid_xpadneo disable_ff=1" > /etc/modprobe.d/99-xpadneo-bluetooth.conf
echo "options hid_xpadneo disable_ff=1" > /etc/modprobe.d/xpadneo.conf
```

I tried both scripts, however, it always rumbles when powering on the controller and connecting to it.

ipkpjersi, 2019-10-06:
So this issue gets even weirder.

It appears my Xbox One controller stopped connecting after my Ubuntu 18.04 updated to kernel `4.15.0-65-generic x86_64` from `4.15.0-64-generic x86_64` so I uninstalled xpadneo, rebooted, installed xpadneo and watched it build for `4.15.0-65-generic x86_64` and now, my controller no longer vibrates when connecting with disable_ff=1 or disable_ff=0. It's literally the opposite issue now. My Xbox One controller at home on my home PC behaves perfectly normally but on this cottage PC this is some very weird behaviour.

I would appreciate any advice or thoughts/ideas about this @atar-axis as this is quite weird.

atar-axis, 2019-10-11:
> Here's the scripts I tried:

Have you tried the included `configure.sh` script too?

ipkpjersi, 2019-10-11:
Yeah, I tried the `configure.sh` script which is why I ended up making my own scripts in the first place. Luckily, I will have access to my computer again tonight (in roughly 10 hours) so hopefully we can really get some troubleshooting done this weekend and figure it out.

I even bought another Bluetooth adapter that's how badly I want to figure this out. It's a Bluetooth 4 class 1 adapter (previously I was using a Bluetooth 4 class 2 adapter). Hopefully we can figure this out.

atar-axis, 2019-10-11:
Okay, lets see... Can you please sum up again what exactly you tried and what exactly didn't work? I don't really get it from the above description.

What's the exact command you used on the configure.sh script to disable FF?

ipkpjersi, 2019-10-12:
Hi,

I'm pretty sure this issue is *mostly* invalid. I was having some strange permission issues with xpadneo conf files combined with the fact (and most important thing) I forgot that you need to rmmod and modprobe to reload the kernel module for the settings changes to take effect.

However, I did notice that configure.sh does still throw some weird warnings and it doesn't seem to toggle xpadneo, but this may have been caused by me messing with something: 
[image]

Here are my new scripts that I am using, I am very happy with how well they work as they are working perfectly for me, and this issue can probably be closed if you want:

xpadneo_disable_ff.sh
```bash
#!/bin/bash
sudo rmmod hid_xpadneo
sudo chmod 777 /etc/modprobe.d/99-xpadneo-bluetooth.conf
sudo chmod 777 /etc/modprobe.d/xpadneo.conf
sudo echo "options hid_xpadneo disable_ff=3" > /etc/modprobe.d/99-xpadneo-bluetooth.conf
sudo echo "options hid_xpadneo disable_ff=3" > /etc/modprobe.d/xpadneo.conf
sudo chmod 777 /etc/modprobe.d/99-xpadneo-bluetooth.conf
sudo chmod 777 /etc/modprobe.d/xpadneo.conf
sudo modprobe hid_xpadneo
```
xpadneo_enable_ff.sh
```bash
#!/bin/bash
sudo rmmod hid_xpadneo
sudo chmod 777 /etc/modprobe.d/99-xpadneo-bluetooth.conf
sudo chmod 777 /etc/modprobe.d/xpadneo.conf
sudo echo "options hid_xpadneo disable_ff=0" > /etc/modprobe.d/99-xpadneo-bluetooth.conf
sudo echo "options hid_xpadneo disable_ff=0" > /etc/modprobe.d/xpadneo.conf
sudo chmod 777 /etc/modprobe.d/99-xpadneo-bluetooth.conf
sudo chmod 777 /etc/modprobe.d/xpadneo.conf
sudo modprobe hid_xpadneo
```

atar-axis, 2019-10-12:
I think so too, thanks you for getting back to me :)

closed 2019-10-12 by atar-axis