atar-axis/xpadneo #135: vmouse support

tarokker, 2019-09-24:
Hi, thank you for this fantastic project which actually works flawlessly (the rumble too!)
I tried the vmouse branch, in order to try to control the mouse pointer on my raspberry deskop, using the desktop, but it looks like it doesn't work properly.

Here's my info:

uname -r
> 4.19.66-v7l+

dkms status

> hid-xpadneo, 0.7.0, 4.19.66-v7l+, armv7l: installed
> vmouse, 0.1, 4.19.66-v7l+, armv7l: installed
> xpad, 0.4: added

some dmesg logs:
> [   72.022277] hid-generic 0005:045E:02E0.0005: unknown main item tag 0x0
> [   72.022900] input: Xbox Wireless Controller as /devices/platform/soc/fe201000.serial/tty/ttyAMA0/hci0/hci0:12/0005:045E:02E0.0005/input/input7
> [   72.023427] hid-generic 0005:045E:02E0.0005: input,hidraw2: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on dc:a6:32:0d:52:fb
> [   72.028307] xpadneo udev: 0005:045E:02E0.0005
> [   72.036498] hid_xpadneo: loading out-of-tree module taints kernel.
> [   72.037108] xpadneo: hello there (kernel 041342)!

Do you think that 'unknown main item tag' log could have something to do with the problem?
I really would like to help. Can I test something or try some change in the code?

Thanks in advance for your support and your greate project!

Regards

atar-axis, 2019-10-03:
> 'unknown main item tag' 

no, not really :)

> I really would like to help. Can I test something or try some change in the code?

What exactly do you mean by "but it looks like it doesn't work properly." :grin:

tarokker, 2019-10-03:
Actually the mouse pointer doesn't move using any of my pad sticks.

I keep pressed the XBox button, in order to activate the driver (this is how I should activate the mouse support, as far I understood, but I tried also without pressing it) but then the mouse pointer doesn't move on the raspbian desktop. Very strange.

Am I missing something?

Consider that the pad work flawlessly through bluetooth connection.

atar-axis, 2019-10-03:
what is the output of `lsmod | grep mouse` after connecting the gamepad?

tarokker, 2019-10-03:
- after pressing the XBox button for 2 seconds and receiving the confirm vibration -

root@beddomini:~# lsmod | grep mouse
root@beddomini:~#

tarokker, 2019-10-03:
more info from dmesg, dunno if this could help:

```
[   42.309269] xpadneo udev: 0005:045E:02E0.0005
[   42.429641] hid_xpadneo: loading out-of-tree module taints kernel.
[   42.430290] xpadneo: hello there (kernel 041342)!
[   42.430392] xpadneo: vmouse support disabled!
[   47.484012] xpadneo 0005:045E:02E0.0005: unknown main item tag 0x0
[   47.484342] input: Xbox Wireless Controller as /devices/platform/soc/fe201000.serial/tty/ttyAMA0/hci0/hci0:11/0005:045E:02E0.0005/input/input9
[   47.484844] xpadneo 0005:045E:02E0.0005: input,hidraw2: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on dc:a6:32:0d:52:fb
[   48.476957] xpadneo udev: ok
[  899.411930] xpadneo 0005:045E:02E0.0006: unknown main item tag 0x0
[  899.412248] input: Xbox Wireless Controller as /devices/platform/soc/fe201000.serial/tty/ttyAMA0/hci0/hci0:11/0005:045E:02E0.0006/input/input10
[  899.412822] xpadneo 0005:045E:02E0.0006: input,hidraw2: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on dc:a6:32:0d:52:fb
[  899.423743] xpadneo udev: 0005:045E:02E0.0006
[  899.440475] xpadneo udev: ok
```

atar-axis, 2019-10-03:
what is the result of `modinfo vmouse` ?

tarokker, 2019-10-03:
root@beddomini:~# modinfo vmouse
filename:       /lib/modules/4.19.66-v7l+/kernel/drivers/hid/vmouse.ko
version:        0.2
license:        GPL
description:    Mouse Emulation, designed to use in other modules
author:         Florian Dollinger
srcversion:     07084BDC48802B5435A4F50
depends:
name:           vmouse
vermagic:       4.19.66-v7l+ SMP mod_unload modversions ARMv7 p2v8

atar-axis, 2019-10-03:
okay, so the module is at least installed but not loaded. I guess you have already tried to restart the system, right?

tarokker, 2019-10-03:
yes, I tried several times :(

atar-axis, 2019-10-03:
try this:
```
shutdown the gamepad
modprobe -r hid_xpadneo
modprobe vmouse
modprobe hid_xpadneo
turn the gamepad on again and connect it
```

does it work now?

tarokker, 2019-10-03:
yes, it worked! O_O
Does it mean it is related to the incorrect modprobe order? (sorry for newbie questions)

atar-axis, 2019-10-03:
nope, that means that the UDEV rule does not work correctly,... hum

atar-axis, 2019-10-04:
In which order do you have installed the two modules?
* vmouse
* xpadneo vmouse branch

The thing is that you need to install it in the order above, if you don't do so then the xpadneo module is not able to detect the vmouse branch and will not automatically load it (the dependdency is optional).

atar-axis, 2019-10-04:
I labeled this as a bug since the documentation is missing a part about the order

tarokker, 2019-10-04:
I'm not totally sure, but I think I installed vmouse first.
I also tried to uninstall / reinstall xpadneo vmouse branch (and vmouse was already installed).

atar-axis, 2019-10-04:
hum, that's strange.

what's the result of `modinfo hid_xpadneo` then?

atar-axis, 2020-01-15:
closed due to inactivity

closed 2020-01-15 by atar-axis