atar-axis/xpadneo #137: It's seems @medusalix have finished the dongle driver on for macOS.

spphinslove, 2019-10-01:
Do you have plan port it to linux?

atar-axis, 2019-10-03:
Absolutely (but I am a bit busy right now). I would also love to see @medusalix joining this project here.

medusalix, 2019-10-04:
Sure thing 😃. I think the first step would be to get the Linux developers to include the VID and PID of the Xbox dongle in their MT76 driver. I've just spent the last ~3 months working on the macOS version (including reverse engineering) and to be honest I'm a little bit exhausted right now 😄. Having a Linux driver for the WLAN chipset would definitely make things a little bit easier.

atar-axis, 2019-10-04:
At least the old one is already included :)
I would love If you would join this project here when you had some to breath :)

NoXPhasma, 2019-10-19:
Wasn't there already a Kernel driver added for the dongle this year? When I connect the dongle I get this in the Kernel log:
```
Okt 19 20:18:23 NoXP-PC kernel: usb 3-1.2: new high-speed USB device number 4 using xhci_hcd
Okt 19 20:18:23 NoXP-PC kernel: usb 3-1.2: New USB device found, idVendor=045e, idProduct=02fe, bcdDevice= 1.00
Okt 19 20:18:23 NoXP-PC kernel: usb 3-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Okt 19 20:18:23 NoXP-PC kernel: usb 3-1.2: Product: XBOX ACC
Okt 19 20:18:23 NoXP-PC kernel: usb 3-1.2: Manufacturer: Microsoft Inc.
Okt 19 20:18:23 NoXP-PC kernel: usb 3-1.2: SerialNumber: 101653
```

And `lsusb` also shows the device with
```
Bus 003 Device 005: ID 045e:02fe Microsoft Corp.
```

Tell me if I understand something wrong here.

medusalix, 2019-10-19:
@NoXPhasma The kernel driver only attaches to the old dongle (PID 02**e6**). The newer dongle is not included in the [list](https://github.com/torvalds/linux/blob/998d75510e373aab5644d777d3b058312d550159/drivers/net/wireless/mediatek/mt76/mt76x2/usb.c#L20) of hardware ids and fails [this](https://github.com/torvalds/linux/blob/998d75510e373aab5644d777d3b058312d550159/drivers/net/wireless/mediatek/mt76/mt76x2/mt76x2.h#L28) check because it's a MT76**32**. Feel free to submit a patch to the Linux developers (I don't know the exact procedure, that's why this issue is on hold).

NoXPhasma, 2019-10-19:
Thanks for clarify this. I don't have any experience with submitting kernel patches either.

inkubux, 2019-10-21:
So technically if I have the old dongle something is possible. Unfortunately I don't know C. But I can help debug

wgpierce, 2019-10-28:
FYI I posted [this](https://github.com/paroj/xpad/issues/14#issuecomment-546751607) on xpad GitHub for this issue.

ddragic, 2019-11-17:
> @NoXPhasma The kernel driver only attaches to the old dongle (PID 02**e6**). The newer dongle is not included in the [...]

I opened a pull request earlier this year to add the new ids in the openwrt/mt76 repo (https://github.com/openwrt/mt76/pull/272 ; hmm, i just noticed I wrote it's a mt7612u in the comment there, I should change it) since it looks like it's were the upstream development is going on, but maybe it should go to the kernel mailing list...

Oh, and thanks for all of your work on this @medusalix

javsanpar, 2019-11-20:
> but maybe it should go to the kernel mailing list...

It would be great if you send the patch to the kernel mailing list. I can help you :smile:

voider1, 2019-11-24:
Anything else we could do to make this work?

medusalix, 2019-11-24:
@voider1 I'm working on a user-mode driver (based on libusb) for the wireless dongle right now as I ran into some problems with the MT76 kernel driver (apart from the missing hardware id). It shares most of its code with the macOS driver, so I expect to have a demo ready in a few weeks.

zypangpang, 2019-12-02:
Very excited and thankful for your work! @medusalix I just bought this wireless controller and found it cannot work on Linux. It really sucks. Hope it' ll work soon.

EnriqueWood, 2019-12-03:
@medusalix I offer myself to help if there's anything I can do. I have a windows 10-Ubuntu 19.04 dual boot setup, a dongle to Xbox one controllers and two controllers

medusalix, 2019-12-08:
[image]

It's working pretty well so far, although I still have quite a few problems/bugs that I want to have fixed before releasing the full source code 😃.

Neverest, 2019-12-09:
Awesome seeing support with the dongle. I can also help test if you need more info. Have the old big dongle and one controller(revised version).

Hedronmx, 2019-12-20:
Will this support only the 2nd version of the dongle? Or will it support both models?

medusalix, 2019-12-21:
It's finally ready to be [released](https://github.com/medusalix/xow) 😄. Support for the older dongle will be added, but I need packet captures done under Windows and with my Linux driver to see why it's not working.

medusalix, 2020-01-12:
@atar-axis Could you close this issue please?

atar-axis, 2020-01-15:
> Could you please close this issue?
:+1: sure

closed 2020-01-15 by atar-axis