atar-axis/xpadneo #139: Controller doesnt work in steamplay Greedfall

Active000, 2019-10-06:
**Describe the bug**
The controller doesnt do anything when in the game Greedfall with steam play.

**To Reproduce**
pair, trust, connect to your controller the usual way.
Start steam with steamplay 4.11-6.
Start the game GreedFall.
Try to go through the menus.
Start a game. Try to move

**Expected behavior**
Navigation in the main menu wont work
The character wont move

**Screenshots/Gifs**

**System information**
Fedora 30
Desktop: KDE
CPU:  AMD FX 8350 Eight core processor
GPU: AMD RX 580
-   `uname -a`
Linux desktop-fn6p92b.home 5.2.17-200.fc30.x86_64 #1 SMP Mon Sep 23 13:42:32 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

-   `dmesg`
https://pastebin.com/Fuh01Fk4

-   `sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;`

/sys/kernel/debug/hid/0005:045E:02FD.0007
2579218694 1006

**Additional context**  
no

atar-axis, 2019-10-11:
did the controller rumble while pairing?

sinkillerj, 2019-10-22:
I suspect the game in particular may be a false positive, more the fact that you are on Fedora 30 which caused issues for me and many others here. The dirty fix from here resolved my issues: https://github.com/atar-axis/xpadneo/issues/107#issuecomment-498858788

Active000, 2019-10-26:
I cant test that any longer. I dont have Fedora 30 any more. Im using ubuntu lts 18.04 and there I have  no problem in the same game. Although It can be tricky to connect it to bluetooth

atar-axis, 2020-01-15:
closed due to inactivity, feel free to reopen if necessary

closed 2020-01-15 by atar-axis