atar-axis/xpadneo #140: 8bitdo SN30 Pro+ works with xpadneo but Steam doesn't recognize it

pedrofleck, 2019-10-12:
**Describe the bug**
(I removed expected behaviour and steps to reproduce because it isn't needed)
I bought a SN30 Pro + and it was working fine, however, rumble wasn't working wireless or wired and after trying some tweaks, xboxdrv, among other things I ended up know about xpadneo and since I tried everything I thought it was an alternative since the controller is recognized as an Xbox One pad when using X-input mode. 
I installed xpadneo and turned the controller on (it was already synced) and it started vibrating without stoping, so I configured its force-feedback to '2' (disabling the triggers rumble) and it worked! I thought the problem was finally solved, I tested it with Wine "joy.cpl", Dolphin and "fftest" and rumble and the buttons/triggers were recognized. Then I opened Steam to play some games and found out the controller wasn't recognized. What can I do? I know it isn't the focus of this driver but it's the only one that worked for me. Thanks!

**System information**
Please enable debugging output <https://atar-axis.github.io/xpadneo/index.html#debugging>
and add at least the following outputs
-   `uname -a`: `Linux Fleck-LinuxMint 5.0.0-31-generic #33~18.04.1-Ubuntu SMP Tue Oct 1 10:20:39 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`
-   `dmesg`
`[out12 05:29] Linux version 5.0.0-31-generic (buildd@lgw01-amd64-046) (gcc version 7.4.0 (Ubuntu 7.4.0-1ubuntu1~18.04.1)) #33~18.04.1-Ubuntu SMP Tue Oct 1 10:20:39 UTC 2019 (Ubuntu 5.0.0-31.33~18.04.1-generic 5.0.21)
[  +0,000000] Command line: BOOT_IMAGE=/vmlinuz-5.0.0-31-generic root=UUID=c271d251-7a3d-4d49-97b2-4e05f000155e ro quiet splash nvidia-drm.modeset=1 nmi_watchdog=0 vt.handoff=1
[  +0,000000] KERNEL supported cpus:
[  +0,000000]   Intel GenuineIntel
[  +0,000000]   AMD AuthenticAMD
[  +0,000000]   Hygon HygonGenuine
[  +0,000000]   Centaur CentaurHauls
[  +0,000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[  +0,000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[  +0,000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[  +0,000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[  +0,000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[  +0,000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[  +0,000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[  +0,000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[  +0,000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[  +0,000000] BIOS-provided physical RAM map:
[  +0,000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009efff] usable
[  +0,000000] BIOS-e820: [mem 0x000000000009f000-0x00000000000fffff] reserved
[  +0,000000] BIOS-e820: [mem 0x0000000000100000-0x000000003fffffff] usable
[  +0,000000] BIOS-e820: [mem 0x0000000040000000-0x00000000403fffff] reserved
[  +0,000000] BIOS-e820: [mem 0x0000000040400000-0x0000000087a2bfff] usable
[  +0,000000] BIOS-e820: [mem 0x0000000087a2c000-0x0000000087a2cfff] ACPI NVS
[  +0,000000] BIOS-e820: [mem 0x0000000087a2d000-0x0000000087a2dfff] reserved
[  +0,000000] BIOS-e820: [mem 0x0000000087a2e000-0x000000008e10efff] usable
[  +0,000000] BIOS-e820: [mem 0x000000008e10f000-0x000000008e58efff] reserved
[  +0,000000] BIOS-e820: [mem 0x000000008e58f000-0x000000008ea6dfff] usable
[  +0,000000] BIOS-e820: [mem 0x000000008ea6e000-0x000000008eb1efff] ACPI NVS
[  +0,000000] BIOS-e820: [mem 0x000000008eb1f000-0x000000008f61efff] reserved
[  +0,000000] BIOS-e820: [mem 0x000000008f61f000-0x000000008f6fefff] type 20
[  +0,000000] BIOS-e820: [mem 0x000000008f6ff000-0x000000008f6fffff] usable
[  +0,000000] BIOS-e820: [mem 0x000000008f700000-0x000000008fffffff] reserved
[  +0,000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[  +0,000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[  +0,000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[  +0,000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[  +0,000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[  +0,000000] BIOS-e820: [mem 0x0000000100000000-0x000000046dffffff] usable
[  +0,000000] NX (Execute Disable) protection: active
[  +0,000000] efi: EFI v2.70 by American Megatrends
[  +0,000000] efi:  ACPI 2.0=0x8ea80000  ACPI=0x8ea80000  SMBIOS=0x8f418000  MEMATTR=0x892e9018  ESRT=0x8a64fd98 
[  +0,000000] secureboot: Secure boot could not be determined (mode 0)
[  +0,000000] SMBIOS 2.8 present.
[  +0,000000] DMI: Micro-Star International Co., Ltd. MS-7B19/B360M GAMING PLUS (MS-7B19), BIOS 1.70 08/22/2019
[  +0,000000] tsc: Detected 2900.000 MHz processor
[  +0,000000] tsc: Detected 2904.000 MHz TSC
[  +0,000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[  +0,000002] e820: remove [mem 0x000a0000-0x000fffff] usable
[  +0,000006] last_pfn = 0x46e000 max_arch_pfn = 0x400000000
[  +0,000007] MTRR default type: write-back
[  +0,000000] MTRR fixed ranges enabled:
[  +0,000001]   00000-9FFFF write-back
[  +0,000001]   A0000-BFFFF uncachable
[  +0,000000]   C0000-FFFFF write-protect
[  +0,000001] MTRR variable ranges enabled:
[  +0,000001]   0 base 00C0000000 mask 7FC0000000 uncachable
[  +0,000001]   1 base 00A0000000 mask 7FE0000000 uncachable
[  +0,000001]   2 base 0090000000 mask 7FF0000000 uncachable
[  +0,000000]   3 disabled
[  +0,000001]   4 disabled
[... 994 lines cut ...]
[  +0,000071] xpadneo: raw_event: 01 00 80 00 80 00 80 00 80 00 00 00 00 00 00 00
[  +7,691748] xpadneo: raw_event: 01 00 80 00 80 ff 72 00 80 00 00 00 00 00 00 00
[  +0,000097] xpadneo: raw_event: 01 00 80 00 80 00 80 00 80 00 00 00 00 00 00 00
[  +7,159453] xpadneo: raw_event: 01 ff 7e 00 80 00 80 00 80 00 00 00 00 00 00 00
[  +0,000103] xpadneo: raw_event: 01 00 80 00 80 00 80 00 80 00 00 00 00 00 00 00
[  +7,696135] xpadneo: raw_event: 01 ff 7b 00 80 00 80 00 80 00 00 00 00 00 00 00
[  +0,000218] xpadneo: raw_event: 01 00 80 00 80 00 80 00 80 00 00 00 00 00 00 00
[  +7,691268] xpadneo: raw_event: 01 ff 78 00 80 00 80 00 80 00 00 00 00 00 00 00
[  +0,000143] xpadneo: raw_event: 01 00 80 00 80 00 80 00 80 00 00 00 00 00 00 00
[  +7,691664] xpadneo: raw_event: 01 ff 76 00 80 00 80 00 80 00 00 00 00 00 00 00
[  +0,000019] xpadneo: raw_event: 01 00 80 00 80 00 80 00 80 00 00 00 00 00 00 00
[out12 08:06] xpadneo: raw_event: 01 00 80 ff 7b 00 80 00 80 00 00 00 00 00 00 00
[  +0,000213] xpadneo: raw_event: 01 00 80 00 80 00 80 00 80 00 00 00 00 00 00 00
[  +7,703530] xpadneo: raw_event: 01 00 80 ff 78 00 80 00 80 00 00 00 00 00 00 00
[  +0,000091] xpadneo: raw_event: 01 00 80 00 80 00 80 00 80 00 00 00 00 00 00 00
[out12 08:07] xpadneo: raw_event: 01 00 80 00 80 00 80 00 80 00 00 00 00 00 00 00
[  +0,000147] xpadneo: raw_event: 01 00 80 00 80 00 80 00 80 00 00 00 00 00 00 00`

-   `sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;`
`/sys/kernel/debug/hid/0005:045E:02E0.000E
05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0a 15 00 25 01 75 01 95 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 80 85 02 a1 00 09 85 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75 08 95 01 81 02 c0 00 
437417229 922`

**Additional context**  
Dolphin mapping worked ok, I didn't have to change any button mapped before and rumble worked perfectly fine. However, when trying to play a Wine game outside of Steam, the button mapping were crazy with the camera moving all the way up and L1 mapped to start, Y mapped to L3, etc.

pedrofleck, 2019-10-13:
Oh god I **fixed** it, that was a long ride... Anyway, I had to set "fake_dev_version" to 1 (but I'm not sure if it was necessary at all) then follow these instructions in [Arch Wiki](https://wiki.archlinux.org/index.php/Gamepad#Joystick_not_working_in_FNA/SDL_based_games) to get a SDL config, added to .profile the following: `export SDL_GAMECONTROLLERCONFIG="050000005e040000e002000030110000,8BitDo SN30 Pro+,platform:Linux,a:b0,b:b1,x:b2,y:b3,back:b6,guide:b8,start:b7,leftstick:b9,rightstick:b10,leftshoulder:b4,rightshoulder:b5,dpup:h0.1,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,leftx:a0,lefty:a1,rightx:a3,righty:a4,lefttrigger:a2,righttrigger:a5,"` and voila! rumble in Steam, Wine and native games! Thanks again for the driver.

atar-axis, 2019-10-13:
Maybe Something we should add to the readme, will Check in later.

sinkillerj, 2019-10-22:
Under Fedora 30 I didin't have to do the SDL nonsense just:
A. The dirty fix outlined here, I've not done deep diving into why but it seams F30 users have to do this period: https://github.com/atar-axis/xpadneo/issues/107#issuecomment-498858788
B. Setting force to 2 as outlined above to disable the triggers. Perhaps xpadneo could detect the controller type and set this on its own in the future?

pedrofleck, 2019-10-22:
@sinkillerj do you have the SN30 Pro+ too? I changed distro to Pop OS 19.10 now and my method worked again, however I had to blacklist xpad module this time, weird. I will try this "dirty fix" some other time, but atm I'm happy with the controller support.

sinkillerj, 2019-10-22:
I do! I installed xpadneo once I realized that its wireless xinput acts as a one controller not a 360 controller and I'm quite happy to have working vibration.

Can't speak on Pop OS personally. Was more just providing input to help the developer regarding this controller.

rsvensson, 2019-11-20:
I had to add the `SDL_GAMECONTROLLERCONFIG` environment variable to get Wine to recognize my SN30 Pro+. Or rather to get Xinput to work through Wine, since it did seem to detect the js and event types just fine with rumble and everything. Probably because Wine uses SDL2 for Xinput support.

atar-axis, 2020-02-20:
Feel free to open a pull request if you think that this should be mentioned at the documentation. I will close this issue since it seems to be fixed for you

closed 2020-02-20 by atar-axis