atar-axis/xpadneo #142: Xbox Elite 2 Controller

Draic, 2019-11-08:
**Is your feature request related to a problem? Please describe.**
I got the new Xbox Elite Series 2 Controller. It connects via bluetooth fine, but xpadneo does not grab the device. 

**Describe the solution you'd like**
I assume it would work fine, if xpadneo would just grab the device like the normal XboxOne controller. 

Without xpadneo in can be used as a generic gamepad, but the mapping seems to be wrong (I believe this is the same as with other Xbox One controllers? not sure; never had one before)

Let me know if there are any device specific infos that I can provide for this to happen

atar-axis, 2019-11-08:
Connect it without xpadneo and Upload the dmesg please, i need the PID :)

Draic, 2019-11-08:
is this what you need?

`[Nov 8 17:10] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[  +0,000005] Bluetooth: HIDP socket layer initialized
[  +0,022990] hid-generic 0005:045E:0B05.0013: unknown main item tag 0x0
[  +0,000169] input: Xbox Elite Wireless Controller as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.5/1-1.5:1.0/bluetooth/hci0/hci0:11/0005:045E:0B05.0013/input/input35
[  +0,000183] input: Xbox Elite Wireless Controller Consumer Control as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.5/1-1.5:1.0/bluetooth/hci0/hci0:11/0005:045E:0B05.0013/input/input36
[  +0,000113] input: Xbox Elite Wireless Controller Keyboard as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.5/1-1.5:1.0/bluetooth/hci0/hci0:11/0005:045E:0B05.0013/input/input38
[  +0,000105] hid-generic 0005:045E:0B05.0013: input,hidraw13: BLUETOOTH HID v9.03 Gamepad [Xbox Elite Wireless Controller] on dc:85:de:67:07:ac
`

atar-axis, 2019-11-08:
So it is 0B05 :)

Hielyr, 2019-11-11:
> So it is 0B05 :) I will add it - stay tuned!

I'm getting mine in today, so I'm very much looking forward to its implementation. If you'd like, I'll grab my `dmesg` output as well just to verify the PID.

Hielyr, 2019-11-11:
Here's my dmesg output just to confirm.

```
[112813.188454] hid-generic 0005:045E:0B05.001A: unknown main item tag 0x0
[112813.188680] input: Xbox Elite Wireless Controller as /devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:0B05.001A/input/input51
[112813.188866] input: Xbox Elite Wireless Controller Consumer Control as /devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:0B05.001A/input/input52
[112813.188967] input: Xbox Elite Wireless Controller Keyboard as /devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:0B05.001A/input/input54
[112813.189071] hid-generic 0005:045E:0B05.001A: input,hidraw6: BLUETOOTH HID v9.03 Gamepad 

```

Draic, 2019-11-16:
seems like this is less trivial to implement than expected

atar-axis, 2019-11-16:
I expect it to be easy but i am currently busy, I will do it as soon as possible. Sorry

duganchen, 2019-11-24:
I bought one today.

It's 0xb05, both before and after the firmware update. After I made the obvious changes to the driver and the udev rule to support the device with that id, all the keys were detected but they were mapped all wrong:

Here's sample evtest output.

Pressing down the left thumbstick seems to be correct:

`Event: time 1574576736.898854, type 1 (EV_KEY), code 317 (BTN_THUMBL), value 0`

Pressing down the right thumbstick is correct:

`Event: time 1574576898.830978, type 1 (EV_KEY), code 318 (BTN_THUMBR), value 0`

Pressing the thumbstick left is detected as a change in the Y axis:

`Event: time 1574576934.661434, type 3 (EV_ABS), code 1 (ABS_Y), value -472`

As is pressing the left thumbstick right:

`Event: time 1574576992.832223, type 3 (EV_ABS), code 1 (ABS_Y), value 1104`

Pressing the left thumbstick up:

`Event: time 1574577041.102846, type 3 (EV_ABS), code 0 (ABS_X), value 1215`

Pulling the left thumbstick down (towards you):

`Event: time 1574577070.233218, type 3 (EV_ABS), code 0 (ABS_X), value -5448`

D-Pad Up:

`Event: time 1574577134.294085, type 3 (EV_ABS), code 17 (ABS_HAT0Y), value -1`

D-Pad Down:

`Event: time 1574577151.124270, type 3 (EV_ABS), code 17 (ABS_HAT0Y), value 1`

D-Pad Left:

`Event: time 1574577166.714541, type 3 (EV_ABS), code 16 (ABS_HAT0X), value -1`

D-Pad Right:

`Event: time 1574577189.974812, type 3 (EV_ABS), code 16 (ABS_HAT0X), value 1`

LB:

`Event: time 1574577218.738831, type 1 (EV_KEY), code 310 (BTN_TL), value 2`

LT:

`Event: time 1574577236.005413, type 3 (EV_ABS), code 10 (ABS_BRAKE), value 1023`

RB:

`Event: time 1574577257.868832, type 1 (EV_KEY), code 311 (BTN_TR), value 2`

RT:

`Event: time 1574577277.526015, type 3 (EV_ABS), code 9 (ABS_GAS), value 110`

Y:

`Event: time 1574577293.574831, type 1 (EV_KEY), code 308 (BTN_WEST), value 2`

X:

`Event: time 1574577314.678831, type 1 (EV_KEY), code 307 (BTN_NORTH), value 2`

A:

`Event: time 1574577342.810831, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 2`

B:

`Event: time 1574577358.692832, type 1 (EV_KEY), code 305 (BTN_EAST), value 2`

Right Stick North:

`Event: time 1574577429.748046, type 3 (EV_ABS), code 2 (ABS_Z), value 37482`

Right Stick West:

`Event: time 1574577460.188631, type 3 (EV_ABS), code 5 (ABS_RZ), value 20446`

Right Stick East:

`Event: time 1574577502.940329, type 3 (EV_ABS), code 5 (ABS_RZ), value 19813`

Right Stick South:

`Event: time 1574577529.079452, type 3 (EV_ABS), code 2 (ABS_Z), value 24092`

Back:

`Event: time 1574577574.273819, type 1 (EV_KEY), code 158 (KEY_BACK), value 2`

Start:

`Event: time 1574577599.424831, type 1 (EV_KEY), code 315 (BTN_START), value 2`

Mode:

Event: time 1574577887.132816, type 1 (EV_KEY), code 316 (BTN_MODE), value 2

SentinelMKX, 2019-12-03:
Umm, my Xbox Elite Series 2 has a different ID, with Bluetooth I can't even connect it so I can't get the ID that way, only with usb=a, I have the newest firmware.
[10489.843144] usb 1-7: New USB device found, idVendor=045e, idProduct=0b00, bcdDevice= 4.08

SentinelMKX, 2019-12-03:
What reason can it be that I have different idProduct?

duganchen, 2019-12-04:
```
[ 5045.701666] usb 3-2: new full-speed USB device number 7 using xhci_hcd
[ 5045.828836] usb 3-2: New USB device found, idVendor=045e, idProduct=0b00, bcdDevice= 4.08
[ 5045.828837] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5045.828838] usb 3-2: Product: Controller
[ 5045.828839] usb 3-2: Manufacturer: Microsoft
[ 5045.828839] usb 3-2: SerialNumber: 3032363330303032373735393430
[ 5045.847656] input: Generic X-Box pad as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/input/input34
[ 5045.847798] usbcore: registered new interface driver xpad
```

It's 0b05 over Bluetooth and 0b00 over USB.

SentinelMKX, 2019-12-04:
Oh, okay, thank you, weird that I can't connect to with it with bluettooth, well

duganchen, 2019-12-04:
Not all Bluetooth adapters worked with even the Xbox One S controller (there's a "Compatible BT Dongles" link in the README of this very project), and some of the ones that do require ERTM to be disabled first.

My own (current) setup is Fedora 31, with an Asus USB-BT-400 dongle, and a file named "bluetooth.conf" in /etc/modprobe.d containing the following:

`options bluetooth disable_ertm=Y`

I got it from here:

https://www.reddit.com/r/RetroPie/comments/bi5bm4/psa_new_method_for_disabling_ertm_fix_controller/?utm_source=share&utm_medium=web2x

Some other ways of disabling it, which may work better on other distros, are documented here:

https://wiki.archlinux.org/index.php/Gamepad

SentinelMKX, 2019-12-15:
When I install xpadneo I can connect to it, when I uninstall it I can't
but with the ertm disabled I can connect to it even without xpadneo so yeah, that's nice, thank you

Mrtenz, 2020-01-25:
Any updates on this? I can connect to my controller with xpadneo installed, but it doesn't rumble (as it does with the regular Xbox controller), and it doesn't seem to work properly.

kakra, 2020-05-05:
@Mrtenz Could you send your device descriptor?

You should be able to obtain it with:
```
xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/*/report_descriptor
```

when `xpadneo` is bound to the device already.

hibes, 2020-05-11:
Now that I have the new kernel offered in Ubuntu 20.04 and added the disable_etrm=1 kernel option again using kernelstub, the device was connecting reliably until this latest reboot.  Now it's connecting, but nothing except tools like `jstest-gtk`, `evtest`, etc., seem to be able to detect it.

While it was working, the button config worked fine in steam big picture mode, which was a huge bonus, but in non-steam games, it wasn't working with controller settings.  Non-steam games which I launch with steam (Add a non-steam game featuer) worked fine, presumable because the `SDL_*` settings it sets, but maybe there's more to it.  I haven't been able to figure out how to reproduce the functioning gameplay without going through steam big picture mode.

Additionally, rumbling seems to cause a crash in the driver or something and cause the controller to stop responding until I reconnect it or wait long enough for it to reconnect itself.

I don't seem to be getting level 3 debugging from the commands like I'd expect.

```
$ sudo bash -c "echo 3 > /sys/module/hid_xpadneo/parameters/debug_level"
```

Then after disconnecting and reconnecting, the output from dmesg is the same.

```
$ cat /sys/module/hid_xpadneo/parameters/debug_level 
3
$ dmesg | grep xpadneo
[  795.454993] xpadneo 0005:045E:0B05.0008: fixing up report size
[  795.454994] xpadneo 0005:045E:0B05.0008: fixing up Rx axis
[  795.454995] xpadneo 0005:045E:0B05.0008: fixing up Ry axis
[  795.454995] xpadneo 0005:045E:0B05.0008: fixing up Z axis
[  795.454996] xpadneo 0005:045E:0B05.0008: fixing up Rz axis
[  795.454996] xpadneo 0005:045E:0B05.0008: fixing up button mapping
[  795.455387] LED trigger xpadneo-98:7a:14:1f:c3:82-battery-charging-or-full failed to register (-17)
[  795.455388] LED trigger xpadneo-98:7a:14:1f:c3:82-battery-charging failed to register (-17)
[  795.455388] LED trigger xpadneo-98:7a:14:1f:c3:82-battery-full failed to register (-17)
[  795.455389] LED trigger xpadneo-98:7a:14:1f:c3:82-battery-charging-blink-full-solid failed to register (-17)
[  795.455417] xpadneo 0005:045E:0B05.0008: battery registered
[  795.457319] xpadneo udev: 0005:045E:0B05.0008
[  795.459089] xpadneo udev: ok
[  815.940239] xpadneo 0005:045E:0B05.0008: input,hidraw0: BLUETOOTH HID v9.03 Gamepad [Xbox Elite Wireless Controller] on 5c:80:b6:70:8b:5d
```

hibes, 2020-05-12:
Turns out copying one or more of Steam's SDL configs was a bad idea.  I suspect it's their `SDL_GAMECONTROLLER_IGNORE_DEVICES` environment variable.

The remaining issues are now the rumble problem and the button config issues outside of steam. I've seen other reports of the rumble issue here: https://github.com/atar-axis/xpadneo/pull/144#issuecomment-618688115

kakra, 2020-05-12:
I'm in the process of removing `debug_level` in favor of using the normal HID kernel log infrastructure. So this setting probably has hardly an effect currently. I may add some logging during the next releases which uses `debug_level` again thus it still exists.

The rumbling crash is an issue with the bluetooth dongle (probably some incompatibility between the controller and the dongle itself, so nothing we could do about it). There's not much if even anything we could do differently when sending rumble commands to the controller: The protocol is clearly defined by the controller, and we are only using standard kernel infrastructure for communicating with the controller. Buying or using a different dongle may help even when it is the same chipset, see #166 and #171. It mostly shows up in newer kernel versions so there's a chance that it is influenced by how the kernel handles the BT dongles which could explain why even the same chipset but different dongle maker sometimes fixes the problem. Please report rumble and connection problems in the appropriate issue, or create a separate issue if unsure. This will be a lot easier for us to track and fix problems. Thank you.

The Xbox Elite controller support is currently incomplete. Could you please find the vendor/device ID when connecting the controller via Bluetooth in Windows? This may help us bypass SDL fixups applied by Steam or other software. Having `jstest` working correctly is a pretty clear sign we'd need such a bypass. Also, please let me know the report descriptor as asked for in https://github.com/atar-axis/xpadneo/issues/142#issuecomment-624070392.

This issue should be about detecting the controller and mapping its buttons and axis correctly only. The rumble command is exactly the same (4 bits + 4 bytes motor strength + 2 bytes pulse width + 1 byte loop count), so I'd like to stay away from putting any rumble issues here.

Draic, 2020-05-12:
I see that rudimentary support has beed added. As I had too much trouble with the elite 2 overall, I returned my controller. Someone else needs to report if the changes now recognize the controller proper. sorry.

hibes, 2020-05-12:
> @Mrtenz Could you send your device descriptor?
> 
> You should be able to obtain it with:
> 
> ```
> xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/*/report_descriptor
> ```
> 
> when `xpadneo` is bound to the device already.

```
$ xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/*/report_descriptor
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0b 15 00 25 01 75 01 95 0b 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 05 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81  u........$...%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 01 09 01 a1 00 09 40 09  ...%.u............@.
000000c8: 41 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 01 a1  A..'......u.........
000000dc: 00 09 43 09 44 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05  ..C.D..'......u.....
000000f0: 01 09 42 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06  ..B..&....u.....%.u.
00000104: 95 01 81 03 05 01 09 45 15 00 26 ff 03 95 01 75 0a 81 02 15  .......E..&....u....
00000118: 00 25 00 75 06 95 01 81 03 05 01 09 37 15 01 25 08 35 00 46  .%.u........7..%.5.F
0000012c: 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35  ;.f..u....Bu.....%.5
00000140: 00 45 00 65 00 81 03 05 09 19 10 29 1e 15 00 25 01 75 01 95  .E.e.......)...%.u..
00000154: 0f 81 02 15 00 25 00 75 01 95 01 81 03 05 0c 0a 82 00 15 00  .....%.u............
00000168: 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 05 0c  %...u.....%.u.......
0000017c: 09 01 a1 01 0a 81 00 15 00 26 ff 00 95 01 75 04 81 02 15 00  .........&....u.....
00000190: 25 00 95 01 75 04 81 03 0a 84 00 15 00 26 ff 00 95 01 75 04  %...u........&....u.
000001a4: 81 02 15 00 25 00 95 01 75 04 81 03 0a 85 00 15 00 26 ff 00  ....%...u........&..
000001b8: 95 01 75 08 81 02 0a 99 00 15 00 26 ff 00 95 01 75 04 81 02  ..u........&....u...
000001cc: 15 00 25 00 95 01 75 04 81 03 0a 9e 00 15 00 26 ff 00 95 01  ..%...u........&....
000001e0: 75 08 81 02 0a a1 00 15 00 26 ff 00 95 01 75 08 81 02 0a a2  u........&....u.....
000001f4: 00 15 00 26 ff 00 95 01 75 08 81 02 0a a3 00 15 00 26 ff 00  ...&....u........&..
00000208: 95 01 75 08 81 02 0a a4 00 15 00 26 ff 00 95 01 75 08 81 02  ..u........&....u...
0000021c: 0a b9 00 15 00 26 ff 00 95 01 75 08 81 02 0a ba 00 15 00 26  .....&....u........&
00000230: ff 00 95 01 75 08 81 02 0a bb 00 15 00 26 ff 00 95 01 75 08  ....u........&....u.
00000244: 81 02 0a be 00 15 00 26 ff 00 95 01 75 08 81 02 0a c0 00 15  .......&....u.......
00000258: 00 26 ff 00 95 01 75 08 81 02 0a c1 00 15 00 26 ff 00 95 01  .&....u........&....
0000026c: 75 08 81 02 0a c2 00 15 00 26 ff 00 95 01 75 08 81 02 0a c3  u........&....u.....
00000280: 00 15 00 26 ff 00 95 01 75 08 81 02 0a c4 00 15 00 26 ff 00  ...&....u........&..
00000294: 95 01 75 08 81 02 0a c5 00 15 00 26 ff 00 95 01 75 08 81 02  ..u........&....u...
000002a8: 0a c6 00 15 00 26 ff 00 95 01 75 08 81 02 0a c7 00 15 00 26  .....&....u........&
000002bc: ff 00 95 01 75 08 81 02 0a c8 00 15 00 26 ff 00 95 01 75 08  ....u........&....u.
000002d0: 81 02 c0 05 0c 09 01 85 02 a1 01 05 0c 0a 23 02 15 00 25 01  ..............#...%.
000002e4: 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09  ..u.....%.u.........
000002f8: 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25  !........%.u.......%
0000030c: 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09  .u......p..%du......
00000320: 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15  Pf..U...&..u........
00000334: 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff  .&..u.....e.U..|..&.
00000348: 00 75 08 95 01 91 02 c0 05 06 09 20 85 04 15 00 26 ff 00 75  .u......... ....&..u
0000035c: 08 95 01 81 02 06 00 ff 09 01 a1 02 85 06 09 01 15 00 25 64  ..................%d
00000370: 75 08 95 01 b1 02 09 02 15 00 25 64 75 08 95 01 b1 02 09 03  u.........%du.......
00000384: 15 00 26 ff 00 75 08 95 01 b1 02 09 04 26 ff 00 75 08 95 3c  ..&..u.......&..u..<
00000398: b2 02 01 c0 06 00 ff 09 02 a1 02 85 07 09 05 15 00 25 64 75  .................%du
000003ac: 08 95 01 b1 02 09 06 15 00 25 64 75 08 95 01 b1 02 09 07 15  .........%du........
000003c0: 00 25 64 75 08 95 01 b1 02 c0 06 00 ff 09 03 a1 02 85 08 09  .%du................
000003d4: 08 15 00 25 64 75 08 95 01 b1 02 09 09 15 00 25 64 75 08 95  ...%du.........%du..
000003e8: 01 b1 02 09 0a 15 00 26 ff 00 75 08 95 01 b1 02 c0 06 00 ff  .......&..u.........
000003fc: 09 04 a1 01 85 09 09 0b 15 00 25 64 75 08 95 01 b1 02 09 0c  ..........%du.......
00000410: 15 00 25 64 75 08 95 01 b1 02 09 0d 15 00 25 64 75 08 95 01  ..%du.........%du...
00000424: b1 02 09 0e 15 00 26 ff 00 75 08 95 01 b1 02 09 0f 26 ff 00  ......&..u.......&..
00000438: 75 08 95 3c b2 02 01 c0 06 00 ff 09 05 a1 01 85 0a 09 10 15  u..<................
0000044c: 00 27 ff ff ff 7f 75 20 95 01 81 02 09 11 15 00 27 ff ff ff  .'....u ........'...
[... cut ...]  00 26 ff 00 75 08 95 02 81 02  .u ........&..u.....
00000474: 09 13 15 00 26 ff 00 75 08 95 01 81 02 c0 06 00 ff 09 06 a1  ....&..u............
00000488: 02 85 0b 09 14 15 00 25 64 75 08 95 01 b1 02 c0 c0 05 01 09  .......%du..........
0000049c: 06 a1 01 85 05 05 07 19 e0 29 e7 15 00 25 01 75 01 95 08 81  .........)...%.u....
000004b0: 02 95 01 75 08 81 03 95 06 75 08 15 00 25 65 05 07 19 00 29  ...u.....u...%e....)
000004c4: 65 81 00 c0                                                  e...
```

hibes, 2020-05-12:
> I'm in the process of removing `debug_level` in favor of using the normal HID kernel log infrastructure. So this setting probably has hardly an effect currently. I may add some logging during the next releases which uses `debug_level` again thus it still exists.

Roger that. Thanks for the clarity!

> Please report rumble and connection problems in the appropriate issue, or create a separate issue if unsure. This will be a lot easier for us to track and fix problems. Thank you.

Understood,

> The Xbox Elite controller support is currently incomplete. Could you please find the vendor/device ID when connecting the controller via Bluetooth in Windows? This may help us bypass SDL fixups [...]

Unfortunately I don't have any Windows installations

> Also, please let me know the report descriptor as asked for in [#142 (comment)](https://github.com/atar-axis/xpadneo/issues/142#issuecomment-624070392).

> This issue should be about detecting the controller and mapping its buttons and axis correctly only. The rumble command is exactly the same (4 bits + 4 bytes motor strength + 2 bytes pulse width + 1 byte loop count), so I'd like to stay away from putting any rumble issues here.

Got it, sorry about that.  Other than that descriptor, is there anything I can do to help? For the moment, I only have Linux. I'll keep an eye for other people around who have a Windows laptop which I might be able to borrow for a minute.  Among the others around me, I've only seen Apple laptops, and I doubt these folks are dual booting Windows.

kakra, 2020-05-12:
Thanks for the descriptor. I may be able to look into it during the weekend. But I fear we'd really need info on how the device identifies when connected to Windows via Bluetooth. Maybe someone else may help? Here should be at least 2 or 3 people who contributed issue reports and are using the Elite controller.

gytisrepecka, 2020-05-21:
I do have Elite 2 controller and could hook it up on virtualized (VirtualBox) Windows 10 Home by sharing USB Bluetooth dongle to virtual machine and pairing joystick via Windows. Would that help?

kakra, 2020-05-21:
Yes, definitely that helps. Please try. I'm not sure how to find the device ID then but it probably shows up in the device manager. We'd also need the HID device descriptor then but I'm even do not know where that is stored or can be looked up in Windows. So that's a secondary objective.

mercuriete, 2020-05-30:
I'm sorry but I have a elite 2 controller and xpadneo is picking up the device properly.
So far I didn't find any issue.
* force feedback working
* battery report properly working
* all button working but __guide button__.
* steam picking up the controller and working properly in proton games.

my dmesg:

```
[   38.880128] hid_xpadneo: loading out-of-tree module taints kernel.
[   62.094346] xpadneo 0005:045E:0B05.0005: fixing up report size
[   62.094348] xpadneo 0005:045E:0B05.0005: fixing up Rx axis
[   62.094350] xpadneo 0005:045E:0B05.0005: fixing up Ry axis
[   62.094351] xpadneo 0005:045E:0B05.0005: fixing up Z axis
[   62.094352] xpadneo 0005:045E:0B05.0005: fixing up Rz axis
[   62.094353] xpadneo 0005:045E:0B05.0005: fixing up button mapping
[   62.095348] xpadneo 0005:045E:0B05.0005: battery registered
[   82.640860] input: Xbox Elite Wireless Controller as /devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-10/1-10:1.0/bluetooth/hci0/hci0:71/0005:045E:0B05.0005/input/input23
[   82.641178] xpadneo 0005:045E:0B05.0005: input,hidraw4: BLUETOOTH HID v9.03 Gamepad [Xbox Elite Wireless Controller] on (censored)
```

kakra, 2020-05-30:
First of all: Yes, please use only master for testing and debugging.

How do you detect the guide button not working? The guide button is not exposed as a classical gamepad button but as a keyboard button instead. So usual joystick tools won't show it working. But while the Steam client is in front, pressing the button should take you to your library or into big screen mode. Also, when a browser is in front, the guide button should navigate to your configured browser start page. It's actually the keyboard home button.

If it doesn't, we'd need a fixup for your guide button. Please post the HID report descriptor as described in the debugging instructions:

    xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor

Also, please run `evtest`, try to not touch the joysticks or triggers to reduce the amount of log, then press each button in this order: A, B, X, Y, LB, RB, Select 🗗, Start ☰, Home (Xbox). There's also LS and RS but when pressing those, you usually also move the sticks and that generates a lot of noise. Also the triggers are not buttons. For the moment, let's just work with the 9 buttons I mentioned.

mercuriete, 2020-05-30:
descriptor:

```
xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0b 15 00 25 01 75 01 95 0b 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 05 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81  u........$...%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 01 09 01 a1 00 09 40 09  ...%.u............@.
000000c8: 41 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 01 a1  A..'......u.........
000000dc: 00 09 43 09 44 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05  ..C.D..'......u.....
000000f0: 01 09 42 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06  ..B..&....u.....%.u.
00000104: 95 01 81 03 05 01 09 45 15 00 26 ff 03 95 01 75 0a 81 02 15  .......E..&....u....
00000118: 00 25 00 75 06 95 01 81 03 05 01 09 37 15 01 25 08 35 00 46  .%.u........7..%.5.F
0000012c: 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35  ;.f..u....Bu.....%.5
00000140: 00 45 00 65 00 81 03 05 09 19 10 29 1e 15 00 25 01 75 01 95  .E.e.......)...%.u..
00000154: 0f 81 02 15 00 25 00 75 01 95 01 81 03 05 0c 0a 82 00 15 00  .....%.u............
00000168: 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 05 0c  %...u.....%.u.......
0000017c: 09 01 a1 01 0a 81 00 15 00 26 ff 00 95 01 75 04 81 02 15 00  .........&....u.....
00000190: 25 00 95 01 75 04 81 03 0a 84 00 15 00 26 ff 00 95 01 75 04  %...u........&....u.
000001a4: 81 02 15 00 25 00 95 01 75 04 81 03 0a 85 00 15 00 26 ff 00  ....%...u........&..
000001b8: 95 01 75 08 81 02 0a 99 00 15 00 26 ff 00 95 01 75 04 81 02  ..u........&....u...
000001cc: 15 00 25 00 95 01 75 04 81 03 0a 9e 00 15 00 26 ff 00 95 01  ..%...u........&....
000001e0: 75 08 81 02 0a a1 00 15 00 26 ff 00 95 01 75 08 81 02 0a a2  u........&....u.....
000001f4: 00 15 00 26 ff 00 95 01 75 08 81 02 0a a3 00 15 00 26 ff 00  ...&....u........&..
00000208: 95 01 75 08 81 02 0a a4 00 15 00 26 ff 00 95 01 75 08 81 02  ..u........&....u...
0000021c: 0a b9 00 15 00 26 ff 00 95 01 75 08 81 02 0a ba 00 15 00 26  .....&....u........&
00000230: ff 00 95 01 75 08 81 02 0a bb 00 15 00 26 ff 00 95 01 75 08  ....u........&....u.
00000244: 81 02 0a be 00 15 00 26 ff 00 95 01 75 08 81 02 0a c0 00 15  .......&....u.......
00000258: 00 26 ff 00 95 01 75 08 81 02 0a c1 00 15 00 26 ff 00 95 01  .&....u........&....
0000026c: 75 08 81 02 0a c2 00 15 00 26 ff 00 95 01 75 08 81 02 0a c3  u........&....u.....
00000280: 00 15 00 26 ff 00 95 01 75 08 81 02 0a c4 00 15 00 26 ff 00  ...&....u........&..
00000294: 95 01 75 08 81 02 0a c5 00 15 00 26 ff 00 95 01 75 08 81 02  ..u........&....u...
000002a8: 0a c6 00 15 00 26 ff 00 95 01 75 08 81 02 0a c7 00 15 00 26  .....&....u........&
000002bc: ff 00 95 01 75 08 81 02 0a c8 00 15 00 26 ff 00 95 01 75 08  ....u........&....u.
000002d0: 81 02 c0 05 0c 09 01 85 02 a1 01 05 0c 0a 23 02 15 00 25 01  ..............#...%.
000002e4: 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09  ..u.....%.u.........
000002f8: 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25  !........%.u.......%
0000030c: 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09  .u......p..%du......
00000320: 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15  Pf..U...&..u........
00000334: 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff  .&..u.....e.U..|..&.
00000348: 00 75 08 95 01 91 02 c0 05 06 09 20 85 04 15 00 26 ff 00 75  .u......... ....&..u
0000035c: 08 95 01 81 02 06 00 ff 09 01 a1 02 85 06 09 01 15 00 25 64  ..................%d
00000370: 75 08 95 01 b1 02 09 02 15 00 25 64 75 08 95 01 b1 02 09 03  u.........%du.......
00000384: 15 00 26 ff 00 75 08 95 01 b1 02 09 04 26 ff 00 75 08 95 3c  ..&..u.......&..u..<
00000398: b2 02 01 c0 06 00 ff 09 02 a1 02 85 07 09 05 15 00 25 64 75  .................%du
000003ac: 08 95 01 b1 02 09 06 15 00 25 64 75 08 95 01 b1 02 09 07 15  .........%du........
000003c0: 00 25 64 75 08 95 01 b1 02 c0 06 00 ff 09 03 a1 02 85 08 09  .%du................
000003d4: 08 15 00 25 64 75 08 95 01 b1 02 09 09 15 00 25 64 75 08 95  ...%du.........%du..
000003e8: 01 b1 02 09 0a 15 00 26 ff 00 75 08 95 01 b1 02 c0 06 00 ff  .......&..u.........
000003fc: 09 04 a1 01 85 09 09 0b 15 00 25 64 75 08 95 01 b1 02 09 0c  ..........%du.......
00000410: 15 00 25 64 75 08 95 01 b1 02 09 0d 15 00 25 64 75 08 95 01  ..%du.........%du...
00000424: b1 02 09 0e 15 00 26 ff 00 75 08 95 01 b1 02 09 0f 26 ff 00  ......&..u.......&..
00000438: 75 08 95 3c b2 02 01 c0 06 00 ff 09 05 a1 01 85 0a 09 10 15  u..<................
0000044c: 00 27 ff ff ff 7f 75 20 95 01 81 02 09 11 15 00 27 ff ff ff  .'....u ........'...
00000460: 7f 75 20 95 01 81 02 09 12 15 00 26 ff 00 75 08 95 02 81 02  .u ........&..u.....
00000474: 09 13 15 00 26 ff 00 75 08 95 01 81 02 c0 06 00 ff 09 06 a1  ....&..u............
[... cut ...]
Event: time 1590859690.727623, -------------- SYN_REPORT ------------
Event: time 1590859691.767777, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90008
Event: time 1590859691.767777, type 1 (EV_KEY), code 315 (BTN_START), value 1
Event: time 1590859691.767777, -------------- SYN_REPORT ------------
Event: time 1590859691.937798, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90008
Event: time 1590859691.937798, type 1 (EV_KEY), code 315 (BTN_START), value 0
Event: time 1590859691.937798, -------------- SYN_REPORT ------------
```

mercuriete, 2020-05-30:
Sorry for the noise,
dont know what happened but I had master from several days ago.

I did a git pull and now the guide button works.
I will do a evtest again

evtest
```
Select the device event number [0-24]: 23
Input driver version is 1.0.1
Input device ID: bus 0x5 vendor 0x45e product 0xb05 version 0x903
Input device name: "Xbox Elite Wireless Controller"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 1 (KEY_ESC)
    Event code 2 (KEY_1)
    Event code 3 (KEY_2)
    Event code 4 (KEY_3)
    Event code 5 (KEY_4)
    Event code 6 (KEY_5)
    Event code 7 (KEY_6)
    Event code 8 (KEY_7)
    Event code 9 (KEY_8)
    Event code 10 (KEY_9)
    Event code 11 (KEY_0)
    Event code 12 (KEY_MINUS)
    Event code 13 (KEY_EQUAL)
    Event code 14 (KEY_BACKSPACE)
    Event code 15 (KEY_TAB)
    Event code 16 (KEY_Q)
    Event code 17 (KEY_W)
    Event code 18 (KEY_E)
    Event code 19 (KEY_R)
    Event code 20 (KEY_T)
    Event code 21 (KEY_Y)
    Event code 22 (KEY_U)
    Event code 23 (KEY_I)
    Event code 24 (KEY_O)
    Event code 25 (KEY_P)
    Event code 26 (KEY_LEFTBRACE)
    Event code 27 (KEY_RIGHTBRACE)
    Event code 28 (KEY_ENTER)
    Event code 29 (KEY_LEFTCTRL)
    Event code 30 (KEY_A)
    Event code 31 (KEY_S)
    Event code 32 (KEY_D)
    Event code 33 (KEY_F)
    Event code 34 (KEY_G)
    Event code 35 (KEY_H)
    Event code 36 (KEY_J)
    Event code 37 (KEY_K)
    Event code 38 (KEY_L)
    Event code 39 (KEY_SEMICOLON)
    Event code 40 (KEY_APOSTROPHE)
    Event code 41 (KEY_GRAVE)
    Event code 42 (KEY_LEFTSHIFT)
    Event code 43 (KEY_BACKSLASH)
    Event code 44 (KEY_Z)
    Event code 45 (KEY_X)
    Event code 46 (KEY_C)
    Event code 47 (KEY_V)
    Event code 48 (KEY_B)
    Event code 49 (KEY_N)
    Event code 50 (KEY_M)
    Event code 51 (KEY_COMMA)
    Event code 52 (KEY_DOT)
    Event code 53 (KEY_SLASH)
    Event code 54 (KEY_RIGHTSHIFT)
    Event code 55 (KEY_KPASTERISK)
[... 181 lines cut ...]
Event: time 1590863288.252775, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90006
Event: time 1590863288.252775, type 1 (EV_KEY), code 311 (BTN_TR), value 1
Event: time 1590863288.252775, -------------- SYN_REPORT ------------
Event: time 1590863288.422784, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90006
Event: time 1590863288.422784, type 1 (EV_KEY), code 311 (BTN_TR), value 0
Event: time 1590863288.422784, -------------- SYN_REPORT ------------
Event: time 1590863290.522438, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90007
Event: time 1590863290.522438, type 1 (EV_KEY), code 314 (BTN_SELECT), value 1
Event: time 1590863290.522438, -------------- SYN_REPORT ------------
Event: time 1590863290.692463, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90007
Event: time 1590863290.692463, type 1 (EV_KEY), code 314 (BTN_SELECT), value 0
Event: time 1590863290.692463, -------------- SYN_REPORT ------------
Event: time 1590863291.242519, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90008
Event: time 1590863291.242519, type 1 (EV_KEY), code 315 (BTN_START), value 1
Event: time 1590863291.242519, -------------- SYN_REPORT ------------
Event: time 1590863291.412543, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90008
Event: time 1590863291.412543, type 1 (EV_KEY), code 315 (BTN_START), value 0
Event: time 1590863291.412543, -------------- SYN_REPORT ------------
Event: time 1590863294.748393, type 4 (EV_MSC), code 4 (MSC_SCAN), value c0223
Event: time 1590863294.748393, type 1 (EV_KEY), code 172 (KEY_HOMEPAGE), value 1
Event: time 1590863294.748393, -------------- SYN_REPORT ------------
Event: time 1590863294.902528, type 4 (EV_MSC), code 4 (MSC_SCAN), value c0223
Event: time 1590863294.902528, type 1 (EV_KEY), code 172 (KEY_HOMEPAGE), value 0
Event: time 1590863294.902528, -------------- SYN_REPORT ------------
```

kakra, 2020-05-31:
> EDIT: Sorry I did not mention... I am using bluethoot. I didn't test with usb cable.
> do you need the same over usb?

No, this driver does not (officially) handle USB (at least it wasn't tested, not sure if it is even the same protocol).

> I did a git pull and now the guide button works.
> I will do a evtest again

Cool. This was fixed by commits https://github.com/atar-axis/xpadneo/commit/695a7a547d657954b1cc8b700633c82a82e0d4ca about 4 weeks ago.

kakra, 2020-05-31:
@Draic Can we close this now?

Draic, 2020-05-31:
sounds like it. Good that it is working now :) thanks everyone

kakra, 2020-05-31:
Thanks. If any issues of the original request persist, feel free to re-open.

closed 2020-05-31 by kakra