atar-axis/xpadneo #145: Xbox one on Jetson Nano

gordonsolar, 2019-11-24:
**Describe the bug**
the installation on the jetson nano does not work without errors.

**To Reproduce**
**1.**
I followed the installation instruction:
sudo apt-get install dkms linux-headers-`uname -r` did not work completetly.
linux-headers-'uname - r' does not exist - so i installed only dkms
**2.**
i downloaded the repo and run the installation script - without obvious errors.
Output:

* replacing version string if necessary
* copying module into /usr/src
* adding module to DKMS

Creating symlink /var/lib/dkms/hid-xpadneo/0.6.0/source ->
                 /usr/src/hid-xpadneo-0.6.0

DKMS: add completed.
* installing module (using DKMS)

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.9.140-tegra -C /lib/modules/4.9.140-tegra/build M=/var/lib/dkms/hid-xpadneo/0.6.0/build/src modules....
cleaning build area...

DKMS: build completed.

hid-xpadneo.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.9.140-tegra/updates/dkms/

Running the post_install script:
Disabling ERTM permanently (requires reboot)...
Disabling ERTM instantly...
Installing UDEV rule to automatically bind xpadneo (necessary on kernel < 4.16 and >= 4.20)...

depmod...

Backing up initrd.img-4.9.140-tegra to /boot/initrd.img-4.9.140-tegra.old-dkms
Making new initrd.img-4.9.140-tegra
(If next boot fails, revert to initrd.img-4.9.140-tegra.old-dkms image)
update-initramfs.......

DKMS: install completed.

**3.**
I used bluetoothctl to connect to the device.
First i experienced an unstable connection with connect yes, no, yes, no ...
After reboot and reinstallation of the driver it connected - the white light of the controller turned permanently on and the output from bluetoothctl seem to be ok.
Output:

[bluetooth]# scan on
Discovery started
[CHG] Controller 5C:87:9C:1F:9B:F8 Discovering: yes
[NEW] Device 9C:AA:1B:E7:FC:A2 Xbox Wireless Controller
[bluetooth]# pair 9C:AA:1B:E7:FC:A2
Attempting to pair with 9C:AA:1B:E7:FC:A2
[CHG] Device 9C:AA:1B:E7:FC:A2 Connected: yes
[CHG] Device 9C:AA:1B:E7:FC:A2 Modalias: usb:v045Ep02E0d0903
[CHG] Device 9C:AA:1B:E7:FC:A2 Modalias: usb:v045Ep02FDd0903
[CHG] Device 9C:AA:1B:E7:FC:A2 UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[CHG] Device 9C:AA:1B:E7:FC:A2 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device 9C:AA:1B:E7:FC:A2 ServicesResolved: yes
[CHG] Device 9C:AA:1B:E7:FC:A2 Paired: yes
Pairing successful
[CHG] Device 9C:AA:1B:E7:FC:A2 ServicesResolved: no
[CHG] Device 9C:AA:1B:E7:FC:A2 Connected: no
[bluetooth]# trust 9C:AA:1B:E7:FC:A2
[CHG] Device 9C:AA:1B:E7:FC:A2 Trusted: yes
Changing 9C:AA:1B:E7:FC:A2 trust succeeded
[bluetooth]# connect 9C:AA:1B:E7:FC:A2
Attempting to connect to 9C:AA:1B:E7:FC:A2
[CHG] Device 9C:AA:1B:E7:FC:A2 Connected: yes
Connection successful
[CHG] Device 9C:AA:1B:E7:FC:A2 ServicesResolved: yes
[Xbox Wireless Controller]# 

**4.**
output from dmesg shows some errors:
[ 1731.161499] xpadneo udev: 0005:045E:02FD.0001
[ 1731.163827] hid_xpadneo: loading out-of-tree module taints kernel.
[ 1731.164398] xpadneo: hello there!
[ 1731.192029] xpadneo 0005:045E:02FD.0001: unknown main item tag 0x0
[ 1731.198240] xpadneo 0005:045E:02FD.0001: unbalanced collection at end of report description
[ 1731.207428] xpadneo 0005:045E:02FD.0001: parse failed
[ 1731.212897] xpadneo: probe of 0005:045E:02FD.0001 failed with error -22
[ 1731.221282] xpadneo 0005:045E:02FD.0001: unknown main item tag 0x0
[ 1731.227576] xpadneo 0005:045E:02FD.0001: unbalanced collection at end of report description
[ 1731.237699] xpadneo 0005:045E:02FD.0001: parse failed
[ 1731.242849] xpadneo: probe of 0005:045E:02FD.0001 failed with error -22
[ 1731.250432] xpadneo 0005:045E:02FD.0001: unknown main item tag 0x0
[ 1731.256693] xpadneo 0005:045E:02FD.0001: unbalanced collection at end of report description
[ 1731.265130] xpadneo 0005:045E:02FD.0001: parse failed
[ 1731.270214] xpadneo: probe of 0005:045E:02FD.0001 failed with error -22
[ 1731.276983] xpadneo udev: ok

**5.**
turn the controller off and on again leads to the controller "rumbling" :-)
dmesg:
[ 2655.881227] xpadneo udev: 0005:045E:02FD.0002
[ 2655.941420] xpadneo 0005:045E:02FD.0002: unknown main item tag 0x0
[ 2655.949051] xpadneo 0005:045E:02FD.0002: input,hidraw0: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 5c:87:9c:1f:9b:f8
[ 2656.939412] xpadneo udev: ok
**6.**
using a small pyhton script allowed the reading of key events
`from evdev import InputDevice, categorize, ecodes

dev = InputDevice('/dev/input/event2')
print(dev)
for event in dev.read_loop():
    if event.type == ecodes.EV_KEY:
        print(categorize(event))`

**Expected behavior**
the installation is not straight forward and maybe some user will give up half the way to the goal.
For me installation and the driver works - many thanks to the author!

If useful, i can provide more debug information to maybe improve the installation process for this special board.

**System information**
Jetson Nano with Ubuntu 18.04.3 LTS (GNU/Linux 4.9.140-tegra aarch64)
The Bluetooth device is an Intel 8265NGW
see installation [https://www.jetsonhacks.com/2019/04/08/jetson-nano-intel-wifi-and-bluetooth/](url)

atar-axis, 2020-01-15:
You are welcome to change the documentation for this board - i am open to PR's :) Currently I am too busy too do this my own

kakra, 2020-05-09:
That device seems to come with a broken device descriptor. We could probably fix that in the driver. Please post the descriptor dump tho I'm not sure how to get it since the kernel already rejects the device before xpadneo has a chance to grab it. `sudo usbhid-dump` may work tho I never managed to enumerate BT devices with it. Maybe fire up wireshark and monitor the BT stack. I any case you may want to unload xpadneo first and prevent it from being loaded.

kakra, 2020-06-11:
Can we close this? Current master has a fixup for the unbalanced collection error probably. Otherwise please open a new bug report with all the details and close this in favor of the new report.

gordonsolar, 2020-06-14:
Sorry for the late response.
I could solve the problem by disabling ertm in .bashrc like it is also described in the setup doc:

#echo "y" > /sys/module/bluetooth/parameters/disable_ertm
sudo bash -c "echo 1 > /sys/module/bluetooth/parameters/disable_ertm"

not very nice, since i have to enter my pwd two times for each login - but is ok for me.

closed 2020-06-14 by gordonsolar

kakra, 2020-06-14:
Why don't you add the parameter to `/etc/modprobe.d/99-bluetooth.conf` or the kernel cmdline (if your kernel has no modules)?

gordonsolar, 2020-06-22:
I tried in the past - but it did not work.
Now i checked the net again and found the sysfsutils method - which works for me:

`sudo apt-get install sysfsutils`
`sudo nano /etc/sysfs.conf` 
and append _/module/bluetooth/parameters/disable_ertm=1_ to the end of the file
and then reboot.

kakra, 2020-06-22:
Yes, that works, too, as long as the module is loaded before sysfsutils.