atar-axis/xpadneo #146: Despite having two pads connected, info about battery is only displayed for the first one.

dragon99919, 2019-11-25:
System: Linux Mint 19.2 Cinnamon Edition

As seen on screenshot below, despite having two connected pads, only battery information about the first one is being displayed.

[image]

atar-axis, 2020-01-15:
Thanks! I am curretly too busy to fix this but i will mark is at a bug so if someone finds time (maybe me in the future)

deathxxx123, 2020-01-15:
Oh in Mint you can see controller battery just in right down corner...?
In ubuntu i need to go in options menu and go for Bluetooth devices, then i can see battery level indicator.

dragon99919, 2020-01-20:
> Oh in Mint you can see controller battery just in right down corner...?
> In ubuntu i need to go in options menu and go in Power options, then i can see battery level.
> [image]

Isn't that 18.04 LTS with Gnome? If yes then I'm pretty sure that even if battery does not show up by default, there are a lot of extensions that should make it work. [That one](https://extensions.gnome.org/extension/817/battery-status/) for example seems to be the most popular regarding the customization (one of the only configurable as top comment says), maybe you could give it a shot.

deathxxx123, 2020-01-20:
I remember now. I install Ubuntu with Minimal options. Maybe in this way, i don't have some packages like battery icons and others. And yes, that is Ubuntu Gnome - Default one from Ubuntu.com
For now, i will leave it in this way.

atar-axis, 2020-02-20:
Doesn't seem to be a driver problem since it works on KDE
[image]

You should try it at the desktop environment community I guess

closed 2020-02-20 by atar-axis