atar-axis/xpadneo #147: Install on Raspberry Pi Hypriotos failed

frank-qcd-qk, 2019-11-25:
**Describe the bug**
Attempt to install on Raspberry Pi 3 Hypriotos, but install failed

**To Reproduce**
Installed dkmg and headers then attempted sudo ./install

**System information**
Linux avlduck3 4.14.34-hypriotos-v7+ #1 SMP Sun Apr 22 14:57:31 UTC 2018 armv7l GNU/Linux
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.34-hypriotos-v7+ (circleci@default-ad29452a-e7b7-4bea-bd85-60df8b208ec4) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03)) #1 SMP Sun Apr 22 14:57:31 UTC 2018
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 256 MiB at 0x1ec00000
[    0.000000] On node 0 totalpages: 242688
[    0.000000] free_area_init_node: node 0, pgdat 80c85b80, node_mem_map bab25000
[    0.000000]   Normal zone: 2133 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 242688 pages, LIFO batch:31
[    0.000000] random: fast init done
[    0.000000] percpu: Embedded 17 pages/cpu @baace000 s38784 r8192 d22656 u69632
[    0.000000] pcpu-alloc: s38784 r8192 d22656 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: 8250.nr_uarts=0 cma=256M bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 cgroup_enable=cpuset cgroup_enable=memory swapaccount=1 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 685784K/970752K available (7168K kernel code, 578K rwdata, 2072K rodata, 1024K init, 706K bss, 22824K reserved, 262144K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
                   lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x80008000 - 0x80800000   (8160 kB)
                     .init : 0x80b00000 - 0x80c00000   (1024 kB)
                     .data : 0x80c00000 - 0x80c90bf4   ( 579 kB)
                      .bss : 0x80c97f70 - 0x80d48ab4   ( 707 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 25307 entries in 75 pages
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000022] Switching to timer-based delay loop, resolution 52ns
[    0.000282] Console: colour dummy device 80x30
[    0.000841] console [tty1] enabled
[    0.000876] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000916] pid_max: default: 32768 minimum: 301
[    0.001228] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001260] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002159] CPU: Testing write buffer coherency: ok
[    0.002586] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002987] Setting up static identity map for 0x100000 - 0x10003c
[    0.003124] Hierarchical SRCU implementation.
[    0.003841] smp: Bringing up secondary CPUs ...
[    0.004616] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.005436] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.006246] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.006354] smp: Brought up 1 node, 4 CPUs
[    0.006424] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.006445] CPU: All CPU(s) started in HYP mode.
[    0.006463] CPU: Virtualization extensions available.
[    0.007316] devtmpfs: initialized
[    0.018218] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.018471] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[... 231 lines cut ...]
[   58.022670] Bluetooth: Core ver 2.22
[   58.022754] NET: Registered protocol family 31
[   58.022760] Bluetooth: HCI device and connection manager initialized
[   58.022781] Bluetooth: HCI socket layer initialized
[   58.022794] Bluetooth: L2CAP socket layer initialized
[   58.022832] Bluetooth: SCO socket layer initialized
[   58.036010] Bluetooth: HCI UART driver ver 2.3
[   58.036025] Bluetooth: HCI UART protocol H4 registered
[   58.036031] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   58.036273] Bluetooth: HCI UART protocol Broadcom registered
[   58.276861] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   58.276868] Bluetooth: BNEP filters: protocol multicast
[   58.276880] Bluetooth: BNEP socket layer initialized

**make.log**
DKMS make.log for hid-xpadneo-0.6.0 for kernel 4.14.34-hypriotos-v7+ (armv7l)
Mon Nov 25 17:28:34 UTC 2019
make: Entering directory '/usr/src/linux-headers-4.14.34-hypriotos-v7+'
  CC [M]  /var/lib/dkms/hid-xpadneo/0.6.0/build/src/hid-xpadneo.o
/bin/sh: 1: scripts/basic/fixdep: Exec format error
scripts/Makefile.build:334: recipe for target '/var/lib/dkms/hid-xpadneo/0.6.0/build/src/hid-xpadneo.o' failed
make[1]: *** [/var/lib/dkms/hid-xpadneo/0.6.0/build/src/hid-xpadneo.o] Error 2
Makefile:1528: recipe for target '_module_/var/lib/dkms/hid-xpadneo/0.6.0/build/src' failed
make: *** [_module_/var/lib/dkms/hid-xpadneo/0.6.0/build/src] Error 2
make: Leaving directory '/usr/src/linux-headers-4.14.34-hypriotos-v7+'

deathxxx123, 2020-01-07:
I'm not a pro, but i think i saw somewhere, for xpadneo support you need at least kernel 4.19+. Maybe you'r kernel do not have modules, required for this compile.

kakra, 2020-06-11:
The install script has seen some changes since then, so is this still an issue? OTOH, this looks a bit like a cross-compilation problem: `Exec format error` is clearly something xpadneo couldn't cause. You should check proper installation of all packages and the OS.

closed 2020-10-17 by frank-qcd-qk