atar-axis/xpadneo #148: GalliumOS (Ubuntu 18.04) with Linux Kernels 5.4 connect disconnect loop.

ryanturner328, 2019-11-25:
**Describe the bug**
I run GalliumOS (Braswell) on a Chromebook. I have updated my kernels to connect to my xbox one controller, and it worked flawlessly for about 2 weeks. All of a sudden I get on today and attempted to connect my controller so I could game and it just doesn't work. I am getting the connect disconnect loop and the bluez error. (Please bear with me, im kinda linux noob)

**To Reproduce**
Steps to reproduce the behavior: I attempt to connect, and pair, and disconnect and nothing happens. I follow the readme from front to back time and time again.

**System information**
Please enable debugging output <https://atar-axis.github.io/xpadneo/index.html#debugging>
and add at least the following outputs

-   `uname -a`
```Linux chrx 5.4.0-050400-generic #201911242031 SMP Mon Nov 25 01:35:10 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux```

-   `dmesg`
```[    0.000000] microcode: microcode updated early to revision 0x411, date = 2019-04-23
[    0.000000] Linux version 5.4.0-050400-generic (kernel@tangerine) (gcc version 9.2.1 20191109 (Ubuntu 9.2.1-19ubuntu1)) #201911242031 SMP Mon Nov 25 01:35:10 UTC 2019
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.4.0-050400-generic root=UUID=7a84fdef-07db-4d7d-b11d-f935a167b73a ro quiet splash boot=local acpi_osi=Linux acpi_backlight=vendor add_efi_memmap intel_pstate=enable i915.modeset=1 tpm_tis.force=1 tpm_tis.interrupts=0 nmi_watchdog=panic,lapic elevator=noop noresume vt.handoff=1
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000001fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000020000000-0x00000000200fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000020100000-0x000000007cbfefff] usable
[    0.000000] BIOS-e820: [mem 0x000000007cbff000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fea00000-0x00000000febfffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed03000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed06000-0x00000000fed06fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed08000-0x00000000fed08fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1cfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed40000-0x00000000fed44fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fed83fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000017fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: GOOGLE Kefka, BIOS  07/26/2019
[    0.000000] tsc: Detected 1600.000 MHz processor
[    0.001459] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.001462] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.001471] last_pfn = 0x180000 max_arch_pfn = 0x400000000
[    0.001476] MTRR default type: uncachable
[    0.001477] MTRR fixed ranges enabled:
[    0.001480]   00000-9FFFF write-back
[    0.001481]   A0000-BFFFF uncachable
[    0.001483]   C0000-FFFFF write-back
[    0.001484] MTRR variable ranges enabled:
[    0.001487]   0 base 000000000 mask F80000000 write-back
[    0.001489]   1 base 07D800000 mask FFF800000 uncachable
[    0.001490]   2 base 07E000000 mask FFE000000 uncachable
[    0.001492]   3 base 0FF800000 mask FFF800000 uncachable
[    0.001494]   4 base 100000000 mask F00000000 write-back
[    0.001495]   5 disabled
[    0.001496]   6 disabled
[    0.001497]   7 disabled
[    0.001602] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.001724] total RAM covered: 6104M
[    0.002536] Found optimal setting for mtrr clean up
[    0.002538]  gran_size: 64K 	chunk_size: 64M 	num_reg: 4  	lose cover RAM: 0G
[    0.002637] e820: update [mem 0x7d800000-0xffffffff] usable ==> reserved
[    0.002643] last_pfn = 0x7cbff max_arch_pfn = 0x400000000
[    0.007690] check: Scanning 1 areas for low memory corruption
[... 649 lines cut ...]
[   11.208641] cht-bsw-rt5645 cht-bsw-rt5645: rt5645-aif1 <-> ssp2-port mapping ok
[   11.208759] cht-bsw-rt5645 cht-bsw-rt5645: ASoC: no DMI vendor name!
[   11.283180] input: chtrt5650 Headset as /devices/platform/808622A8:00/cht-bsw-rt5645/sound/card0/input10
[   12.468264] cdc_acm 1-3:1.1: failed to set dtr/rts
[   22.003600] cdc_acm 1-3:1.1: failed to set dtr/rts
[   22.007592] cdc_acm 1-3:1.1: failed to set dtr/rts
[   28.019529] cdc_acm 1-3:1.1: failed to set dtr/rts
[   75.514149] wlp2s0: authenticate with d4:04:cd:b8:96:94
[   75.526964] wlp2s0: send auth to d4:04:cd:b8:96:94 (try 1/3)
[   75.528815] wlp2s0: authenticated
[   75.533913] wlp2s0: associate with d4:04:cd:b8:96:94 (try 1/3)
[   75.535371] wlp2s0: RX AssocResp from d4:04:cd:b8:96:94 (capab=0x1011 status=0 aid=1)
[   75.548021] wlp2s0: associated
[   75.550039] wlp2s0: Limiting TX power to 30 (30 - 0) dBm as advertised by d4:04:cd:b8:96:94
[   75.605115] IPv6: ADDRCONF(NETDEV_CHANGE): wlp2s0: link becomes ready
[   78.624169] intel_sst_acpi 808622A8:00: FW Version 01.0b.02.02
[   78.863123] Bluetooth: RFCOMM TTY layer initialized
[   78.863135] Bluetooth: RFCOMM socket layer initialized
[   78.863151] Bluetooth: RFCOMM ver 1.11 ```

-  ``sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;``
(nothing happened here)

**Additional context**  
Add any other context about the problem here.

deathxxx123, 2020-01-06:
What is this bluetooth? Copy output there after this command:
`hciconfig -a`

And what controller you have?
Use this command in terminal and then connect and disconnect controller, then copy last 20 lines of command output here:
`dmesg -wH`

Maybe you'r ERTM is not dissabled. Try disable it:
Method 1:
`echo -e "options bluetooth disable_ertm=Y" | sudo tee '/etc/modprobe.d/xbox_bluetooth_ertm.conf'`
Then restart and try connecto controller.
Method 2:
Edit /etc/default/grub
and add the
"bluetooth.disable_ertm=1"
to the GRUB_CMDLINE_LINUX line. , so after edit it will looks like this:
`GRUB_CMDLINE_LINUX="bluetooth.disable_ertm=1"`

Then run
`sudo grub-mkconfig -o /boot/grub/grub.cfg`
and reboot

atar-axis, 2020-02-20:
closing this issue due to inactivity

closed 2020-02-20 by atar-axis