atar-axis/xpadneo #155: default axis mapping slightly wrong

D33M0N, 2020-01-07:
I am rather newbie in this field, but for me in Manjaro linux using AUR version of this driver I see the default axes mappings are:
0:ABS_X
1:ABS_Y
2:ABS_Z
3:ABS_RX
4:ABS_RY
5:ABS_RZ
6:ABD_HAT0X
7:ABD_HAT0Y

and after every computer reboot I have manually change them (with jstest-gtk) to:
0:ABS_X
1:ABS_Y
3:ABS_RX
4:ABS_RY
2:ABS_Z
5:ABS_RZ
6:ABD_HAT0X
7:ABD_HAT0Y

Not sure is this (this particular) linux or linux in general or driver issue. Using controllers 1708.
( If it is my linux problem then any hint how to make the change permanent would be nice )

atar-axis, 2020-02-20:
I am not sure what the problem is, please try to explain a bit more precise. I will close this issue as "not reproducible" as long as I am waiting for your answer.

closed 2020-02-20 by atar-axis

jefersonnavarro, 2020-05-06:
https://github.com/atar-axis/xpadneo/issues/182#issue-613584729