atar-axis/xpadneo #156: EndlessOS linux support

deathxxx123, 2020-01-10:
_Is your feature request related to a problem? Please describe._
I cannot install this on EndlessOS, i need alternative way. Everyone can try from live-usb with EndlessOS. It's with presistence feature by default, so no need for installation on HDD. EndlessOS is using flatpak packets.

_Describe the solution you'd like_
Maybe a flatpak app of XPADNEO will be a solution, but there is no exist this in flathub.

_Describe alternatives you've considered_
Flatpak, or can be a some alternative way?

_Additional context_
It will be great if someone can help with that.

Kiffmet, 2020-01-14:
AFAIK it is currently impossible to ship kernel mode drivers via flatpak.

deathxxx123, 2020-01-14:
> AFAIK it is currently impossible to ship kernel mode drivers via flatpak.

Yes, i was wondering about that. Anyway it's not a good gaming distro.

deathxxx123, 2020-01-14:
I contact with Endless OS support, and they seems to be a interested to have a flatpak version of xpadneo. Maybe they should contact with you directly, to make possible flatpak version of this.
I'm only a medium linux user with some terminal commands knowledge.

atar-axis, 2020-01-15:
Sorry but I am currently too busy to do anything like that - but I am open for PR's ;)

deathxxx123, 2020-01-15:
What is "PR's"?
I hope you are working on XPADNEO for some more improvements, or maybe add support for PS4 controller too.... :)

kakra, 2020-05-09:
PR is a pull request, that means: Try to implement that feature and submit it as a PR here so it may be considered for inclusion.

But TBH, I don't think xpadneo can work as a flatpak package. After all, flatpak is about isolation, it won't allow a package or application to inject kernel modules. In that case, `xow` may work better: It's a driver implemented in user space.

kakra, 2020-06-11:
I'm closing this: Flatpaks are isolated applications with reduces privileges. They cannot contain kernel drivers, nor could they inject drivers into the kernel. That's against all design decisions that flatpak was made for.

If EndlessOS wants to include the driver, it has to do so as a non-flatpak package or as part of the kernel. `xpadneo` can be patched into the kernel source relatively easy, so it could be carried as a patch.

closed 2020-06-11 by kakra