atar-axis/xpadneo #158: System goes to sleep without using keyboard/mouse just using Xbox One controller

ipkpjersi, 2020-01-20:
**Describe the bug**
When I use the Xbox One controller, it does not count as an input. If I don't move my mouse or use my keyboard, my PC automatically goes to the screensaver and then goes to sleep as if the Xbox One controller was not a device at all. If I use my mouse or keyboard whilst I am also playing a game my PC does not go to screensaver and then sleep.

**To Reproduce**
Use Ubuntu 18.04 LTS with the Xfce DE and Xscreensaver 5.36:
[image]

**Expected behavior**
I expect using the Xbox One controller to count as input and not let the system go to screensaver and then sleep.

If you require any additional details please let me know.

atar-axis, 2020-02-20:
That's a good point - will add that as soon as I can (I fear it will take some time, too busy at the moment).

ipkpjersi, 2020-05-03:
Is there any additional information or debugging I can provide that would help with this task or do you generally know exactly how it would work?

kakra, 2020-05-09:
Usually, games should inhibit the screensaver themselves while in-game. What types of games are affected? Proton games should work afair: Their Wine version should have a patch that inhibits the screensaver while a fullscreen game is running.

The problem may be that the game controller does not count as a keyboard or mouse, so it may be ignored by Xorg. Or, other option: The game controller does not generate events of type `EV_KEY` and thus it's ignored. Could you try v0.7? It has remapped the Xbox logo button to a keyboard event. Please check if you can stop the screensaver by pressing that button after the screensaver kicked in. If that works, we may synthesize some key events every once in a while.

ipkpjersi, 2020-05-09:
It did still go to sleep with 0.7, and pressing the xbox logo button (and I think even just the start button) woke it up. I'm not sure if phantom/synthetic key events would work - would that get detected as "game input" or is it just done at the OS-level? I'm not really sure how this works.

kakra, 2020-05-09:
Okay, so all we need is sending a keyboard event of type `EV_KEY` once in a while. That would actually generate a real keypress so we should find a key that usually does nothing in Linux or the usual desktop environments. Here's a list of available codes we could synthesize:

https://elixir.bootlin.com/linux/latest/source/include/uapi/linux/input-event-codes.h#L177

(every `KEY_*` entry could work)

ipkpjersi, 2020-05-09:
Just tested this with 0.6 again, the guide button does not wake up the screen even though the other buttons still work and I can still hear the menu in the game. 0.7 is definitely an improvement and it's already halfway solved this issue.

I have no idea what a good key to send would be so I probably can't help with that, sadly.

kakra, 2020-05-09:
The guide button sent a game controller event in v0.6, that isn't picked up by anything on the desktop usually. Thus, I converted it to `KEY_HOME`. When your browser has focus, the guide button will switch to your homepage. If nothing else grabs this event, usually Steam will pick it up and show up. Pressing it a second time will then fire up its tenfoot launcher (Steam Big Picture mode). In theory, it should also open the Steam overlay while in a game. I didn't test it yet, and I'm not sure if `KEY_HOME` is really the correct event or if Steam could also work with another key event. I'm not completely happy with it because browsers can grab the key.

ipkpjersi, 2020-05-09:
I forgot to mention that regardless of how this is implemented, I feel like it should probably either be able to be toggled on or toggled off as a module parameter in case it doesn't work as expected or doesn't work for everyone etc.

kakra, 2020-06-10:
I'll keep the idea in mind... Maybe when I've got time to work on Proton again. But the driver is clearly the wrong place to implement this.

Until then, I suggest to use a script like this as a starting point (untested):
```bash
#!/bin/bash
: ${1?:needs a parameter to execute}
xset s off
"$@"
xset s on
```

Put it in `/usr/local/bin/inhibit-screensaver` and then add it to your games launch options in Steam:
```
inhibit-screensaver %command%
```

ipkpjersi, 2020-06-24:
I tried xset which didn't work but I did find a couple other good solutions. :+1: 

I tried `xset` with a bunch of params, unfortunately none of them worked (either all in the same script, or only with one of them one at a time):
```bash
#!/bin/bash
xset s off 
xset s noblank
xset s -dpms
"$@"
```

However, I did find a couple bash applications/scripts that both seemed to do the trick so hopefully mentioning them here may help anyone else who runs into this issue. The first one I tried was `joystickwake`, a Python3 script which you can use to send commands (to prevent screensavers) when joystick activity is detected: 
```
joystickwake [15:15:11] DEBUG js0 activity
joystickwake [15:15:11] INFO waking the screen
joystickwake [15:15:11] DEBUG X settings waker running command: xset dpms force on s reset
joystickwake [15:15:11] DEBUG XScreenSaver waker running command: xscreensaver-command -deactivate
joystickwake [15:15:11] DEBUG custom waker running command: xscreensaver-command -deactivate
joystickwake [15:15:11] DEBUG X settings waker succeeded
joystickwake [15:15:11] DEBUG js0 activity
xscreensaver-command: not active: idle timer reset.
joystickwake [15:15:11] DEBUG XScreenSaver waker succeeded
```

The second one is GameMode by the awesome Linux-friendly company Feral Interactive, it can disable the screensaver outright when running via Steam launch options `gamemoderun %command%`

Hopefully this information helps anyone else who runs into this issue. This is a huge improvement for me.

kakra, 2020-06-24:
Thanks for the heads-up.

Yes, I know `gamemode`, I contributed the Proton/Wine detection routines to it and some other stuff. The Wine detection is still a PR. It should usually work for you but maybe you're using something that invokes `xscreensaver` by other means than Xorg itself?

Apparently, we cannot run user space programs like `xscreensaver-command` from within the driver. But your solution may be useful in the docs. I wonder what the performance cost of this is, tho. Does it run the commands on each joystick activity? That could be quite a lot in a short time...

ipkpjersi, 2020-06-24:
For `joystickwake` it only issues the `screensaver-command` and/or similar commands every X number of seconds, I think the default is 30 seconds, but you can pass an interval parameter to it to specify how often you want it to run. Doesn't seem to affect performance and seems to work really well, plus it uses Python 3 instead of Python 2 so that's nice for future-proofing/forwards compatibility.

kakra, 2020-06-24:
If you'd like to have that added to the documentation, feel free to open a PR.