atar-axis/xpadneo #159: Bluetooth : impossible to connect more than 2 gamepad

Norde, 2020-01-22:
Hello atar-axis and thanks for this great driver :)

I've encounter some weird bug :
- It's almost impossible to connect more than 2 gamepad in bluetooth.
I've got 4 Xbox one S gamepad (rev 1708) and in most of the case, only two can be connected.

Sometimes, after some reboot, the 3rd and 4th gamepad connect (about 1min after the first two, with disconnection and reconnection of one of the two first gamepad)

I can't determine if this is a bug related to xpadneo or bluetooth
Some suggestions are welcome :blush:

### **System information**
Hardware : raspberry pi 3B and 3B+
Software : Ubuntu 18.04 and 19.10, raspbian strech and buster
Kernel : all kernel from 4.14 and 4.19 (v7 and v8, ie: 32 and 64bits)
Bluez : 5.43 and 5.50

### **Debuggin**
Debugging output of xpadneo is enabled
Bug appear on fresh install with only git, dkms, kernel headers and xpadneo installed.

**Nothing appears in dmseg except the result of the connection of the first two controllers:**

> [  +0.000567] hid-generic 0005:045E:02FD.0009: input,hidraw2: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on b8:27:eb:c8:5f:87
> [  +0.013499] xpadneo udev: 0005:045E:02FD.0009
> [  +0.152942] xpadneo 0005:045E:02FD.0009: unknown main item tag 0x0
> [  +0.000385] input: Xbox Wireless Controller as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.0/bluetooth/hci0/hci0:69/0005:045E:02FD.0009/input/input15
> [  +0.000625] xpadneo 0005:045E:02FD.0009: input,hidraw2: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on b8:27:eb:c8:5f:87
> [  +0.738756] NET: Registered protocol family 38
> [  +0.252638] xpadneo udev: ok
> [  +1.346004] hid-generic 0005:045E:02FD.000A: unknown main item tag 0x0
> [  +0.000802] input: Xbox Wireless Controller 2 as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.0/bluetooth/hci0/hci0:71/0005:045E:02FD.000A/input/input16
> [  +0.000762] hid-generic 0005:045E:02FD.000A: input,hidraw3: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller 2] on b8:27:eb:c8:5f:87
> [  +0.012070] xpadneo udev: 0005:045E:02FD.000A
> [  +0.157611] xpadneo 0005:045E:02FD.000A: unknown main item tag 0x0
> [  +0.000393] input: Xbox Wireless Controller 2 as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.0/bluetooth/hci0/hci0:71/0005:045E:02FD.000A/input/input17
> [  +0.000546] xpadneo 0005:045E:02FD.000A: input,hidraw3: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller 2] on b8:27:eb:c8:5f:87
> [  +0.990768] xpadneo udev: ok

**Btmon return some failed connection:** 

> = bluetoothd: Can't get HIDP connection info           19.136944
> < HCI Command: Create Connection (0x01|0x0005) plen 13           [hci0] 19.136870
>         Address: 5C:BA:37:63:B0:C1 (Microsoft Corporation)
>         Packet type: 0xcc18
>           DM1 may be used
>           DH1 may be used
>           DM3 may be used
>           DH3 may be used
>           DM5 may be used
>           DH5 may be used
>         Page scan repetition mode: R2 (0x02)
>         Page scan mode: Mandatory (0x00)
>         Clock offset: 0x0000
>         Role switch: Allow slave (0x01)
> '> HCI Event: Command Status (0x0f) plen 4           [hci0] 19.148318
>       Create Connection (0x01|0x0005) ncmd 1
>         Status: Command Disallowed (0x0c)
> '> ACL Data RX: Handle 69 flags 0x02 dlen 7           [hci0] 23.643748
>       Channel: 65 len 3 [PSM 0 mode 0] {chan 0}
>         a1 04 8a                                         ...
> '> HCI Event: Connect Complete (0x03) plen 11           [hci0] 24.264386
>         Status: Page Timeout (0x04)
>         Handle: 0
>         Address: 5C:BA:37:63:B0:C1 (Microsoft Corporation)
>         Link type: ACL (0x01)
>         Encryption: Disabled (0x00)
> @ MGMT Event: Connect Failed (0x000d) plen 8           {0x0002} [hci0] 24.264451
>         BR/EDR Address: 5C:BA:37:63:B0:C1 (Microsoft Corporation)
>         Status: Connect Failed (0x04)
> @ MGMT Event: Connect Failed (0x000d) plen 8           {0x0001} [hci0] 24.264451
>         BR/EDR Address: 5C:BA:37:63:B0:C1 (Microsoft Corporation)
>         Status: Connect Failed (0x04)

**And journalctl return "can't get HIDP connection info" for bluetooth:**

> Jan 22 18:47:45 raspberrypi kernel: hid-generic 0005:045E:02FD.0009: unknown main item tag 0x0
> Jan 22 18:47:45 raspberrypi kernel: input: Xbox Wireless Controller as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.0/bluetooth/hci0/hcJan 22 18:47:45 raspberrypi kernel: hid-generic 0005:045E:02FD.0009: input,hidraw2: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on b8:27:eb:c8:5f:8Jan 22 18:47:45 raspberrypi unknown: xpadneo udev: 0005:045E:02FD.0009
> Jan 22 18:47:45 raspberrypi kernel: xpadneo 0005:045E:02FD.0009: unknown main item tag 0x0
> Jan 22 18:47:45 raspberrypi kernel: input: Xbox Wireless Controller as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.0/bluetooth/hci0/hcJan 22 18:47:45 raspberrypi kernel: xpadneo 0005:045E:02FD.0009: input,hidraw2: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on b8:27:eb:c8:5f:87
> Jan 22 18:47:46 raspberrypi sudo[898]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/btmon
> Jan 22 18:47:46 raspberrypi sudo[898]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
> Jan 22 18:47:46 raspberrypi kernel: NET: Registered protocol family 38
> Jan 22 18:47:46 raspberrypi unknown: xpadneo udev: ok
> Jan 22 18:47:47 raspberrypi systemd-udevd[907]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
> Jan 22 18:47:47 raspberrypi systemd-udevd[904]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
> Jan 22 18:47:48 raspberrypi kernel: hid-generic 0005:045E:02FD.000A: unknown main item tag 0x0
> Jan 22 18:47:48 raspberrypi kernel: input: Xbox Wireless Controller 2 as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.0/bluetooth/hci0/Jan 22 18:47:48 raspberrypi kernel: hid-generic 0005:045E:02FD.000A: input,hidraw3: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller 2] on b8:27:eb:c8:5fJan 22 18:47:48 raspberrypi unknown: xpadneo udev: 0005:045E:02FD.000A
> Jan 22 18:47:48 raspberrypi kernel: xpadneo 0005:045E:02FD.000A: unknown main item tag 0x0
> Jan 22 18:47:48 raspberrypi kernel: input: Xbox Wireless Controller 2 as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.0/bluetooth/hci0/Jan 22 18:47:48 raspberrypi kernel: xpadneo 0005:045E:02FD.000A: input,hidraw3: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller 2] on b8:27:eb:c8:5f:87 Jan 22 18:47:49 raspberrypi unknown: xpadneo udev: ok
> Jan 22 18:47:49 raspberrypi systemd-udevd[933]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
> Jan 22 18:47:49 raspberrypi systemd-udevd[928]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
> Jan 22 18:48:05 raspberrypi bluetoothd[333]: Can't get HIDP connection info
> Jan 22 18:48:05 raspberrypi bluetoothd[333]: Can't get HIDP connection info
> Jan 22 18:48:10 raspberrypi bluetoothd[333]: connect error: Host is down (112)
> Jan 22 18:48:15 raspberrypi bluetoothd[333]: connect error: Host is down (112)

deathxxx123, 2020-01-25:
Maybe you'r bluetooth device is too slow? Maybe for that i'n buying BT 5 on PCI-E.
The controller itself is using BT 4.2 hardware.

Norde, 2020-01-27:
No it's not that, the bandwidth is more than enough for 4 controllers (and everything works fine when the 4 controllers manage to be connected at the same time).

atar-axis, 2020-02-20:
I am sorry but I cannot reproduce that here - I have currently connected three gamepads :)
[image]

closed 2020-02-20 by atar-axis

Norde, 2020-02-20:
@atar-axis What OS and hardware do you use ?

It seems to be a bluetooth driver issue then :)

atar-axis, 2020-02-20:
I am under Arch Linux on my R3 Acer Laptop, I don't know the chipset at the moment - if you need it I will look it up for you

Norde, 2020-02-20:
Wich rev and kernel ?

I don't think it's hardware related (at least not specific to a chipset) since this bug occurs with internal bluetooth of Rpi and also with two usb bluetooth dongle (with two different bluetooth chip).

Norde, 2020-02-21:
I am also interested in the reference of your bluetooth chipset if you have time to look :)