atar-axis/xpadneo #162: Xbox one s controller no guide button funtionality after firmware update

blastermaster77, 2020-02-24:
**Describe the bug**
The guide button no longer works after updatating the firmware from a win10pc, also it reconize it with usb but buttons dont work.

**To Reproduce**
Steps to reproduce the behavior:
Update firmware with win10 upate app
**Expected behavior**
guide button and usb should work

**Screenshots/Gifs**
If applicable, add screenshots or animated gifs to help explain your problem.

**System information**
Please enable debugging output <https://atar-axis.github.io/xpadneo/index.html#debugging>
and add at least the following outputs
-   `Linux blastermaster-X570-GAMING-X 5.3.0-41-generic #33-Ubuntu SMP Fri Feb 14 17:26:47 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

-   `[Xbox Wireless Controller] on 00:1a:7d:da:71:11
[   54.643847] xpadneo udev: 0005:045E:02FD.000A
[   54.647914] xpadneo: hello there!
[   54.766247] xpadneo 0005:045E:02FD.000A: unknown main item tag 0x0
[   54.766301] input: Xbox Wireless Controller as /devices/pci0000:00/0000:00:08.1/0000:09:00.3/usb5/5-3/5-3.4/5-3.4:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000A/input/input25
[   54.766397] xpadneo 0005:045E:02FD.000A: input,hidraw9: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 00:1a:7d:da:71:11
[   55.749687] xpadneo udev: ok
[   85.922309] input: Microsoft X-Box 360 pad as /devices/virtual/input/input26
[  213.440928] input: Microsoft X-Box 360 pad as /devices/virtual/input/input27
`

**Additional context**  
Add any other context about the problem here.

atar-axis, 2020-02-25:
Please add the output of `sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;`

blastermaster77, 2020-02-28:
I get nothing when I run that command

> Please add the output of `sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;`

I get nothing when I run that command.

blastermaster77, 2020-02-28:
[image]

bserem, 2020-04-06:
I just installed xpadneo and the guide button is not working for me too. The button numbering on steam is being kept, but `Button 8` is skipped and can't be registered when I press it.

Here is the output for the command you asked for:
```
$ sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;
/sys/kernel/debug/hid/0005:045E:02E0.0004
05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 32 09 35 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 02 09 c5 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 02 09 c4 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0f 15 00 25 01 75 01 95 0f 81 02 15 00 25 00 75 01 95 01 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 05 06 09 20 85 04 15 00 26 ff 00 75 08 95 01 81 02 c0 00 
3745764076 1006

/sys/kernel/debug/hid/0005:045E:02FD.0003
05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 32 09 35 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 02 09 c5 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 02 09 c4 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0f 15 00 25 01 75 01 95 0f 81 02 15 00 25 00 75 01 95 01 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 05 06 09 20 85 04 15 00 26 ff 00 75 08 95 01 81 02 c0 00 
3745764076 1006
```
Note: I have two (2) controllers connected. Model number 1708 (Wirelerss controller for Xbox One).

Probably related: the battery is showing 55% to both of them, which I really doubt is true.
I've found a WIn10 pc and updated my devices around the beginning of March.

Hope I helped.

Rixasha, 2020-04-15:
The Guide button isn't working for me either with an Xbox One S controller with the newest firmware on offer, 4.8.1923.0. Here's my output, the bytes are the same as above:

    /sys/kernel/debug/hid/0005:045E:02FD.0021
    05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 32 09 35 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 02 09 c5 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 02 09 c4 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0f 15 00 25 01 75 01 95 0f 81 02 15 00 25 00 75 01 95 01 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 05 06 09 20 85 04 15 00 26 ff 00 75 08 95 01 81 02 c0 00 
    3745764076 1006

Considering the charge mentioned in the previous post, mine was also showing as 55% for hours and hours until eventually and rather quickly dropping to 10%. I didn't think this too odd, it's pretty well in keeping with how it shows on the Xbox with rechargeable NiMH batteries.

bserem, 2020-04-16:
Didn't know about the charge reporting issue with NiMH batteries. Thanks!

kakra, 2020-05-06:
@bserem Could you try my PR #178 with both of your controllers? It would be helpful as it seems they are both in different connection modes. I hope that's still the case. With my PR, both modes should be visible as `0x2E0` to applications. `dmesg` should tell you for one controller that it pretends a different connection mode. All button mappings should be fixed then once and for all.

kakra, 2020-05-08:
The issue may still occur and I think I may have found what's causing it: After firmware upgrade, when connecting to Bluetooth for the first time, the controller shows itself as firmware 4.08, the buttons are and axes may be messed up then (although my current PR #178 mostly fixes it except the guide button and the thumb buttons). If you reconnect the controller (by re-pairing it), it probably shows up as firmware 9.03 - now the buttons all work correctly with my PR. So we are probably back to the old SDL problem where user space tries to fix-up mappings. I'm having an idea how to fix it but I'm not sure how to force my controller back into 4.08 firmware announce for testing.

First connect (v4.08):
```
[51453.240811] xpadneo 0005:045E:02E0.001C: fixing up report size
[51453.240813] xpadneo 0005:045E:02E0.001C: fixing up Rx axis
[51453.240815] xpadneo 0005:045E:02E0.001C: fixing up Ry axis
[51453.240816] xpadneo 0005:045E:02E0.001C: fixing up Z axis
[51453.240817] xpadneo 0005:045E:02E0.001C: fixing up Rz axis
[51453.240818] xpadneo 0005:045E:02E0.001C: fixing up button mapping
[51453.241139] input: Xbox Wireless Controller as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2.7/2-1.2.7.2/2-1.2.7.2:1.0/bluetooth/hci0/hci0:75/0005:045E:02E0.001C/input/input25
[51453.241240] xpadneo 0005:045E:02E0.001C: input,hidraw13: BLUETOOTH HID v4.08 Gamepad [Xbox Wireless Controller] on 00:1a:7d:da:71:15
[51454.234270] xpadneo 0005:045E:02E0.001C: battery registered
```

vs. after Bluetooth re-pairing:
```
[56408.457029] xpadneo 0005:045E:02FD.001D: fixing up report size
[56408.457032] xpadneo 0005:045E:02FD.001D: fixing up Rx axis
[56408.457033] xpadneo 0005:045E:02FD.001D: fixing up Ry axis
[56408.457034] xpadneo 0005:045E:02FD.001D: fixing up Z axis
[56408.457036] xpadneo 0005:045E:02FD.001D: fixing up Rz axis
[56408.457038] xpadneo 0005:045E:02FD.001D: fixing up button mapping
[56408.457410] xpadneo 0005:045E:02FD.001D: pretending XB1S Windows wireless mode (changed PID from 0x02FD to 0x02E0)
[56408.457477] input: Xbox Wireless Controller as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2.7/2-1.2.7.2/2-1.2.7.2:1.0/bluetooth/hci0/hci0:72/0005:045E:02FD.001D/input/input26
[56408.457656] xpadneo 0005:045E:02FD.001D: input,hidraw13: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 00:1a:7d:da:71:15
[56409.470039] xpadneo 0005:045E:02FD.001D: battery registered
```

So all we need to do is probably faking v9.03 if we detect the controller with PID 0x2E0.

closed 2020-05-08 by kakra

bserem, 2020-05-08:
You are way faster than me! I just did a first test, they both connect but I can only register buttons with one of them (testing on steam). The guide button does work.

I might have to remove both of them and re-pair them. I'll report back if there are any issues.
Thanks

kakra, 2020-05-08:
Before you re-pair them, please let us do some tests... I think you have them both in different connection modes. Please post the output of recent lines from `dmesg` for each of them. You can use `dmesg -w` to get a live log while you do your tests.