atar-axis/xpadneo #163: Modification to 99-xpadneo-bluetooth.conf will be reset by dkms post install script

terencode, 2020-02-25:
**Problem*
I need to set permanent settings to `99-xpadneo-bluetooth.conf` like `disable_ff` and `fake_dev_version` due to me using a 8bitdo controller, however, this file gets overwritten by the post-install dkms script everytime I update the kernel.

**Possible solution**
Check if the file is already created and do not touch it if it's the case.

**Alternatives**
I'm not aware of an other way to set parameters for out-of-tree kernel modules.

**Additional context**
Distro: ArchLinux, xpadneo installed via https://aur.archlinux.org/packages/xpadneo-dkms-git/

kakra, 2020-05-09:
Please send the device descriptor and PID/VID/version info of the controller.

terencode, 2020-05-09:
lsusb reports `Bus 001 Device 016: ID 045e:028e Microsoft Corp. Xbox360 Controller`

I don't know how to get the device descriptor.

kakra, 2020-05-09:
When the device is bound to a driver, you can dump it like this:
```
xxd -c20 -g1 /sys/module/<MODULENAME>/drivers/<DRIVERNAME>/<BUS:VEN:DEV:NUM>/report_descriptor
```
where you replace the CAPITALS with whatever you find there.

kakra, 2020-05-09:
This is a USB device not wireless, correct?

terencode, 2020-05-09:
>This is a USB device not wireless, correct?
It can do both. Do you need the same in bluetooth mode?

Is the updated comment above enough? 
I navigated to `/sys/module/xpad/drivers/usb:xpad/1-11.1.1:1.0` but I do not see a file named report_descriptor:

```
authorized  bAlternateSetting  bInterfaceClass  bInterfaceNumber  bInterfaceProtocol  bInterfaceSubClass  bNumEndpoints  driver  ep_01  ep_81  input  modalias  power  subsystem  supports_autosuspend  uevent
```

kakra, 2020-05-09:
We'd need the HID version of the connection, that's a different protocol embedded into the USB protocol. If your controller doesn't support HID, xpadneo cannot currently support the device. So in `/sys/module/xpad/drivers` there should be an entry starting with `hid:` - if not, you could try binding the device to a different module. Currently, your device does not use xpadneo so I wonder why you're having issues with the `99-xpadneo-bluetooth.conf` file at all.

This all may change in Bluetooth mode, so, yes please, try that. The verbose output of `lsusb` isn't needed, Bluetooth devices won't show up there (only the dongle itself and that's uninteresting to `xpadneo`).

kakra, 2020-05-09:
> I navigated to `/sys/module/xpad/drivers`

BTW: `xpad` is not `xpadneo` - it's a completely unrelated and different driver.

terencode, 2020-05-09:
> BTW: xpad is not xpadneo - it's a completely unrelated and different driver.

Ah lol that's why...

Here you go :)

```
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0a 15 00 25 01 75 01 95 0a 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 06 95 01 81 03 05 01 09 80 85 02 a1 00 09 85 15 00 25 01  u.................%.
000000b4: 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09  ..u.....%.u.........
000000c8: 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25  !........%.u.......%
000000dc: 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09  .u......p..%du......
000000f0: 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15  Pf..U...&..u........
00000104: 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff  .&..u.....e.U..|..&.
00000118: 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75  .u........... ..&..u
0000012c: 08 95 01 81 02 c0                                            ......
```

kakra, 2020-05-09:
Could you append the recent lines from `dmesg` that resulted from the connection made by xpadneo?

terencode, 2020-05-09:
```
[  +9,693150] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[  +0,000003] Bluetooth: HIDP socket layer initialized
[  +0,000457] hid-generic 0005:045E:02E0.000B: unknown main item tag 0x0
[  +0,000089] input: 8BitDo SN30 Pro+ as /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.000B/input/input36
[  +0,000535] hid-generic 0005:045E:02E0.000B: input,hidraw2: BLUETOOTH HID v9.03 Gamepad [8BitDo SN30 Pro+] on d4:d2:52:a9:4f:d3
[  +0,001451] xpadneo udev: 0005:045E:02E0.000B
[  +0,002721] hid_xpadneo: unknown parameter 'fake_dev_version' ignored
[  +0,028503] xpadneo 0005:045E:02E0.000B: fixing up report size
[  +0,000255] xpadneo 0005:045E:02E0.000B: battery registered
[  +0,000003] xpadneo 0005:045E:02E0.000B: pretending XB1S Linux firmware version (changed version from 0x00000903 to 0x00000903)
[  +0,000036] input: 8BitDo SN30 Pro+ as /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.000B/input/input37
[  +0,000084] xpadneo 0005:045E:02E0.000B: input,hidraw2: BLUETOOTH HID v9.03 Gamepad [8BitDo SN30 Pro+] on d4:d2:52:a9:4f:d3
[  +0,000381] xpadneo udev: ok
```

kakra, 2020-05-09:
Ok thanks, with this information I could natively embed support for this into xpadneo. You shouldn't need to take care of the `99-xpadneo-bluetooth.conf` file then: It will be installed and your system will use it correctly for this controller.

Would that be how you'd expect this to work?

terencode, 2020-05-09:
Well I would think the `options bluetooth disable_ertm=y` applies globally and there might be drawbacks to enabling it, hence why it's not there by default. 

I tested it without this option and it seems to work fine, so I thought about having a way to not have this option when the device is not an official controller.

kakra, 2020-05-09:
Ah okay... I also let ERTM enabled despite having the genuine controller. It connects just fine when using a kernel patch that ignores an incompatible Bluetooth L2CAP negotiation with the device. But such a patch apparently cannot be deployed with dkms.

terencode, 2020-05-09:
With dkms you can override already existing modules, for example I patch the i2c-piix4 module to allow changing the RGB of my motherboard: https://aur.archlinux.org/packages/i2c-piix4-aura-dkms/

It shows up like this: 
`i2c-piix4-aura, 5.6.7, 5.6.11-arch1-1, x86_64: installed (original_module exists)`

Maybe it's too complicated to do it with your use-case though.

kakra, 2020-05-09:
I'll create an option do prevent installing the file by placing a `/dev/null` symlink. The descriptor perfectly matches the genuine controller in Windows mode, so we'd need no special support for it. It even has the same descriptor bugs (like having a bogus NUL byte at the end).

BTW: `fake_dev_version` is no longer supported as an option. Do you still have any axis mapping problems?

kakra, 2020-05-09:
Could you try my patch? It allows you to place a symlink to `/dev/null` instead of the config file before doing the dkms install. If you didn't place the symlink, it will log a command to use so you can prevent the installation in the future.

terencode, 2020-05-09:
> BTW: fake_dev_version is no longer supported as an option. Do you still have any axis mapping problems?

I think I never had problems, just some left over. Working fine without it.

>Could you try my patch?

Seems to work but when uninstalling you also need to change the post_remove script so that it doesn't remove the conf file if it is symlinked to `/dev/null`.

kakra, 2020-05-09:
> Seems to work but when uninstalling you also need to change the post_remove script so that it doesn't remove the conf file if it is symlinked to /dev/null.

Oh, how true you are...

kakra, 2020-05-09:
Please test again...

terencode, 2020-05-09:
All seems good now!

closed 2020-05-09 by kakra

terencode, 2020-05-11:
I don't know if it's worth to open a new issue but even though the gamepad is correctly handled by xpadneo, libinput seems to still have something to say about it: `(EE) libinput bug: Event for missing capability CAP_POINTER on device "8BitDo SN30 Pro+"` even though it clearly isn't handling this device.
This happens each time I press the B/A/Y/X keys.
Can you do something about it?

kakra, 2020-05-12:
I think this should become a new issue, this one was solely about the ERTM override. In the new issue, please let us know where you're seeing the log including some more context. It doesn't seem to be a kernel message, `Xorg.log` probably?

PR #185 (2020-05-09): Add support for disabling the ERTM override
Closes #163