atar-axis/xpadneo #169: sudo bluetoothctl

ModernLobby, 2020-03-19:
**Describe the bug**
After installing xpadneo and try to type sudo bluetoothctl on Pop OS nothing happens after that

**To Reproduce**
Pop Os install Xpadneo and type cd xpadneo and after the install done type sudo bluetoothctl

**Expected behavior**
sudo bluetoothctl
[bluetooth]# scan on

**Screenshots/Gifs**
https://imgur.com/a/LBOyKdv

and i tried with xpadneo with this command same results

**System information**

**Additional context**  
I got the Xbox One Controller with the USB dongle supplied with.

currybullen, 2020-04-01:
If you're using the official Microsoft dongle you are looking for a different driver. Have a look at this project: https://github.com/medusalix/xow

PetrusZ, 2020-04-14:
You should start bluetooth.service. `sudo systemctl start bluetooth.service`

Anthony6444, 2020-04-16:
after typing `sudo bluetoothctl`:
enter your account password, then type `scan on` at the `[bluetooth]#` prompt

kakra, 2020-05-09:
This looks like an issue of the distribution. If `bluetoothctl` doesn't show a prompt, that's completely outside of the scope of this project. Try asking the Pop Os folks first, this seems unrelated to xpadneo.

kakra, 2020-06-01:
Moving to wifi project support... Xpadneo currently only supports third-party Bluetooth dongles.

closed 2024-04-03 by github-actions[bot]