atar-axis/xpadneo #172: custom button mappings

toby63, 2020-04-02:
**Problem:**
(Some) Games don't allow for changing the buttons, so if I for example would like to use Button A for a different command, this is not possible.

**Solution:**
Custom button mappings with xpadneo.
Change the buttons, so that A is for example recognized as X.
Because Games are different, different profiles would be convenient.

**Alternatives:**
1. Software like sc-controller [github repo](https://github.com/kozec/sc-controller) offers something like that, but does not function right now (at least for me).
2. Am I right assuming that i could change the source code of xpadneo to achieve this?

kakra, 2020-05-03:
The problem with this is that we cannot currently remap buttons at runtime of the driver, this is baked into the loading phase which injects the mapping corrections into the HID layer of the kernel. So your point of 2 may contain a pitfall.

If we wanted to support different button mapping profiles, this must be done outside of the initialization phase in `xpadneo_event()` hook. But that's not ideal because we would discard the original button event and inject a newly generated. The behavior fixup of this already does this and has a TODO of not doing it there. May instead fixup the raw report itself? Most other drivers in the kernel seem to prefer that way.

toby63, 2020-05-03:
@kakra Thx for the answer.

kakra, 2020-05-04:
My current branch #178 should make this much easier for you: It will simplify the whole process of remapping by patching the controller protocol into a single format for all connection modes. Now it's a matter of adding or changing a few lines to remap buttons.

But during testing, I figured that the Linux HID layer may not be too happy about it: It is really important to always map exactly 10 controller buttons or it will mess up the layout of the second last button. So, the `BTN_*` must not be changed to something else except swapping with each other, and the `KEY_*` codes must not be changed at all.

There's two opportunities now with #178 to swap the layout: Either you mess around with the bit masks in `xpadneo_raw_event()` (which is complicated but works without games seeing what you did) or you change the mapping in the struct `xpadneo_usage_map` (which may be visible to games and they may fix it back). In the latter case, the order doesn't matter, only the `0x900..` to `BTN_*` mapping.

With some programming skills this could be converted into a runtime module parameter.

toby63, 2020-05-05:
> My current branch #178 should make this much easier for you [...]

Sounds good.
Thank you for your effort :+1:.

kakra, 2020-05-06:
I've updates my PR to account for the different report sizes of the XBE2 controllers... Otherwise you won't be able to remap buttons at that part of the code.

kakra, 2020-07-07:
This feature will be queued for version 0.9 which will include a remapper and a profile switcher for 4 different profiles. Profile switching will be possible at anytime by pressing a button combination on the controller.