atar-axis/xpadneo #174: udev rules do not seem to work

PetrusZ, 2020-04-14:
**Describe the bug**
After connecting via bluetoothctl, the controller does not at work.

**To Reproduce**
git clone https://github.com/atar-axis/xpadneo.git
cd xpadneo
sudo ./install.sh

bluetoothctl
[bluetooth]# scan on
[bluetooth]# pair <MAC>
[bluetooth]# trust <MAC>
[bluetooth]# connect <MAC>

**Expected behavior**
After connecting, controller works.

**System information**
Can't enable debugging, because I don't have /sys/module/hid_xpadneo dir.
-   `uname -a` Linux Avalon-PC 5.6.0-gentoo-x86_64 #6 SMP Tue Apr 14 01:27:54 CST 2020 x86_64 Intel(R) Xeon(R) CPU E3-1231 v3 @ 3.40GHz GenuineIntel GNU/Linux

-   `dmesg`

```
[  432.129781] hid-generic 0005:045E:02FD.0005: unknown main item tag 0x0
[  432.129964] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:02FD.0005/input/input22
[  432.130141] input: Xbox Wireless Controller Consumer Control as /devices/virtual/misc/uhid/0005:045E:02FD.0005/input/input23
[  432.130205] hid-generic 0005:045E:02FD.0005: input,hidraw4: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 24:41:8c:13:8f:a2
[  432.133074] xpadneo udev: 0005:045E:02FD.0005
[  432.174688] xpadneo udev: ok

```

-   `sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;`
```
/sys/kernel/debug/hid/0005:045E:02FD.0005
05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 32 09 35 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 02 09 c5 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 02 09 c4 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0f 15 00 25 01 75 01 95 0f 81 02 15 00 25 00 75 01 95 01 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75 08 95 01 81 02 c0 00
2579218694 1006
```

**Additional context**  
Using systemd. `udevadm control --reload && udevadm trigger` doesn't also work.

➜ modinfo hid_xpadneo
modinfo: ERROR: Module hid_xpadneo not found.

➜ dkms status
hid-xpadneo, 0.6.0, 5.6.0-gentoo-x86_64, x86_64: installed (WARNING! Diff between built and installed module!)

kakra, 2020-05-09:
`hid-generic` has already bound to the controller, it should say `xpadneo` in that line instead. I'll come up with an idea how to diagnose it, maybe some udev rule you're using is taking higher priority.

kakra, 2020-05-09:
DKMS does not work in Gentoo... Please use `make -C hid-xpadneo install` instead.

PetrusZ, 2020-05-17:
```
➜ sudo make -C hid-xpadneo install
Alias tip: _ make -C hid-xpadneo install
make: Entering directory '/home/petrus/Project/xpadneo/hid-xpadneo'
make: *** No rule to make target 'install'.  Stop.
make: Leaving directory '/home/petrus/Project/xpadneo/hid-xpadneo'
```

```
➜ sudo make -C hid-xpadneo modules_install
Alias tip: _ make -C hid-xpadneo modules_install
make: Entering directory '/home/petrus/Project/xpadneo/hid-xpadneo'
make -C /usr/src/linux INSTALL_MOD_DIR="kernel/drivers/hid" LD=ld.bfd M=/home/petrus/Project/xpadneo/hid-xpadneo/src modules_install
make[1]: Entering directory '/usr/src/linux-5.6.7-gentoo'
cat: /home/petrus/Project/xpadneo/hid-xpadneo/src/modules.order: No such file or directory
  DEPMOD  5.6.7-gentoo-x86_64
make[1]: Leaving directory '/usr/src/linux-5.6.7-gentoo'
make: Leaving directory '/home/petrus/Project/xpadneo/hid-xpadneo'
```

kakra, 2020-05-17:
Oh yes, my fault. It should be `modules modules_install`. Sorry for the confusion. I'll edit my post above.

PetrusZ, 2020-05-18:
After `make -C hid-xpadneo modules modules_install`,

```
➜ sudo make -C hid-xpadneo modules modules_install
Alias tip: _ make -C hid-xpadneo modules modules_install
make: Entering directory '/home/petrus/Project/xpadneo/hid-xpadneo'
make -C /usr/src/linux INSTALL_MOD_DIR="kernel/drivers/hid" LD=ld.bfd M=/home/petrus/Project/xpadneo/hid-xpadneo/src modules
make[1]: Entering directory '/usr/src/linux-5.6.13-gentoo'
  CC [M]  /home/petrus/Project/xpadneo/hid-xpadneo/src/hid-xpadneo.o
  MODPOST 1 modules
  CC [M]  /home/petrus/Project/xpadneo/hid-xpadneo/src/hid-xpadneo.mod.o
  LD [M]  /home/petrus/Project/xpadneo/hid-xpadneo/src/hid-xpadneo.ko
make[1]: Leaving directory '/usr/src/linux-5.6.13-gentoo'
make -C /usr/src/linux INSTALL_MOD_DIR="kernel/drivers/hid" LD=ld.bfd M=/home/petrus/Project/xpadneo/hid-xpadneo/src modules_install
make[1]: Entering directory '/usr/src/linux-5.6.13-gentoo'
  INSTALL /home/petrus/Project/xpadneo/hid-xpadneo/src/hid-xpadneo.ko
  DEPMOD  5.6.13-gentoo-x86_64
make[1]: Leaving directory '/usr/src/linux-5.6.13-gentoo'
make: Leaving directory '/home/petrus/Project/xpadneo/hid-xpadneo'

➜ lsmod
Module                  Size  Used by
nvidia_drm             45056  -2
nvidia_modeset       1077248  -2
nvidia              19968000  -2

➜ sudo modprobe hid-xpadneo
Alias tip: _ modprobe hid-xpadneo
Password:

➜ lsmod
Module                  Size  Used by
hid_xpadneo            16384  -2
nvidia_drm             45056  -2
nvidia_modeset       1077248  -2
nvidia              19968000  -2
```

I can't find Xbox controller on scan:

```
➜ bluetoothctl
Agent registered
[CHG] Controller 24:41:8C:13:8F:A2 Pairable: yes
[bluetooth]# scan on
Discovery started
[CHG] Controller 24:41:8C:13:8F:A2 Discovering: yes
[CHG] Device 77:A1:A0:E9:4B:AD RSSI: -67
[CHG] Device 77:A1:A0:E9:4B:AD TxPower: 12
[CHG] Device 77:A1:A0:E9:4B:AD ManufacturerData Key: 0x004c
[CHG] Device 77:A1:A0:E9:4B:AD ManufacturerData Value:
  10 05 47 1c 93 d1 d7                             ..G....
[CHG] Device C0:9F:05:EA:D6:5B RSSI: -77
[CHG] Device EC:83:50:47:87:B8 RSSI: -77
[DEL] Device 77:A1:A0:E9:4B:AD 77-A1-A0-E9-4B-AD
[DEL] Device 6E:BB:7C:F1:D8:E5 6E-BB-7C-F1-D8-E5
[DEL] Device C0:9F:05:EA:D6:5B OPPO A59m
[CHG] Device EC:83:50:47:87:B8 RSSI: -68
[NEW] Device C0:9F:05:EA:D6:5B OPPO A59m
[CHG] Device C0:9F:05:EA:D6:5B UUIDs: 00001112-0000-1000-8000-00805f9b34fb
[CHG] Device C0:9F:05:EA:D6:5B UUIDs: 00001203-0000-1000-8000-00805f9b34fb
[CHG] Device C0:9F:05:EA:D6:5B UUIDs: 0000111f-0000-1000-8000-00805f9b34fb
[CHG] Device C0:9F:05:EA:D6:5B UUIDs: 00001116-0000-1000-8000-00805f9b34fb
[CHG] Device C0:9F:05:EA:D6:5B UUIDs: 00001115-0000-1000-8000-00805f9b34fb
[CHG] Device C0:9F:05:EA:D6:5B UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Device C0:9F:05:EA:D6:5B UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Device C0:9F:05:EA:D6:5B UUIDs: 0000110f-0000-1000-8000-00805f9b34fb
[CHG] Device C0:9F:05:EA:D6:5B UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[NEW] Device 77:A1:A0:E9:4B:AD 77-A1-A0-E9-4B-AD
[bluetooth]#
```

I give up xpadneo on gentoo. Thanks for your help.

closed 2020-05-18 by PetrusZ

kakra, 2020-05-18:
Actually, I'm using it on Gentoo and developing on Gentoo. It should work.