atar-axis/xpadneo #175: pcap captures

gabrielgbs97, 2020-04-19:
I do not own a xbox one s controller, but would like to get some pcap captures, or something like ASCII debug txt. I would like to emulate the xbox BT HID gamepad from a pc to another PC/Android as target platforms.

kakra, 2020-05-08:
I wonder how that would interact between PC and Android as Android clearly seems to expect its own special mapping.

Usually, from what I see, the device descriptor is pretty much okay: It maps keycode offsets directly and correctly to kernel event codes. But neither the legacy js interface nor SDL uses these offsets but instead expects hard-wired numbers for axis and key codes.

I'm not sure if dumping hidraw would reveal the original or the fixed up HID reports. But actually those reports are probably what you want to replay/proxy to the other box.

kakra, 2020-06-08:
There's a bunch of `btmon` logs in some of the recent issue reports, maybe you can work with those. It's not a pcap capture but ASCII text.

gabrielgbs97, 2020-06-09:
Thank you for the reference! I will try

closed 2020-06-09 by gabrielgbs97