atar-axis/xpadneo #177: Firmware 4: button mapping messed up

kakra, 2020-05-02:
Since upgrading the controller from firmware 3 to firmware 4 branch, the buttons are messed up again. It looks like the controller now sends a report descriptor identical to the Windows descriptor (same length, same bytes, I compared my dmesg output with the source code). Since they are supposed to have different lengths but are now identical, the heuristics in `check_report_behaviour()` no longer works and the buttons are messed up. Manually replacing the `case WINDOWS` to force `perform_mapping = map_hid_to_input_linux` corrects this problem and ABXY are fine again. But still, the following buttons no longer work and generate no input events:

Button "SELECT" (left of Xbox Logo button):
```
xpadneo: raw_event: 01 c3 81 98 84 43 80 ca 85 00 00 00 00 00 00 00 01
xpadneo: raw_event: 01 c3 81 98 84 43 80 ca 85 00 00 00 00 00 00 00 00
```

Button "MENU" (right of Xbox Logo button):
```
xpadneo: raw_event: 01 c3 81 98 84 43 80 ca 85 00 00 00 00 00 00 08 00
xpadneo: raw_event: 01 c3 81 98 84 43 80 ca 85 00 00 00 00 00 00 00 00
```

Button "GUIDE" (the Xbox Logo button):
```
xpadneo: raw_event: 01 c3 81 98 84 43 80 ca 85 00 00 00 00 00 00 10 00
xpadneo: raw_event: 01 c3 81 98 84 43 80 ca 85 00 00 00 00 00 00 00 00
```

I think we need a different heuristics or some manual override which mapping profile is used.

kakra, 2020-05-02:
Thumb button also no longer work:

Thumb left:
```
xpadneo: raw_event: 01 b1 88 e2 82 17 7d 96 82 00 00 00 00 00 00 20 00
xpadneo: raw_event: 01 b1 88 e2 82 17 7d 96 82 00 00 00 00 00 00 00 00
```

Thumb right:
```
xpadneo: raw_event: 01 6b 87 5a 84 43 73 ab 75 00 00 00 00 00 00 40 00
xpadneo: raw_event: 01 6b 87 5a 84 fe 74 54 76 00 00 00 00 00 00 00 00
```

kakra, 2020-05-02:
I think what we are facing now is a situation with
```
xdata->report_behaviour == LINUX && xdata->report_descriptor == WINDOWS
```

The report descriptor shows Windows mode upon binding the device but the reports come in as 17-byte reports like Linux mode does it. We could fix this in `padneo_event()` but the dead buttons won't show up there because they are set to ignored already during mapping setup.

kakra, 2020-05-02:
Okay, this firmware is messed up... That's probably why the letters MS can be found in "messed", so it's MS'ed up. ;-)

I completely removed the device from `bluetoothctl` now to start over, with `untrust` and `remove` so it doesn't auto pair again. I restarted the controller (pull batteries, turn back on), followed by putting it into pairing mode (hold connect on the controller).

Now, in bluetoothctl I scanned for devices, paired the device and trusted it again.

Everything is back to normal except the Guide button being silent, it doesn't show up in `evtest`. It probably has been moved to another HID code / key code now in the firmware. In my first tests two years ago, the controller generated two events for the Guide button, one button event and one keyboard event. The firmware probably fixed that, and discarded the button event. The Guide button is supposed to send a keyboard event only, and that's probably filtered out by the mapping routing (aka set to `MAP_IGNORE`, or `MAP_AUTO` and the kernel ignores it).

At least I can get back into my games, I don't need this button anyways.

ipkpjersi, 2020-05-03:
Is downgrading firmware possible or is that not really possible with Xbox One controllers?

kakra, 2020-05-03:
I used a Windows laptop, connected the controller via USB, and used the Xbox app to update the firmware.

kakra, 2020-05-05:
@ipkpjersi BTW: The button is fixed in my latest branch. It's an issue with how the Linux input layer treats buttons: Some software like `evtest` shows the event code, other software uses the scan code - and that wasn't correct. I ended up fixing the report descriptor instead so that button number and scan code match again.

closed 2020-05-08 by kakra