atar-axis/xpadneo #179: Run 'modprobe hid_xpadneo' after every boot

jefersonnavarro, 2020-05-03:
The driver seems to work fine, but every time the computer is started I must run 'modprobe hid_xpadneo' to get it to work.

What are the alternatives to make it work since boot ?

**System information**

-   `uname -a`
Linux leticia-localdomain 4.18.0-193.el8.x86_64 #1 SMP Fri Mar 27 14:35:58 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

-   `dmesg | grep xpadneo`
[  271.038720] xpadneo: hello there!
[  282.402226] xpadneo 0005:045E:02E0.0002: unknown main item tag 0x0
[  282.402788] xpadneo 0005:045E:02E0.0002: input,hidraw1: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 7c:b2:7d:81:86:97
[  282.409669] xpadneo udev: 0005:045E:02E0.0002
[  282.410408] xpadneo udev: ok
[  985.357017] xpadneo: raw_event: 04 87

-   `find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;`
/sys/kernel/debug/hid/0005:045E:02E0.0002
05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0a 15 00 25 01 75 01 95 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 80 85 02 a1 00 09 85 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75 08 95 01 81 02 c0 00 
437417229 922

jefersonnavarro, 2020-05-05:
I found the reason.

The udev man page ( man udev ) says:

RUN{type}
           Add a program to the list of programs to be executed after processing all the rules for a specific event, depending on "type":
           "program"
               Execute an external program specified as the assigned value. If no absolute path is given, the program is expected to live in /usr/lib/udev; otherwise, the absolute path must be specified.

In the file /etc/udev/rules.d/99-xpadneo.rules there is not the path to modprobe:

RUN:="/bin/sh -c 'echo xpadneo udev: $kernel > /dev/kmsg; modprobe hid_xpadneo; echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind; echo $kernel > /sys/bus/hid/drivers/microsoft/unbind; echo $kernel > /sys/bus/hid/drivers/xpadneo/bind; echo xpadneo udev: ok > /dev/kmsg'"

I do not know if every distribution puts modprobe in the same folder, but mine is in /usr/sbin/

which modprobe
/usr/sbin/modprobe

I edited the file, adding the folder and then it just worked.

kakra, 2020-05-09:
Closing as duplicate of #167

closed 2020-05-09 by kakra