atar-axis/xpadneo #180: Terrible controller performance for a short period

silverhammermba, 2020-05-04:
I have been using this with an Xbox One controller on Arch Linux and the controller performance has been awful in several ways:
* Missed inputs
* Low-resolution joystick input
* Stuck joysticks

I originally assumed this was due to a bad bluetooth receiver, however, if I connect the _same_ controller to a _different_ computer (with a different bluetooth receiver) running Windows 10, I get the exact same problems, _but only for a short duration_.

For about 1-2 minutes the control is terrible and nearly unusable, then suddenly it's as if the OS switches to some hi-performance mode and it works perfectly. I have never witnessed this switch occurring on Arch.

Does this make any sense with your understanding of how the controller works over bluetooth? Is there any way to replicate this result on Linux?

kakra, 2020-05-04:
Afair, I had problems with my previous bluetooth controller with laggy behavior similar to what you describe. The difference is the BT version, the previous one support v2 or v3 (I don't remember) while my current one support v4.

But I worked on a major code design overhaul which defers some work from the main input thread to a kernel worker thread. It may make a difference for your observation. Would you mind trying my branch? It's still WIP, so maybe it completely messes up how the controller works for you. In that case I very much appreciate some feedback so I can fix it. The new branch works fine for me.

rhyst, 2020-05-24:
I'm also experiencing this issue, with the lots of missed inputs and the controller generally feeling very unresponsive. Sometimes it will work for a few minutes but it seems to inevitably become unusable.

Is there any debug information I can provide to help diagnose this?

silverhammermba, 2020-05-24:
Turns out I should have done more testing on Windows 10. It might be _slightly_ better than Linux, but I still get occasional periods of awful performance where inputs get stuck or it fails to respond entirely for several seconds. And this is with me sitting right next to my computer, so I'm pretty sure it doesn't have to do with signal strength.

The fix: plugged it into my USB port because this controller is clearly garbage when it comes to bluetooth. Get a PS3 controller instead, they work perfectly with bluetooth on Linux.

kakra, 2020-05-24:
I'm guessing the problem may actually be the Bluetooth dongle you're using. I'm seeing no such problems. So to get some more insight:

1. Please show the output of `sudo btmon | tee btmon-performance.log` when the problem occurs and during pairing/connecting.
2. Please show the output of `lsusb -v -s##` where `##` is the device number of your dongle according to a plain `lsusb`.

Evidence is clearly condensing around the controller working correctly only with very specific Bluetooth dongles. But that it might be *slightly* better in Windows also tells us that the Linux drivers could improve. It looks like kernel 5.7 brings a lot of fixes around Bluetooth (maybe it even fixes the L2CAP problem with the controller, didn't try yet).

adam-becker, 2020-05-27:
I'm having this same issue in Arch, the driver seemed to work fine enough under Debian though. I'll see if I can capture this happening with `btmon`, but here's my `lsusb` output for now:

[lsusb.log](https://github.com/atar-axis/xpadneo/files/4685992/lsusb.log)

silverhammermba, 2020-05-27:
I don't have much time to spend on debugging this but I've tried three different bluetooth receivers (two built-in to different computers I own, one a small USB dongle) and all of them work perfectly with my PS3 controllers on Linux. All of them seem to have the same issues with the Xbox One controller (and one is a dual boot so I've been able to try it on Windows).

It may be that there are some software improvements to be made in this project, but it seems to me that there's a lot of evidence indicating that the Xbox One controller has some kind of exceptional hardware requirement compared to other bluetooth controllers. Maybe you need some super-sensitive receiver or something.

kakra, 2020-05-28:
> It might also be worth noting that the rumble on connect stutters a _lot_ but the stuttering feels evenly spaced, like there's a timing issue somewhere.

@adam-becker We initialize the controller with three short pulses per rumble effect (weak, strong, triggers). This tests if the pulsed rumble effect is correctly supported. There's at least one clone or semi-compatible model out there that doesn't support it. It's tracked in #192. So if your are seeing three short `bzz bzz bzz`pulses per rumble effect, that's actually working correctly.

kakra, 2020-05-28:
> It may be that there are some software improvements to be made in this project, but it seems to me that there's a lot of evidence indicating that the Xbox One controller has some kind of exceptional [...]

@silverhammermba Yes it seems so. I'm pinning this issue so other users may more easily find it. I'll watch the updates to the Bluetooth stack of Linux when kernel 5.7 is released. Let's see if that improves things. But apparently we probably cannot fix it in the driver except after starting work on supporting the original dongle that comes with the controller. But that's completely different technology, it's a wifi chipset not Bluetooth.

adam-becker, 2020-06-01:
I just want to note that the driver worked flawlessly for me at a certain point, and the only thing that's changed is the driver. Still using the same bluetooth dongle and same controller. I think something changed in the driver that introduced this, but I'm not entirely sure what.

kakra, 2020-06-01:
@adam-becker While I also think it is in the kernel, you could bisect the driver between 0.6 and master and see if we could pinpoint it down to a change in the driver. But you probably can reproduce the issue with 0.6.

kakra, 2020-06-02:
Maybe-related: https://github.com/atar-axis/xpadneo/issues/194

If someone who can reproduce the issue and is familiar with bisecting the kernel, they could try doing that. Nailing down the issue to a kernel commit could help us find the problem in the Bluetooth stack.

kakra, 2020-06-02:
@adam-becker Can you narrow down the kernel range to two point-0 releases of the kernel between which it broke?

kakra, 2020-06-02:
Please refer to #198.

adam-becker, 2020-06-02:
@kakra Narrowing it down to that level is going to be pretty tedious. I can boot back into debian later and see if the driver still works there, and I can make a note of the version of the kernel and the driver. Should give something to start on at least.

kakra, 2020-06-02:
Yes... If I'd be able to reproduce here, I would take your starting point, run git bisect and spawn a qemu machine using that kernel as boot kernel, and passing the dongle through. But except vastly different connect times from kernel version to version, I don't have any problems. I upgraded to 5.4.43 now (from .39) and the connect times seem to have doubled at least.

Please post your results to #198 (if the prereqs match), maybe we can narrow it down more easily by having all kernel version reports in one place.

adam-becker, 2020-06-16:
The issue for me on this appears to have been missing firmware for my bluetooth chipset. On arch the `linux-firmware` package didn't have what i needed, and prevented me from installing the correct one. Seems to work find since fixing that though.

kakra, 2020-06-17:
This is a good point, we should document that. Also, the controller should have the latest firmware (which can currently only be obtained and installed through Windows, tho).

kakra, 2020-06-27:
Meh...

rhyst, 2020-06-27:
Apologies for not responding earlier. I just got one of the Bluetooth adapters from the list and then couldn't be bothered to unpair/pair while it was working well.

I saw you briefly suggested this might fix the issue so I updated and tried again with my laptops internal bluetooth adapter. Attached is the btmon output. I don't think my adapter appears in lsusb so I haven't attached any output from that.

[btmon-performance.log](https://github.com/atar-axis/xpadneo/files/4841245/btmon-performance.log)

The btmon output is from ~20 seconds of attempting (while playing Hollow Knight). The lag was immediate and continuous so this log should be very representative of the issue.

kakra, 2020-06-27:
Did you try with current master? I just updated it today with a work-around for a probable bug/race in the controller firmware...

rhyst, 2020-06-27:
I think I am on latest master (8e7eba3). I pulled about an hour ago and then ran the uninstall/install scripts (the update script said I had an up to date version).

kakra, 2020-06-27:
> [btmon-performance.log](https://github.com/atar-axis/xpadneo/files/4841245/btmon-performance.log)
> 
> The btmon output is from ~20 seconds of attempting (while playing Hollow Knight). The lag was immediate and continuous so this log should be very representative of the issue.

This log doesn't even contain communication to or from the controller. It looks like you're having to Bluetooth dongles, hci0 (Qualcomm) and hci1 (Cambridge Silicon Radio, CSR). Both should show up in `lsusb` as they're connected to USB.

I can only guess that `btmon` monitored `hci0` only but your controller is connected to `hci1`. `hci0`(Qualcomm) seems to transmit a lot of data. Do you use Bluetooth audio?

Hmm, probably it's just missing logs from connection time? At least I see packets with `a1 01` which look like input reports from the controller. So it's connected to the Qualcomm bluetooth not to one from the list. It doesn't look like any rumble packet was sent, so I suppose we can exclude a problem introduces by rumble.

> I pulled about an hour ago and then ran the uninstall/install scripts (the update script said I had an up to date version).

There may be an issue with the update check on non-final versions. Master still shows `v0.7`, but we are really just before `v0.8`: `v0.7-32-g8e7eba3`. Please run uninstall and then install separately. I should really fix how versioning is applied to the code - putting it statically into `VERSION` is not suitable for master builds, just for tagged builds.

rhyst, 2020-06-27:
Woops, sorry I did have my bluetooth headphones connected. Here is a log with just the controller and should contain the connection as well. Same deal, immediately laggy, couldn't even navigate a menu.

[btmon-performance.log](https://github.com/atar-axis/xpadneo/files/4841288/btmon-performance.log)

kakra, 2020-06-27:
Signal strength seems to be in the "very good" range... So I don't think this is a range to signal strength issue. But it's fluctuating a lot, I guess you've moved towards the receiver at some point, then moved back.

I cannot see if it lags behind but your system sees way too few input reports, at some point only 3 per second even while it seems you're actively using the controller. It should send input reports at a much higher rate. From what I see you probably experience a lag of 300ms for input from the controller - and even that is not smooth. With those numbers, I expect that every input in the menu comes 300ms late (which your brain can definitely tell, it has to be more like 20-30ms to start perceiving immediate reaction), and due to the low rate it will probably jump multiple menu items per stick movement. Can you confirm that?

Could you try to connect the controller to `hci1` instead? Did you also properly update the driver as I was outlining above?

rhyst, 2020-06-27:
I'm basically 30cm away from the adapter, for both the external and internal adapters so the signal strength should be fine.

It's hard to tell what the lag is on an individual input as it seems to miss most of them. For that previous log I was basically continuously moving the left analog stick.

Here is the log from connecting to `hci1` and moving the analogue stick:

[btmon-performance.log](https://github.com/atar-axis/xpadneo/files/4841341/btmon-performance.log)

I ran `uninstall.sh` and then `install.sh` to update the driver as you said before collecting the previous log, so it should have been using the latest driver.

kakra, 2020-06-27:
Unrelated to your original problem but would you mind testing #207 to see if version detection then works correctly? You may comment there.

kakra, 2020-06-27:
> Here is the log from connecting to `hci1` and moving the analogue stick:
> 
> [btmon-performance.log](https://github.com/atar-axis/xpadneo/files/4841341/btmon-performance.log)

That looks much better as you are getting input reports around every 10ms now. Does it work better?

kakra, 2020-06-28:
> I have been using this with an Xbox One controller on Arch Linux and the controller performance has been awful in several ways:
> 
> * Missed inputs
> * Low-resolution joystick input

This may result from the same issue as https://github.com/atar-axis/xpadneo/issues/180#issuecomment-650639451 and is either an incompatible Bluetooth chipset, or the chipset is missing the `linux-firmware` package from your distribution, or it is some result of power management in the Bluetooth driver or chipset.

> * Stuck joysticks

This is probably related to the rumble protocol and a fix for that is in `master`.

> I originally assumed this was due to a bad bluetooth receiver, however, if I connect the _same_ controller to a _different_ computer (with a different bluetooth receiver) running Windows 10, I get the exact same problems, _but only for a short duration_.

This hints towards my first theory. The second one may be unrelated, except that Windows shows evidence of slowing down the rumble protocol below 100 Hz. An implementation of such a throttle is in `master` now.

> For about 1-2 minutes the control is terrible and nearly unusable, then suddenly it's as if the OS switches to some hi-performance mode and it works perfectly. I have never witnessed this switch occurring on Arch.

This hints towards broken power management as outlined above. Apparently, `xpadneo` has no control over that. You may want to research if you can supply a module parameter to the driver of your Bluetooth chipset (which is usually the generic `btusb` driver).

rhyst, 2020-06-28:
> That looks much better as you are getting input reports around every 10ms now. Does it work better?

Using the `hci1` adapter works perfectly, except sometimes it is slightly difficult to connect it (which can be worked around by unpairing and re-pairing).

kakra, 2020-06-28:
Yes, the connection problems are known for this chipset. For me, it works although it sometimes takes up to 1 minute to connect, and it `btmon` you'll see that it tries to connect multiple times until it eventually succeeds. This may be related to the kernel version (I'm on 5.4), Bluetooth subsystem version (I'm on 2.22 according to `btmon`), bluez version (I'm on 5.54, distribution package), bluetoothd version (I'm on 1.14 according to `btmon`), disabling ERTM (but I did not) or adding the L2CAP patch to the kernel (which I did).

The Dual Shock 2 controller seems to have similar issues, so it's probably a kernel or bluez related issue.

For me, the problem was relaxed with later versions of kernel 5.4, and was fixed by using a dongle with the same chipset but from a different brand (previously no-name, now TP-LINK).

kakra, 2021-01-31:
Is this still an issue? It may be a duplicate of #272, as ERTM affects how packets are transmitted.

rhyst, 2021-02-01:
Seems to work now for me even with the adapter that was causing issues before. I only tested for a few minutes but that was enough to get unresponsive controls before. I think I'm on `v0.9-23-g6162dbc` (pulled master just before testing) of xpad neo and `Ubuntu 20.04` with kernel `5.4.0-65-generic`.

kakra, 2021-02-01:
It was probably fixed by some kernel updates and was related to Bluetooth. There have been a lot of fixes lately, one important fix is currently queued (L2CAP patch) which ultimately enables using ERTM again. I'm closing this then.

cybercyst, 2021-02-11:
I know this is closed, but I had this exact same issue, and reverting my xpadneo driver to pre 0.9 fixed it. The controller works beautifully again. No changes to bluetooth main.conf configurations or anything

kakra, 2021-02-11:
@cybercyst So you are saying that this is a post-0.9 issue only? Are you skilled enough with git that you could bisect between v0.8 and v0.9 to find the commit that is causing this?

cybercyst, 2021-02-18:
yeah, i've done git bisects before, it's more of a time issue -- i'll see if i can eke some out

cybercyst, 2021-03-27:
Rebuilt master, made sure all my xpadeo dkms drivers were removed and things seem a bit better now. Also apparently needed SDL_JOYSTICK_HIDAPI=0 for steam to get the proper bindings.
All in all things seem OK now!

kakra, 2021-03-27:
I wonder what was doing the trick, maybe because I moved the loading position of the udev rule?

cybercyst, 2021-03-31:
I think I needed to read the README more closely
Adding ```SDL_JOYSTICK_HIDAPI=0``` seems to have been the fix
Without this the game above in question (and steam in big picture mode) acts like it is dropping a bunch of inputs. I think in reality is is getting double the inputs? I'm not 100 sure, but the environmental variable has things working well for me.

kakra, 2021-03-31:
If you're suspecting double-input, this may be relevant:
https://github.com/ValveSoftware/Proton/issues/4707

Thanks for your feedback, that points me into a direction of an issue I'm currently observing with my XBE2 controller.

kakra, 2021-03-31:
Well the problem is somewhere in the Bluetooth stack which we have absolutely no control over from the driver. And I don't know how the Bluetooth protocol works with all its technical specs and papers. I can only provide debugging and testing efforts to the Bluetooth developers.

For my XBE2 controller, I'm currently seeing the same issue as the original reporter, I believe. But some months ago, it was working just fine. But this seems to be purely a thing of luck: Un-pair and re-pair the controller and it may totally change behavior or even refuse to connect at all until you pair it to different machine (or a different protocol using the Xbox dongle). But whatever state it is in, it always connects fine to Android or Windows for me (even with the same Bluetooth dongle), instantly and without issues. So that draws a circle of evidence around the Linux Bluetooth stack. It's not an RF environment issue (radio pollution), it's not a distance issue (signal strength), something goes wrong in the handshake and communication, and maybe it's some timing or out-of-order issue. But apparently it's not an xpadneo issue as the xpadneo driver cannot control Bluetooth, we just get a handle/comm channel for the HID data.

kakra, 2021-05-01:
Do you have kdeconnect installed and running?

kakra, 2021-06-20:
After some more investigation, there may be two causes for it, and, sadly, both are unrelated/unsolvable by xpadneo:

1. If this is the new BLE model of the controller (that one with the Share button), it may lag under certain not yet fully known conditions: https://github.com/bluez/bluez/issues/156
2. The Linux Bluetooth stack may drop/delay Bluetooth packets which are important for us while an inquiry is running, usually, this may happen on a regular basis or just at random times (one such occurrence is with the kdeconnectd daemon), resulting in "terrible performance" or even dropped connections, not all models are affected in the same way: https://github.com/bluez/bluez/issues/123

closed 2022-04-30 by kakra