atar-axis/xpadneo #181: Trigger Force Feedback - direction_rumble_test does not work

jefersonnavarro, 2020-05-05:
I tried to test trigger force feedback by running misc/examples/c_directional_rumble_test/direction_rumble_test js0 30000.

I got the messages "Directional Rumble Test
Press ENTER to upload Rumble effects"

When enter key was pressed I got the messages "Error while uploading
: Bad file descriptor"

The trigger force feedback did not work.

What is the problem and how to fix it ?

kakra, 2020-05-05:
Do not use `js0` but the event number from `/dev/input/event*`. You can run `evtest` to find the number.

jefersonnavarro, 2020-05-06:
I do not have evtest package in my repositories but managed to find out the event. In my case it was event19, so I ran direction_rumble_test 19 30000. It works. Thank you.

closed 2020-05-06 by jefersonnavarro