atar-axis/xpadneo #184: Uninstaller didn't remove DKMS module

ipkpjersi, 2020-05-09:
**Describe the bug**
I wanted to uninstall 0.7 so I could test 0.6 some more. I tried uninstalling with the uninstall.sh provided by 0.7, and it reported `found 0 registered instance(s) on your system` even though DKMS still reports that it is installed

**To Reproduce**
Steps to reproduce the behavior:
1. Install xpadneo 0.7 with install.sh
2. Try uninstalling xpadneo 0.7 with uninstall.sh

**Expected behavior**
I expect it to fully uninstall xpadneo 0.7

**Screenshots/Gifs**
[image]

**System information**

-   `Linux myhostname 4.15.0-99-generic #100-Ubuntu SMP Wed Apr 22 20:32:56 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`
-   https://gist.github.com/ipkpjersi/35bcc262b0502cf2519b2e39d5f79b27

ipkpjersi, 2020-05-09:
I rebooted my PC after running `sudo dkms uninstall hid-xpadneo/0.7`however it seems to remain installed.

kakra, 2020-05-09:
I'm not familiar with DKMS as my system does not support it (Gentoo Linux). Is this a regression, i.e. did it work correctly with v0.6? I didn't touch any of the install routines during my work on v0.7.

ipkpjersi, 2020-05-09:
This seems to be a regression, I don't remember having any problems with this with 0.6.

I can't install 0.6, and now when I try to run `sudo dkms uninstall hid-xpadneo/0.7` again, it says it is not installed, even though `sudo dkms status` reports it as installed still.

kakra, 2020-05-09:
There's probably a left-over file somewhere... Maybe use the search engine of your choice to find how to manually remove a DKMS module from the system. If you've found the culprit file, let me know. Maybe also post the complete output of the installer script when running with `bash -x install.sh` - it should spool a lot of debug output.

ipkpjersi, 2020-05-09:
Weird, I can re-install xpadneo DKMS module with `sudo dkms install hid-xpadneo/0.7`.

The output of your command:

[image]

I am trying `sudo dkms remove -m hid-xpadneo -v "0.7" --all` now and I hope that works.

Here is where my xpadneo files are located, I'm slightly afraid of removing the .ko files.
[image]

kakra, 2020-05-09:
Ah yes I see where this comes from: The uninstaller is looking for xpadneo x.y.z as version number but I omitted the .0 suffix, so we only have x.y as in 0.7

kakra, 2020-05-09:
Can you please try my patch?

ipkpjersi, 2020-05-09:
I am unable to install your patch version, it fails with:

```
./lib/installer.sh: line 10: syntax error: unexpected end of file
* replacing version string if necessary
./install.sh: line 23: get_dkms_versions_installed: command not found
```

kakra, 2020-05-09:
Please try again...

ipkpjersi, 2020-05-09:
I have tried commit af6b31f39cccfbc2852041c9c036c70433fa584a directly and I can confirm that installing does work, however, uninstalling still does not work:
[image]

kakra, 2020-05-09:
Could you post the output of `dkms status` as plain text instead of screenshots so I can copy and paste it? Thanks. (you can put triple-backticks around the text to format it as monospaced text)

ipkpjersi, 2020-05-09:
```
hid-xpadneo, 0.7, 4.15.0-99-generic, x86_64: installed
```

kakra, 2020-05-09:
Please do not try to shortcut things, post the complete output. I'm trying to figure out why it worked with my test data but not with your real data.

ipkpjersi, 2020-05-09:
Sorry, I thought you only needed the one line.

Here is the full output of dkms status:

```bash
$ sudo dkms status
bcmwl, 6.30.223.271+bdcom, 4.15.0-96-generic, x86_64: installed
bcmwl, 6.30.223.271+bdcom, 4.15.0-99-generic, x86_64: installed
hid-xpadneo, 0.7, 4.15.0-99-generic, x86_64: installed
nvidia, 440.82, 4.15.0-96-generic, x86_64: installed
nvidia, 440.82, 4.15.0-99-generic, x86_64: installed
openrazer-driver, 2.8.0, 4.15.0-99-generic, x86_64: installed
virtualbox, 5.2.34, 4.15.0-96-generic, x86_64: installed
virtualbox, 5.2.34, 4.15.0-99-generic, x86_64: installed
zfs, 0.8.3, 4.15.0-96-generic, x86_64: installed
zfs, 0.8.3, 4.15.0-99-generic, x86_64: installed
```

Could it be because dkms is not running as root? 

Keep in mind, I am fairly sure that this is a regression, as I am pretty sure I remember the uninstall script was working in 0.6 for me.

kakra, 2020-05-09:
Yes it worked in 0.6 because it matched the version number 0.6.0 but it wouldn't match 0.6 or 0.7...

Ok, not sure why you need root permissions to run just `dkms status` but at least in your last test it worked, although you should have used `sudo $(type -p dkms)` instead. Could you try that? I'm using the type command to test if dkms is installed. When typing `type -p dkms` it should show the full path of dkms.

kakra, 2020-05-09:
I've updated the patch. Please try to rerun `sudo bash -x ./uninstall.sh`.

kakra, 2020-05-09:
> I've updated the patch. Please try to rerun `sudo bash -x ./uninstall.sh`.

There was a bug previously, I've redone the whole concept now. Please try all actions if you don't mind: install, uninstall, configure, and update. Thanks.

ipkpjersi, 2020-05-09:
I just tried uninstalling again and it did not work, here is the output:

```
$ sudo bash -x ./uninstall.sh
++ dirname ./uninstall.sh
+ source ./lib/installer.sh
++ '[' 0 -ne 0 ']'
+++ type -p which
++ DKMS_BIN=/usr/bin/which
++ : /usr/bin/which
++ INSTALLED=($(get_dkms_versions_installed))
+++ get_dkms_versions_installed
+++ /usr/bin/which status
+++ awk '-F, ' '-e$1 == "hid-xpadneo" { print $2 }'
+++ sort -nu
+ echo '* unloading current driver module'
* unloading current driver module
+ modprobe -r hid_xpadneo
+ echo '* looking for registered instances'
* looking for registered instances
+ echo 'found 0 registered instance(s) on your system'
found 0 registered instance(s) on your system
```

kakra, 2020-05-09:
Yes, I see... Since I don't have dkms, I've put a different command into the type command for testing and forgot to revert it. I'll push an update now, stand by...

ipkpjersi, 2020-05-10:
Yep, that did the trick, it found and uninstalled my DKMS module, and then after that I installed it successfully too. Both uninstall and install are working great. I think this may be solved now.

closed 2020-05-10 by kakra