atar-axis/xpadneo #188: Cannot uninstall/reinstall driver if already installed

Mikaka27, 2020-05-11:
**Describe the bug**
Cannot uninstall/install driver.
Function:
```
get_dkms_versions_installed() {
	${DKMS_BIN} status | awk -F', ' -e'$1 == "hid-xpadneo" { print $2 }' | sort -nu
}
```
is incorrect, it produces the following error:
`awk: not an option: -e$1 == "hid-xpadneo" { print $2 }`

It seems that standard Kubuntu 19.10 does not have awk version that is required for this to work.
Doing `awk -e` produces:
`awk: not an option: -e`

**To Reproduce**
Download repo, switch to v0.7.1 tag and install.
Then try to uninstall, you get:
```
awk: not an option: -e$1 == "hid-xpadneo" { print $2 }
* unloading current driver module
* looking for registered instances
found 0 registered instance(s) on your system
```

As a result of that, when you try to install again you get:
```
awk: not an option: -e$1 == "hid-xpadneo" { print $2 }
* replacing version string if necessary
* copying module into /usr/src
* adding module to DKMS
Error! DKMS tree already contains: hid-xpadneo-0.7
You cannot add the same module/version combo more than once.
```

**Expected behavior**
Driver should be uninstalled, installed correctly.

**Screenshots/Gifs**
Not needed

**System information**
Kubuntu 19.10 x64

**Additional context**  
No context

Mikaka27, 2020-05-11:
This can be fixed by installing gawk, however in that case this should be listed in `Prerequisites`.

kakra, 2020-05-11:
We should try to keep the prereqs as small as possible...

Could you try if replacing the line in `lib/installer.sh` helps:
```
# replace "-e" with "-- "
get_dkms_versions_installed() {
	${DKMS_BIN} status | awk -F', ' -- '$1 == "hid-xpadneo" { print $2 }' | sort -nu
}
```

This should make it compatible with busybox-awk and thus non-gnu awk.

Mikaka27, 2020-05-11:
Both options (with `--` and with nothing) work for me.

kakra, 2020-05-11:
Okay, thanks. I'll craft a patch from this info.

ewbteewbte, 2020-05-14:
> We should try to keep the prereqs as small as possible...
> 
> Could you try if replacing the line in `lib/installer.sh` helps:
> 
> ```
> # replace "-e" with "-- "
> get_dkms_versions_installed() {
> 	${DKMS_BIN} status | awk -F', ' -- '$1 == "hid-xpadneo" { print $2 }' | sort -nu
> }
> ```
> 
> This should make it compatible with busybox-awk and thus non-gnu awk.
> 
> **Update:** I'm not sure if `--` works with non-gnu awk. Maybe we should use neither `-- ` nor `-e`. Could you test both variants?

Hey, same issue here on Pop_OS 20.04 but only "-- " worked for me.

kakra, 2020-05-15:
Then I think we go with the `-- ` variant.

closed 2020-05-21 by kakra

Yasand123, 2022-04-27:
I have a similar issue, when uninstalling using the `uninstall.sh` script, it outputs the following:
```
* unloading current driver module
* looking for registered instances
found 0 registered instance(s) on your system
```
No errors. It just won't uninstall.

When trying to install again via `install.sh` it tells me the module is already installed:
```
* creating dkms.conf
* adding hid-xpadneo-v0.9-96-g4fd620c folder to /usr/src
* installing module (using DKMS)
Module hid-xpadneo/v0.9-96-g4fd620c already installed on kernel 5.17.4-arch1-1 (x86_64).
```

- `Linux ArchLinux 5.17.4-arch1-1`
- `git version 2.35.1`