atar-axis/xpadneo #192: Continuous rumble after connecting

RPGHank, 2020-05-24:
After connecting the controller via Bluetooth it rumbles continuously

I am using Zorin OS 15.2 with the 8BitDo SN30 Pro+ in XInput mode

kakra, 2020-05-24:
Please disconnect the controller and run `sudo btmon | tee btmon-rumble.log`, then connect the controller. Post the output here (or rather attach the file).

RPGHank, 2020-05-25:
> Please disconnect the controller and run `sudo btmon | tee btmon-rumble.log`, then connect the controller. Post the output here (or rather attach the file).

[upload.zip](https://github.com/atar-axis/xpadneo/files/4675153/upload.zip)

kakra, 2020-05-25:
Thanks, the contents are the same, the log file is just a mirror of the console output (due to `tee`). So I only needed the file.

From the file I see that it is sending the rumble commands correctly:

It sends three rumble commands:
```
        a2 03 01 0a 0a 14 28 05 05 03                    ......(...
        a2 03 02 0a 0a 14 28 05 05 03                    ......(...
        a2 03 0c 0a 0a 14 28 05 05 03                    ......(...
```

This can be seen a seconds time. I guess you just connected a second time within the same log:
```
        a2 03 01 0a 0a 14 28 05 05 03                    ......(...
        a2 03 02 0a 0a 14 28 05 05 03                    ......(...
        a2 03 0c 0a 0a 14 28 05 05 03                    ......(...
```

This command series should pulse the following motors 330 ms each: weak rumble motor, strong rumble motor, both trigger rumble motors in parallel. Actually, it's a test if all rumble motors work and are present.

Per each rumble, it should play the following effect: Pulse the motor for 50 ms, wait 50 ms, repeat 3 times. So you'll get a `bzz bzz bzz` (3x 50 ms pulse + 50 ms pause) for each motor. This tests is the rumble effects work correctly. It'll confirm that the firmware supports looping, pulse sustain, and pulse release.

Please answer the following questions:

1. Did all three phases play? You're probably having both the weak and strong rumble, not sure about the trigger rumble.
2. Did all three phases stop before the next phases started? Or did they build up, thus weak rumble starts, 330 ms later strong rumble starts and weak rumble continues to play, etc.
3. Did the pulsing `bzz bzz bzz` work correctly or did it play continuously per 330 ms period?

I'm currently guessing that the controller may become confused by trigger rumble if that's not supported so we may need a work-around for that. But if trigger rumble is supported, it may not support the rumble pulse and loop parameters (sustain, release, loop count). If that's the case, we may need to send and explicit rumble stop after initialization. For normal operation that should not matter as the kernel force feedback driver will send stop commands by itself.

I'm also guessing that you do not get the `bzz bzz bzz` 50 ms pulses but instead a continuous effect.

Sorry for quite the quite pictographic `bzz bzz bzz` - I hope you'll understand what I mean. ;-)

RPGHank, 2020-05-25:
1)Perhaps the firmware works differently because usually when connecting to Windows or to Linux before I installed XPad it immediately gives a short (50ms? maybe) buzz in which both the strong and weak play together
2)When connecting to Linux after a hour or so it gives its customary buzz (similar to when connecting to Windows) and after about 2 seconds both the motors start buzzing simultaneously. When connecting around the same time I disconnect it buzzing starts immediately without the short buzz
3)It doesn't give bzz bzz bzz and as far as I can reckon it is a continuous simultaneous buzzing of the strong and weak motors
The controller doesn't have trigger rumble of any sort if that helps
(And I'm sorry if this wasn't the information you were looking for)

kakra, 2020-05-25:
> (And I'm sorry if this wasn't the information you were looking for)

I think that'll help. But I have to make up my mind about it. And this week will be pretty busy (as the previous weeks and forthcoming weeks) so I can only dedicate time to this project in the weekends.

What immediately comes to mind it that we'd need some sort of quirk handling like other Linux drivers do. But I'm not quite sure how we can actually differentiate your controller model from the one it's emulating. Maybe you could help here. Please follow the instructions of the [bug report template](https://github.com/atar-axis/xpadneo/blob/7536b6f9455d407d249622c041efaef492bf0635/.github/ISSUE_TEMPLATE/bug_report.md). Some of that information was already collected so you could skip some steps.

Thanks for helping with improving the driver.

RPGHank, 2020-05-26:
Describe the bug: Continuous rumble when connecting via Buetooth

To Reproduce: Using Zorin OS Core 15.2 and 8BitDo SN30 Pro+ in XInput mode

Expected behavior: For the controller to give a short rumble and then be able to use the rumble on the controller

Screenshots/Gifs If applicable, add screenshots or animated gifs to help explain your problem.

System information:
[Upload.zip](https://github.com/atar-axis/xpadneo/files/4681045/Upload.zip)
(I hope the naming of the files is clear)

PS I don't need a fix immediately since in the weird way the controller works when I plug it in without  connecting to Bluetooth its recognized as X360 and rumble works that way (plus I'm still struggling to get non-native games on Linux)

kakra, 2020-05-26:
Please run the xxd command while the controller is connected via Bluetooth. Also, the `lsusb` output is missing. And I'd prefer if files would be attached as single uncompressed attachments each. Just drag files into the issue comment box to upload them unless they are 1 MB or bigger. Short plain text output should be posted just inline with the text by using ` ``` ` (three backticks) markers before and after.

I think we could try identifying the device by its name which it is sending via the Bluetooth protocol. But I don't like doing that idea, it feels wrong, names are hollow words, their handling is cumbersome and error-prone in the kernel. We should try to avoid that as best we can. I'll take a look at other drivers and how those handle similar problems.

RPGHank, 2020-05-27:
[xdd.docx](https://github.com/atar-axis/xpadneo/files/4686163/xdd.docx)

In lsusb I couldn't identify anything related to bluetooth (is it because I'm using inbuilt bluetooth instead of a dongle)

Here's the list that appeared in case you can identify something
[lsusb.docx](https://github.com/atar-axis/xpadneo/files/4686270/lsusb.docx)
ad

friesnadrink, 2020-06-02:
I am having the same issue with the same model of controller on Arch Linux. Here's my info, in case it is useful.

Describe the bug: Connecting an 8bitdo sn30 pro+ over bluetooth causes the controller to rumble continuously until a program sends a rumble signal to it, at which point the controller starts working as expected.

To Reproduce: install xpadneo then connect 8bitdo sn30 pro+ (in xinput mode) over bluetooth

Expected behavior: controller connects, runs brief rumble tests, then stops rumbling until a program initiates rumble.

Screenshots/Gifs: N/A

System information:

    Linux archlinux 5.6.14-arch1-1 #1 SMP PREEMPT Wed, 20 May 2020 20:43:19 +0000 x86_64 GNU/Linux
[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/4719712/xpadneo-lsusb.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/4719713/xpadneo-dmesg.txt)
[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/4719714/xpadneo-btmon.txt)
[xxd.txt](https://github.com/atar-axis/xpadneo/files/4719715/xxd.txt)

kakra, 2020-06-03:
@friesnadrink Could you also describe if the three rumble effects play in sequence (i.e. the previous effect stops when starting the next) or if they build up (the second plays in parallel etc). This may be a bit difficult to tell apart as you only see two effects and the strong rumble is quite slow to start.

friesnadrink, 2020-06-03:
@kakra As far as I can tell it just starts a strong rumble and doesn't stop until it receives a rumble signal from a game. I cannot distinguish any separate rumble signals during the initial rumble tests. Also, I'm not really sure how it is supposed to feel either as I don't have a real xbox controller to test.

kakra, 2020-06-03:
@friesnadrink Does the controller even support two different rumble strengths? This should be easy to tell as games usually make use of that.

friesnadrink, 2020-06-03:
@kakra the product video shows that it has a left and right motor and that both can vary in strength:
https://youtu.be/Dq4bJ5Qpn8M?t=52
Also, I've been using the input settings in Cemu on wine to easily send a rumble signal to it quickly because adjusting the rumble strength there works to stop the continuous rumble.

friesnadrink, 2020-06-04:
@kakra So i got the t6 screwdriver needed to open the controller and found out that it is both motors that continuously spin. They start right after the bluetooth connection is established and there is no sign of the 3 separate rumble signals.

kakra, 2020-06-05:
@friesnadrink They should start in sequence (with a delay of 330 ms) - can you confirm that?

BTW: Thanks a lot for trying that even when I didn't ask you to open the controller casing.

friesnadrink, 2020-06-05:
@kakra Here is the sequence of events as best I can describe them:
1. turn controller on in xbox mode (x + start)
2. controller searches for bluetooth host
3. controller establishes connection with bluetooth host
4. controller starts to rumble and does not stop
The delta between 3 and 4 could be 330ms, but that's a fraction of a second and I just can't be sure if it's 150, 330, 500 or some other number. There is no sequence to the rumble it's just off, then on.

I don't claim to know exactly what's going on inside, but it seems to me that the controller is sent a data packet that it doesn't interpret correctly, resulting in perpetual rumble. Do any of the test signals include feedback for the triggers? If they do then that would be my best guess as to what is causing it, since the 8bitdo does not have that feature.

kakra, 2020-06-05:
Okay, so at least our driver can properly control each motor individually. I think we gathered enough info and I'll try to come up with a solution over the course of the weekend.

For your curiosity: The controller doesn't support the pulse parameters of the rumble protocol. It can just turn motors on at a specific strength (with strength 0 = off). But the Xbox controller has support for also supplying a parameter for the pulse sustain and pulse release duration in steps of 10ms, after which the motors are stopped automatically (motor starts at beginning of the pulse sustain delay and stops at the beginning of pulse release delay). Another parameters let's us loop sustain and release for serveral times. The init routing expects exactly this to be working.

So the rumble implementation is incomplete. This is usually not a problem as the kernel driver for force feedback sends commands to us in 20ms resolution and also sends a stop signal.

What we need now is a quirk flag for disabling the pulse parameters and handle that correctly, and a quirk flag for disabling trigger rumble support (the latter is already in place under a different name). During init the quirk flags will be respected to do the proper thing and stopping the motors explicitly.

Now I only need to figure out the best way to identify the controller and enabling the proper quirk flags. That's the harder part.

I'm still a bit clueless about why both motors continue to rumble. Do they? Or did I misunderstand this? The second command (330 ms after the first) should automatically stop the first motor. If it doesn't we probably need and explicit stop signal per motor and I have to rethink the quirk flags a little.

friesnadrink, 2020-06-05:
Thank you for your efforts on this. This driver is by far the best I've used and it sounds like it's about to get even better! 

I think you understood correctly, both motors turn on at the same time and continue rumbling.

kakra, 2020-06-06:
Ah thanks, this is very cool. I managed to download the mp4 with `youtube-dl` and played it back frame by frame: Actually, both motors start spinning in the same frame while they should start 20 frames apart (60 fps × 330 ms ≅ 20 frames).

So I imagine that the driver cannot currently control the motors independently... We should try to figure out how to do that correctly.

friesnadrink, 2020-06-06:
I made another video with the following code changes, just to test some of the values:
https://youtu.be/lPrXDD_qr84

- increased the delay to 1s
- test only magnitude_weak and magnitude_strong
- the ff_report members are all 0 except tested variable

```
static void xpadneo_welcome_rumble(struct hid_device *hdev)
{
	struct ff_report ff_pck;

	memset(&ff_pck.ff, 0, sizeof(ff_pck.ff));

	ff_pck.report_id = XPADNEO_XB1S_FF_REPORT;
	ff_pck.ff.magnitude_weak = 0;
	ff_pck.ff.magnitude_strong = 0;
	ff_pck.ff.magnitude_right = 0;
	ff_pck.ff.magnitude_left = 0;
	ff_pck.ff.pulse_sustain_10ms = 0;
	ff_pck.ff.pulse_release_10ms = 0;
	ff_pck.ff.loop_count = 0;
	
	ff_pck.ff.enable = 0;

	ff_pck.ff.magnitude_weak = 40;
	//ff_pck.ff.enable = FF_RUMBLE_WEAK;
	hid_hw_output_report(hdev, (u8 *)&ff_pck, sizeof(ff_pck));
	mdelay(1000);
	ff_pck.ff.magnitude_weak = 0;

	ff_pck.ff.magnitude_strong = 20;
	//ff_pck.ff.enable = FF_RUMBLE_STRONG;
	hid_hw_output_report(hdev, (u8 *)&ff_pck, sizeof(ff_pck));
	mdelay(1000);
	ff_pck.ff.magnitude_strong = 0;
	
	hid_hw_output_report(hdev, (u8 *)&ff_pck, sizeof(ff_pck));
	
	//ff_pck.ff.enable = FF_RUMBLE_TRIGGERS;
	//hid_hw_output_report(hdev, (u8 *)&ff_pck, sizeof(ff_pck));
	//mdelay(330);
}
```

As the video shows, the driver can control each motor independently. I'll do some more tests and let you know how it goes.

kakra, 2020-06-06:
Yeah, I suspected something like that: It seems to ignore the enable flags of the motors. That's the only thing that explains it. I'm currently creating a small test program to send the rumble commands via hidraw to the device (thus bypassing the driver). It will allow setting individual flags and values by some keypresses.

kakra, 2020-06-06:
@friesnadrink I've added a hidraw test program for you to play with. Inspect `dmesg` after the controller connected to find the hidraw device name. To build the program, go to `misc/examples/c_hidraw` and run `make`.

Then execute the program with `sudo ./hidraw /dev/hidraw##` according to the correct hidraw device picked from `dmesg`.

**Caution:** Don't run the program for hidraw devices other than the controller, as it may hang or freeze the device (or even make it crash or reprogrammed).

You should be able to find which parameters work and which don't.

The original Xbox controller works like this:

The enable flags (keys 1-4) control which rumble motors are programmed. Sending a long command (parameters sustain, release, loop with high numbers) only updates the motors for the flags enabled, the other motors continue to play their last command until finished. So it's actually not a enable/disable flag, but a flag for which motors are reprogrammed.

Your controller seems to completely ignore those flags and only looks at the strength parameters for deciding if a motor should start or stop. Also, I'm guessing that it ignores the pulse parameters (sustain, release, loop count).

This is a small ncurses application, you may need to install `ncurses-dev` or whatever your distribution calls it. It looks like this:
```
--[  MotorEnable  ]---------------[ Strength in % ][  10ms ][cnt]--
03  1   1   1   1   0   0   0   0  040 030 020 020  005 005  003
    |   |   |   |                   |   |   |   |    |   |    |
   STR WEA LTR RTR  -   -   -   -  STR WEA LTR RTR  SUS REL  LOP
   [1] [2] [3] [4]                 q/a w/s e/d r/f  h/j k/l  u/i    <-- keys, or Enter = SEND, Ctrl+C = ABORT
```

03 is the command and cannot be changed, the other parameters are changed with the keys indicated in the last line.

friesnadrink, 2020-06-07:
I went through all the meaningful combinations of values and only the STR% and WEA% appear to do anything (they work correctly). As far as I can tell, all of the enable toggles on the left do nothing. Same goes for all these other values: LTR%, RTR%, SUS, REL, and LOP.

kakra, 2020-06-07:
Ok thanks, this result solves the last part of the puzzle.

kakra, 2020-06-07:
Can you try my pull request at https://github.com/atar-axis/xpadneo/pull/199

It should detect your controller and write a message to dmesg that quirk 0x0007 is active.

You can override the behavior by adding a module parameter `quirks=MAC:flags` where MAC is the Bluetooth MAC of your controller (not the dongle MAC!) and flags is the sum of different flags, see `modinfo hid-xpadneo`.

friesnadrink, 2020-06-08:
It does indeed work.

```
[ 9284.032873] hid-generic 0005:045E:02E0.0008: unknown main item tag 0x0
[ 9284.033015] input: 8BitDo SN30 Pro+ as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-6/1-6:1.0/bluetooth/hci0/hci0:512/0005:045E:02E0.0008/input/input30
[ 9284.033113] hid-generic 0005:045E:02E0.0008: input,hidraw6: BLUETOOTH HID v9.03 Gamepad [8BitDo SN30 Pro+] on 38:00:25:4d:63:6c
[ 9284.036255] xpadneo udev: 0005:045E:02E0.0008
[ 9284.180454] xpadneo 0005:045E:02E0.0008: fixing up report size
[ 9284.180681] xpadneo 0005:045E:02E0.0008: battery registered
[ 9284.180685] xpadneo 0005:045E:02E0.0008: pretending XB1S Linux firmware version (changed version from 0x00000903 to 0x00000903)
[ 9284.180726] input: 8BitDo SN30 Pro+ as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-6/1-6:1.0/bluetooth/hci0/hci0:512/0005:045E:02E0.0008/input/input31
[ 9284.180831] xpadneo 0005:045E:02E0.0008: input,hidraw6: BLUETOOTH HID v9.03 Gamepad [8BitDo SN30 Pro+] on 38:00:25:4d:63:6c
[ 9284.180833] xpadneo 0005:045E:02E0.0008: controller quirks: 0x0007
[ 9284.841675] xpadneo udev: ok
```

Thanks so much!

kakra, 2020-06-08:
Does it do the connection rumble correctly, and does rumble in games work correctly?

friesnadrink, 2020-06-08:
Yes, it does the weak then strong initial rumbles correctly (stops correctly too), and it works in native, steam, and wine games perfectly without any special settings.

closed 2020-06-08 by kakra