atar-axis/xpadneo #194: Controller connected... but is it?

bige11, 2020-05-28:
Following the instructions I pair, trust, and connect my xbox one controller via bluetooth. The terminal tells me that the connection is successful, but the controller's light never stops pulsating (at first it keeps pulsating fast, then it slows). After a while, it turns off altogether - interrupting the connection. Also, when the connection is established, the controller does not rumble. If I open the controller settings in Steam, the controller seems to be recognized - but I am not sure it really works - as I said, it turns off shortly after.

[Xbox Wireless Controller 7EED803C5A1D]# info C8:3F:26:3B:23:76
Device C8:3F:26:3B:23:76 (public)
	Name: Xbox Wireless Controller 7EED803C5A1D
	Alias: Xbox Wireless Controller 7EED803C5A1D
	Class: 0x00000508
	Icon: input-gaming
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
	UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)

I am using Ubuntu 20.04 (5.4.0-33-generic)

Wireless card: Intell Killer AX1650 (bluetooth 5.0)

kakra, 2020-05-28:
As long as the controller's light pulsates, it is not connected. Fast pulsate means "pairing mode", slow pulsate means "trying to connect".

These steps should work:

1. Turn the controller on
2. Enable pairing in Bluetooth
3. Hold connect on the controller until the LED pulsates fast
4. Scan in Bluetooth
5. Connect to the device in Bluetooth when found
6. The controller should immediately stop pulsating
7. Trust the device in Bluetooth

If the controller changes from fast to slow pulsate, the pairing mode timed out before you finished the above steps. If it changes to slow pulsate in step 5-6 instead of stopping pulsate, the connection was not successful. This may be due to a missing link key. Try purging the device from your Bluetooth database first before pairing it again, this usually solves the issue.

You can verify if the link key exists by going to `/var/lib/bluetooth/DONGLE-MAC/CONTROLLER-MAC` and looking at the `info` file. The following command should work:

    grep ^Key /var/lib/bluetooth/*/C8:3F:26:*/info

If you ever connected different BT dongles, you may see the key multiple times. A key should exist for the dongle+controller MAC pair you are actually using currently. You may want to delete the MAC directory of your controller and restart Bluetooth to start over from scratch. I needed to do that once to get the connection working again, somehow Bluetooth refused to add a link key. Without a link key you'd need to manually confirm the connection each time the controller is trying to connect.

I wonder why the modalias line is missing for you... My output:
```
# bluetoothctl
Agent registered
[CHG] Device C8:3F:26:A9:2D:6C Connected: yes
[Xbox Wireless Controller]# info
Device C8:3F:26:A9:2D:6C (public)
        Name: Xbox Wireless Controller
        Alias: Xbox Wireless Controller
        Class: 0x00000508
        Icon: input-gaming
        Paired: yes
        Trusted: yes
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v045Ep02FDd0903
[CHG] Device C8:3F:26:A9:2D:6C Connected: no
[CHG] Device C8:3F:26:A9:2D:6C Connected: yes
```

As you can see, the controller connects, then immediately disconnects and reconnects during initialization. That could actually explain the problem you're observing: When missing the link key and the controller immediately disconnects and reconnects after pairing, the new connection cannot be automatically established. You would need to manually issue another connect command.

bige11, 2020-05-29:
Thank you for your reply.

I looked into the `info` file located in `/var/lib/bluetooth/58:96:1D:94:E8:5E/C8:3F:26:3B:23:76` directory. The key was there:
```
[General]
Name=Xbox Wireless Controller 7EED803C5A1D
Class=0x000508
SupportedTechnologies=BR/EDR;
Trusted=true
Blocked=false
Services=0000111e-0000-1000-8000-00805f9b34fb;00001124-0000-1000-8000-00805f9b3>

[LinkKey]
Key=EAE20B2265D062DBE9F9209144D70C55
Type=4
PINLength=0

[DeviceID]
Source=2
Vendor=1118
Product=736
Version=2307
```
At any rate, I removed the controller directory, restarted the system, and repeated the connection procedure:
```
alvise@alvise-XPS-13-9300:~$ sudo bluetoothctl
[sudo] password for alvise: 
Agent registered
[CHG] Controller 58:96:1D:94:E8:5E Pairable: yes
[bluetooth]# scan on
Discovery started
[CHG] Controller 58:96:1D:94:E8:5E Discovering: yes
[NEW] Device 62:3A:54:7A:20:E0 62-3A-54-7A-20-E0
[NEW] Device 51:3C:96:26:39:B1 51-3C-96-26-39-B1
[NEW] Device 6F:AC:8B:F8:C9:37 6F-AC-8B-F8-C9-37
[NEW] Device 75:64:73:86:DA:F6 75-64-73-86-DA-F6
[NEW] Device 6B:76:07:5F:1F:56 6B-76-07-5F-1F-56
[NEW] Device 4C:EF:B7:9B:5A:86 4C-EF-B7-9B-5A-86
[NEW] Device 43:30:B1:AC:91:CA 43-30-B1-AC-91-CA
[NEW] Device 41:3A:23:FB:B9:65 41-3A-23-FB-B9-65
[NEW] Device 04:52:C7:B3:AB:60 LE-Bose Revolve SoundLink
[CHG] Device 75:64:73:86:DA:F6 RSSI: -70
[NEW] Device 6E:FA:0B:95:3D:B2 6E-FA-0B-95-3D-B2
[CHG] Device 75:64:73:86:DA:F6 RSSI: -83
[NEW] Device C8:3F:26:3B:23:76 Xbox Wireless Controller 7EED803C5A1D
[bluetooth]# scan off
Discovery stopped
[CHG] Controller 58:96:1D:94:E8:5E Discovering: no
[CHG] Device C8:3F:26:3B:23:76 RSSI is nil
[CHG] Device 6E:FA:0B:95:3D:B2 TxPower is nil
[CHG] Device 6E:FA:0B:95:3D:B2 RSSI is nil
[CHG] Device 04:52:C7:B3:AB:60 TxPower is nil
[CHG] Device 04:52:C7:B3:AB:60 RSSI is nil
[CHG] Device 41:3A:23:FB:B9:65 TxPower is nil
[CHG] Device 41:3A:23:FB:B9:65 RSSI is nil
[CHG] Device 43:30:B1:AC:91:CA TxPower is nil
[CHG] Device 43:30:B1:AC:91:CA RSSI is nil
[CHG] Device 4C:EF:B7:9B:5A:86 TxPower is nil
[CHG] Device 4C:EF:B7:9B:5A:86 RSSI is nil
[CHG] Device 6B:76:07:5F:1F:56 TxPower is nil
[CHG] Device 6B:76:07:5F:1F:56 RSSI is nil
[CHG] Device 75:64:73:86:DA:F6 TxPower is nil
[CHG] Device 75:64:73:86:DA:F6 RSSI is nil
[CHG] Device 6F:AC:8B:F8:C9:37 TxPower is nil
[CHG] Device 6F:AC:8B:F8:C9:37 RSSI is nil
[CHG] Device 51:3C:96:26:39:B1 TxPower is nil
[CHG] Device 51:3C:96:26:39:B1 RSSI is nil
[CHG] Device 62:3A:54:7A:20:E0 TxPower is nil
[CHG] Device 62:3A:54:7A:20:E0 RSSI is nil
[bluetooth]# pair C8:3F:26:3B:23:76
Attempting to pair with C8:3F:26:3B:23:76
[CHG] Device C8:3F:26:3B:23:76 Connected: yes
[CHG] Device C8:3F:26:3B:23:76 Modalias: usb:v045Ep02E0d0903
[CHG] Device C8:3F:26:3B:23:76 UUIDs: 0000111e-0000-1000-8000-00805f9b34fb
[CHG] Device C8:3F:26:3B:23:76 UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[CHG] Device C8:3F:26:3B:23:76 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device C8:3F:26:3B:23:76 ServicesResolved: yes
[CHG] Device C8:3F:26:3B:23:76 Paired: yes
Pairing successful
[CHG] Device C8:3F:26:3B:23:76 ServicesResolved: no
[CHG] Device C8:3F:26:3B:23:76 Connected: no
[bluetooth]# connect C8:3F:26:3B:23:76
Attempting to connect to C8:3F:26:3B:23:76
[CHG] Device C8:3F:26:3B:23:76 Connected: yes
Connection successful
[CHG] Device C8:3F:26:3B:23:76 ServicesResolved: yes
[Xbox Wireless Controller 7EED803C5A1D]# trust C8:3F:26:3B:23:76
[CHG] Device C8:3F:26:3B:23:76 Trusted: yes
Changing C8:3F:26:3B:23:76 trust succeeded
[Xbox Wireless Controller 7EED803C5A1D]# 
```
Once again, the terminal told me the procedure was successful, but the controller's light did not stop pulsating (fast, then slow), until it turned off altogether. If I try to reconnect the controller, I have to repeat the whole procedure (and I have the same results).
This is the new data of the `info` file (a new key is present):
```
[General]
Name=Xbox Wireless Controller 7EED803C5A1D
Class=0x000508
SupportedTechnologies=BR/EDR;
Trusted=true
Blocked=false
Services=0000111e-0000-1000-8000-00805f9b34fb;00001124-0000-1000-8000-00805f9b3>

[LinkKey]
Key=10D619C699A00DAC1544D1ED3E61D9E7
Type=4
PINLength=0

[DeviceID]
Source=2
Vendor=1118
Product=736
Version=2307
```
So, I am back to square one :/

kakra, 2020-05-29:
So to have a clear statement: If you issue the connect command twice, the connection is successful?

bige11, 2020-05-29:
Every time I issue the connect command, the terminal reports
```
[bluetooth]# connect C8:3F:26:3B:23:76
Attempting to connect to C8:3F:26:3B:23:76
[CHG] Device C8:3F:26:3B:23:76 Connected: yes
Connection successful
[CHG] Device C8:3F:26:3B:23:76 ServicesResolved: yes
```
but the controller's light keeps pulsating until it turns off completely. So, the connection is really not successful.

bige11, 2020-05-29:
Moreover, after the controller turned off, if I turn it on again and issue the connect command, the terminal reports
```
[bluetooth]# connect C8:3F:26:3B:23:76
Attempting to connect to C8:3F:26:3B:23:76
Failed to connect: org.bluez.Error.Failed
```
In order to have the terminal reporting that the connection is successful, as above, I have to put the controller into pairing mode again and then issue the connect command.

kakra, 2020-05-29:
Is your `bluetoothd` daemon working properly and stable? The message `Failed to connect: org.bluez.Error.Failed` should clearly never appear and is not related to a hardware or pairing issue, it indicates your `bluetoothd` died or disconnected from dbus.

bige11, 2020-05-30:
How can I check? (Apologies, I'm kind of new to linux.) At any rate, I've made a fresh install of Ubuntu a few days ago and I've been using my connection regularly with a bluetooth speaker; never had a problem with it. So, I suppose my computer's bluetooth is working properly.

Thanks again for your interest!

kakra, 2020-05-30:
Thinking about this again, it may just say that `bluetoothctl` received the reply from `bluetoothd` via dbus... If something crashes, you should see it in `dmesg` or `journalctl -eu bluetoothd`.

Also, please try running `bluetoothctl`, turn on the controller but do NOT send a `connect` command. Since the link key is established, the controller should be able to automatically connect. You should be able to see this by just running `bluetoothctl` and letting it run idle (without typing any commands). If the issue still exists, we'd need to dig a little further.

bige11, 2020-05-30:
I run `bluetoothctl` and turned on the controller, but nothing happens: the controller just keeps pulsating and the terminal does not give any feedback (if I do the same thing with my bluetooth speaker the terminal confirms the connection and I can stream the audio).

kakra, 2020-05-30:
While there is a general problem with the controller not fully compatible with many Bluetooth dongles, this is most likely an issue that can be fixed by just using a different dongle.

To completely reset the controller, connect it to a different Bluetooth adapter (i.e. your mobile, or another computer), then pair again with the previous system. I also recommend to upgrade the firmware from a Windows computer. This also seems to reset the Bluetooth state and you'd need to pair the controller again.

If all this does not work, then try a different dongle. I'm having good results with this:
https://www.amazon.de/dp/B07NQ5YGDW?psc=1&ref=ppx_pop_dt_b_product_details

But that may totally depend on kernel version and probably also USB chipset. I'm running kernel 5.4 LTS.

envyniv, 2020-05-30:
same issue as op here, i recently updated the kernel, may that be it?

kakra, 2020-05-31:
> same issue as op here, i recently updated the kernel, may that be it?

Did it work before?

bige11, 2020-05-31:
> To completely reset the controller, connect it to a different Bluetooth adapter (i.e. your mobile, or another computer), then pair again with the previous system. I also recommend to upgrade the [...]

I successfully connected the controller to my phone, and then tried again with my pc, but it didn't work. I don't have access to a windows machine now - as soon as I do I'll update the firmware and see if it works...

envyniv, 2020-06-02:
> > same issue as op here, i recently updated the kernel, may that be it?
> 
> Did it work before?

yes, it did.

kakra, 2020-06-02:
Then it is probably related to a change in the kernel. Since I cannot reproduce it, someone else may want to try a bisect between the working kernel and the non-working kernel (point 0 releases, or otherwise the bisect may be unpredictable).

kakra, 2020-06-02:
Please refer to #198.

zethra, 2020-12-06:
I'm having this issue too.

alvilag, 2021-07-07:
Hello, I'm coming back to this issue a year later just to let you know that in this time frame I changed a few distros with different kernel versions and the controller didn't connect properly on any of them--just as described above. Today, finally, I had the chance to upgrade the controller's firmware on a Windows machine via usb and now the controller works perfectly fine on Linux (tested with Kubuntu 21.04). So I guess it was not a problem related to kernel versions. I don't know if this information is still relevant to you, but I thought I should share it in case it might be. 

In any case, thanks again for the help and attention!

kakra, 2021-07-07:
Microsoft should really provide a way to upgrade the controller software through Linux, maybe by using fwupd. That would save us a lot of headache.

@all, please upgrade your firmware first before reporting problems. Thanks. ;-)

closed 2021-07-07 by kakra