atar-axis/xpadneo #198: Bluetooth connection stability broken after kernel update

kakra, 2020-06-02:
This is a information collection issue to pinpoint the kernel change that broke the Bluetooth stability with the controller. I'll ask anyone who wants to help to provide the following information:

**Please provide kernel versions as shown by `uname -a` if possible because distributions often implement their own patchlevel versioning.**

1. Which distribution are you running?
2. Last known working kernel version
3. First known broken kernel version
4. Output of `lsusb` and take note of the device number of your USB dongle
5. File generated with `lsusb -v -s## | tee xpadneo-lsusb.txt` where `##` is the device number picked in the previous step
6. Disconnect your controller, run `sudo btmon | tee xpadneo-btmon.txt` and connect the controller, post the resulting file.

Thanks!

envyniv, 2020-06-03:
1. Ubuntu
2. I'm not sure, but i think any 5.3?
3. current; 5.4.0-33-generic
4/5. [xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/4722057/xpadneo-lsusb.txt)
6. [xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/4722058/xpadneo-btmon.txt)
hope i managed to help in some way. :)

envyniv, 2020-06-08:
i know i shouldn't chat here, but am i the only one that seeme to corelate this issue with the kernel update?

kakra, 2020-06-08:
I think this will get some attention with the next version bump.

kakra, 2020-06-18:
> So ,will next version support xbox-one-s? ( it just couldn't connected pc(ubuntu-1804) through bluetooth)

I don't want to be rude but this issue is solely about collecting information with regressions. Please look at the other issues, the one's linked above are a good starting point. Otherwise feel free to open a new issue report. I'll mark off-topic comments as off-topic now. Thanks in advance for understanding and your cooperation.

kakra, 2020-06-20:
## HEADS UP

Anyone affected may tray #204 and report in #189 or #180.

envyniv, 2020-06-25:
> ## HEADS UP
> 
> Anyone affected may try #204 and report in #189 or #180. After the latest discoveries, this may prominently affect people with high-HZ kernel configurations (a 1000 Hz kernel seems to show the problem much more likely than a 100 Hz kernel, I'm using 300 Hz and didn't have many issues but many game-optimized kernels often opt-in for 1000 Hz, tho I think that's overkill, 300 Hz may looks odd but fits best here as it cleanly divides by most common FPS rates, 25 Hz, 30 Hz, 50 Hz, 60 Hz, 100 Hz).

kernel hz? how can i check for kernel hz?

envyniv, 2020-07-14:
[mark this as off-topic]
after a month, I decided to retry connecting my controller to my computer. it now works. I have no idea what has changed. i didn't update the driver, I, pretty much, did not do anything. my guess is that the latest kernel patched the issue? I don't know anything about anything. So sorry i couldn't help.

kakra, 2020-07-14:
There's also an issue with establishing an encrypted connection between the controller and the Linux Bluetooth stack. The first problem is (from my tests) often a missing link key. This can be fixed by retrying to remove the controller in `bluetoothctl` and then pairing it again, immediately followed by `trust`, repeat 1-3 times until the link key shows up in `/var/lib/bluetooth/HOST-MAC/PEER-MAC/info`. During pairing you may need to press buttons on the controller (or click the connect button again) to confirm, but that may also be just placebo - at least clicking the connect button sends data to the Bluetooth stack for some reason. Once the link key is there, it now longer constantly disconnects and reconnects due to a missing or non-matching link key. But still, the controller may take up to 60s for establishing the encryption. I don't know whats wrong but often the handshaking fails until it eventually works after 3-5 times. I wonder if the firmware tries multiple encryption methods until it finds one that works with Linux. It may well be related to the kernel or the bluez version after the initial link key issue was resolved.

Will mark as off-topic later.

envyniv, 2020-07-18:
ight that's good to know, if this problem arises again and the fix you've proposed doesn't work, i'll be sure to hit this up again.

richardtatum, 2020-10-15:
I have been meaning to post this for a while, which means my data is somewhat out of date, so I can only apologise.

1. Arch Linux 5.8.14-zen1
2. 5.8.10-ish?
3. 5.8.10-ish onwards
4. `CONFIG_HZ_1000=y CONFIG_HZ=1000`
5.  [xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/5383157/xpadneo-lsusb.txt)
6.  [xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/5383180/xpadneo-btmon.txt)
7. Whether or not this is helpful I do not know, but [here](https://github.com/atar-axis/xpadneo/files/5383183/xpadneo-bluetoothctl.txt) is the bluetoothctl logs for trying to connect whilst piping the above btmon output.
- ERTM is disabled.

After updating the kernel, it started connecting very slowly (30s+) then stopped connecting altogether. I switched to windows and updated the firmware on the controller, then switched back to linux and I was no longer able to connect via bluetooth. Connecting via bluetoothctl usually end in the same way, I remove the controller from paired devices, I scan for it (sometimes it takes a very long time to show up, or does not show at all and I have to turn the controller off/on and put back into pairing mode), trust, pair and attempt to connect (which always fails).

If you need anything further, please let me know.

kakra, 2020-10-15:
I'm seeing a similar problem after upgrading to latest 5.4.y: firmware was already up to date for me but I had a lot of connection dropouts and establishing connection is very slow (30+ seconds), so I removed the controller from Bluetooth and paired it again, and now connection never happens automatically, I always need to use "connect" command manually (or via GUI). But at least it seems stable once connected. The XBE2 controller acts very differently: It absolutely has no problem establishing the Bluetooth connection (so there seems to be no general kernel problem), but it stalls in the HID layer for around 20s before xpadneo finally initializes it (probably due to its huge HID descriptor), then it drops the connection as soon as the first few real rumble packets are sent by games (the notify rumble works, tho).

There are some unofficial logs about XB1S firmware upgrades, some of them contain notes about improving Bluetooth connection quality and fixing issues with connecting to some Android devices. So there are definitively changes in the Bluetooth part of the controller.

Initially found here: https://answers.microsoft.com/en-us/xbox/forum/all/new-release-of-controller-update-23-october-2019/4ce5abe6-86f8-4490-bda6-c3943c31713e

Then followed the link to "ASA Insider": https://news.xbox.com/en-us/2019/10/23/xbox-insider-release-notes-alpha-skip-ahead-ring-2004-191021-2100/

From there it gets quite messy to follow the update logs but it lists a lot of Xbox firmware updates which also include the controller firmwares.

I think there kernel is missing some bits and pieces that the controller actually expects (and which Android may do differently), as the controller seems to have stable Bluetooth connections with both Windows and Android. OTOH, I believe the Bluetooth implementation in the controller firmware is not very well done (and not very tolerant for deviating from implementation expectations).

kakra, 2020-10-18:
@richardtatum 
```
[CHG] Device 5C:BA:37:D6:DE:E9 ServicesResolved: yes
[CHG] Device 5C:BA:37:D6:DE:E9 Paired: yes
Pairing successful
[CHG] Device 5C:BA:37:D6:DE:E9 ServicesResolved: no
[CHG] Device 5C:BA:37:D6:DE:E9 Connected: no
```

Yeah there's something happening that immediately kicks the controller again (or it kicks itself). This may end in an endless cycle of auto-connecting and then immediately disconnecting again. Only host-initiated connection attempts do seem to work currently, at least most of the times, sometimes you'll have to retry and hit the right timing by luck.

kakra, 2020-10-18:
## HEADS UP

Kernel 5.4.72 brought some patches regarding Bluetooth pairing. All chronology later kernels (5.8.16, 5.9.1) should have the same patches. For my controller, this doesn't make any difference but you may want to try.

richardtatum, 2020-10-19:
Thanks @kakra. I thought I'd try with the latest update (5.9.1-zen2 for me) but no luck. After my initial edit of being able to connect to the controller by omitting the 'pair' step, I have not been able to do it since. Now I am at a stage where the controller does not actually show up in the discovered devices in `bluetoothctl` when the scan is on. For now I am just using the controller in wired mode until there are some further patches to the kernel.

kakra, 2020-10-19:
Scan usually doesn't show your device if it is already known. Use `devices` in `bluetoothctl` to find the MAC address to remove, then use `remove <MAC>`. Now, with `scan on` the controller should show up, you may need to hold the connect button so it doesn't try to reconnect to host it is no longer associated with. Worked for me when I ran into that situation.

Status: Will mark off-topic later as this is not for discussions.

schmitmd, 2020-10-19:
Confirming. My XBES2 controller was suffering from the 20s delay on initial connect. Since updating to kernel v5.9.1, connection is now basically instantaneous.

kakra, 2020-10-19:
@schmitmd I cannot confirm this for 5.4.72. But the actual commits tell that this pairing problem only came apparent after some Bluetooth changes introduced with 5.9. So older kernels may miss some other changes. I'm currently not updating to 5.9 to do any confirmations because I'm waiting for it going LTS which will be around new year. Since kernel development is around 2 months early now compared to the previous years, even 5.10 may become LTS.

kakra, 2021-01-01:
The kernel Bluetooth stack solved a bug for BLE devices in privacy mode: https://bugzilla.kernel.org/show_bug.cgi?id=209745, seen since 5.9, solved since 5.10.4, not sure if it has been backported yet. Updated topmost post accordingly.

kakra, 2021-01-01:
> Confirming. My XBES2 controller was suffering from the 20s delay on initial connect. Since updating to kernel v5.9.1, connection is now basically instantaneous.

I can confirm the same for kernel 5.10.4 (upgraded from 5.4.85).

kakra, 2021-01-08:
To all participants of this thread: Do you still see issues with kernel 5.10.4 or later?

richardtatum, 2021-02-16:
Sorry for the incredibly late response on this, but yes. I am still having the exact same issues on 5.10.16.

kakra, 2021-02-17:
Yes, it's important to pair the controller to a different machine and remove it from the Bluetooth stack. Otherwise one partner may fall back to a partially broken and old state when re-pairing.

kakra, 2021-04-09:
I've added bug reports to the initial report above to cover instabilities due to continuous discovery.

kakra, 2021-04-12:
Identified two possible problems for connection stability and initial connection: `kdeconnectd` is somehow involved as we see continuous Bluetooth discoveries going on, overwhelming the RF environment, resulting in the controller losing connection. Also, the initial connect may depend on proper ordering of some HCI events but this needs further investigation. I've updated the initial post on this issue.

kakra, 2021-06-20:
Updated initial post regarding Bluetooth LE issues.

1985a, 2021-09-29:
In my case, this is the solution
Xbox controller model **1914**
In the main.conf Bluetooth file add the following

```
[General]
ControllerMode = Dual
JustWorkingRepairing = confirm

[LE]
MinConnectionInterval = 7
MaxConnectionInterval = 9
ConnectionLatency = 0

```

kakra, 2021-10-05:
Some general updates on the topic since I was finally able to test the issue on a Windows system which had multiple Bluetooth dongles connected (that system is extensively used for VR, and the user connected all the Bluetooth dongles that came with the various accessories):

* Devices become really confused when multiple CSR 4.0 dongles are connected
* The PC has an internal Intel AX200 Bluetooth transceiver
* We had to remove all but one CSR 4.0 dongle, Windows doesn't seem to handle multiple dongles very well, the GUI doesn't expose a way to select which dongle to pair to
* We had to disable the internal AX200 Bluetooth device, none of the various Bluetooth devices could pair reliably to it
* With only one Bluetooth transceiver left and enables, devices connected just fine and immediately

The tests included an XB1S controller which we updated to the latest firmware first. Connection with the internal AX200 Bluetooth was almost always impossible, the connection was unreliable, and most of the time it connected in some limbo state where Windows would say "connected" but the controller keeps flashing. This reminds me a lot of the behavior we can see in Linux. The other Bluetooth devices had similar issues: While they usually were able to connect, pairing and connected took a long time, some devices needed several tries before they worked. This results in the conclusion that the AX200 chipset is just borked and incompatible. Don't use it.

With only one CSR 4.0 dongle connected, all devices were able to pair and connect almost instantly. Even the XB1S controller connected instantly. This works a lot better than I'm used to it in Linux: In Linux, it CAN connect instantly, but often it takes several seconds. This results in the conclusion that there are still issues with the bluez daemon.

The kernel drivers for the chipsets may have an impact on that but I think most of the drivers are using btusb anyways, the driver should be in good shape, and only some new chipset revisions or cheap China clones may need btusb driver updates or quirk fixes. This leads to the conclusion that you want to use CSR chipsets most likely because those are supported very well and work with the gamepad. There's only the problem with a lot of CSR China clones on the market which you should try to avoid (but may not be possible because who knows).

1985a, 2022-08-03:
I don't know if someone can see help with this but I felt the need to share it.
I'd like to say at the moment the kernel 5.18.15 and these devices, are working together so nice.

Before, I made several attempts in order to get this controller to work on BT mode, but it never any success.

This is my Bluetooth main conf file, that I'm using, and as I say, it's working like some kind of magic after updated the firmware.

PD: excuse me my bad English, it is not my mother language.
```
[General]
ControllerMode = dual
JustWorksRepairing = confirm
[BR]
[LE]
MinConnectionInterval=7
MaxConnectionInterval=9
ConnectionLatency=0
MinConnectionInterval=9
MaxConnectionInterval=12
[GATT]
[AVDTP]
[Policy]
[AdvMon]

```

dtantono, 2022-10-23:
same issue, it does not work on Ubuntu 22.04