atar-axis/xpadneo #203: Connected Xbox One controller has faulty right analog stick range

CrashOveride95, 2020-06-16:
## Describe the bug
When connecting an Xbox One controller  and testing the controls in ``jstest-gtk``, all controls work correctly except for the right analog stick and the triggers.  The right analog stick's controls will always be on the left, and moving the stick moves it up and down. Pressing the triggers will make it move right and left.

## Steps to Reproduce
Connect a Xbox One controller via ```bluetoothctl``
Open ``jstest-gtk``
Test the control functions.

## Expected behavior
I expected that the right analog stick would behave as normal, staying right in the center.

## Screenshots/Gifs
[image]
This GIF shows me first moving the left analog stick, then pressing the triggers, then moving the right stick, then pressing the buttons.

## System information

```console
# uname -a
Linux crash-rogstrixg731gug731gu 5.4.0-37-generic #41-Ubuntu SMP Wed Jun 3 18:57:02 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0b 15 00 25 01 75 01 95 0b 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 05 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81  u........$...%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05  ...%.u..............
000000c8: 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07  ..#...%...u.....%.u.
000000dc: 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75  ........!........%.u
000000f0: 04 95 01
[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/4788243/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/4788244/xpadneo-dmesg.txt)

 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25  .......%.u......p..%
00000104: 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75  du......Pf..U...&..u
00000118: 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00  .........&..u.....e.
0000012c: 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 05 06 09 20  U..|..&..u......... 
00000140: 85 04 15 00 26 ff 00 75 08 95 01 81 02 c0                    ....&..u......
1461450476 1558

```

## Controller and Bluetooth information

I am using the bluetooth hardware in my laptop. It's worked fine before in Windows.

## Additional context

CrashOveride95, 2020-06-16:
[xpadneo-btmon.txt.gz](https://github.com/atar-axis/xpadneo/files/4788255/xpadneo-btmon.txt.gz)
[xpadneo-dmesg.txt.gz](https://github.com/atar-axis/xpadneo/files/4788256/xpadneo-dmesg.txt.gz)

kakra, 2020-06-16:
This looks familiar... Seems the legacy JS interface got broken. Could you test with `evtest` and see if the axes work correctly there?

BTW: Could you repair your initial post? It seems you accidentally dropped the files right into the xxd dump.

CrashOveride95, 2020-06-17:
I have fixed my initial post, sorry about that.

I tried ``evtest``, and I got results, however I am having a hard time figuring out if it's actually working. In programs such as RetroArch the axes do not work.

Atrate, 2020-06-21:
Can also reproduce without the xpadneo driver

kakra, 2020-06-21:
`jstest-gtk` uses the old legacy joystick interface, this expects axes to have a very well defined order - and many mondern controllers just don't do that, they have HID descriptors for the purpose of each axis. The old joystick API just takes whatever it finds in order of event codes, I believe. I'm not sure what we can do about it from within `xpadneo` but I believe it worked right at some point.

The problem is that the event interface of this controller has `X/Y` (left stick), `Rx/Ry` (right stick) and 'Z/Rz` (triggers) in that order for the 3 axis. But the event numbers for these order them to `X/Y/Z` and `Rx/Ry/Rz`, so it ends up in the joystick interface as `X/Y`, `Z/Rx`, `Ry/Rz`. The left stick is correct, the right stick has its X axis shifted to Y and uses the left trigger as X axis instead, and the left trigger is messed up, too, and reacts to the right Y axis.

The HID interface defines `X/Y/Z` as the three axis for movement, and `Rx/Ry/Rz` as the three axis for camera rotation. This is fine for a 3D controller, but applying this to a game controller and games is left up to interpretation. Actually, the Linux interpretation is at least somewhat correct for this:

When looking at a classical FPS/3rd person game, from a birds-eye view, the left stick controls the players X and Y position. The right stick rotates the around the Z axis (so it should Rz here) and the X axis (Rx is correct), but the triggers are usually not part of any movement or rotation, so we are left with Rz (tho it should be Z) and Ry. So in theory it should be `X/Y/Rz/Rx/Z/Ry`. The controller exposes the triggers as `Brake/Accel` in Linux mode. But no game expects these.

But from intuition this makes no sense at all. So Windows uses `X/Y` and `Rx/Ry` (as some sort of right X and right Y) and the two axis remaining are for the triggers, while Linux simply follows the logical order `X/Y/Z/Rx/Ry/Rz`.

The event interface does not do a very good job of giving actual implementation examples for a modern games control (just old crappy ones with a dpad and four buttons). But it is quite clear about the purpose of each axis: Rz = turn left/right, Rx = pitch forward/backward, Ry = roll left/right.

I think the binding of controls to physical actions is one of the biggest problems of HID here. While it makes sense to let a software know what each value is supposed to control, it cannot be applied very well to general purpose game controllers. After all: Who says what an axis actually does? It may change from game to game.

Microsoft introduced another level of mess by putting a special Linux mode into the controller: It presents us `X/Y` for the left stick, the a blind axis `Z` which does nothing, the `Rx/Ry` for the right stick (so it matches with Windows), another blind axis `Rz`, and then `Brake/Accel` for the triggers. The old joystick API then sees 8 axis but we only have 6. The joystick API has no knowledge about the purpose of each axis, it just presents a game 8 axis then. While this Linux mode matches mostly with the HID specification (except that Rx/Ry do not match turn/pitch but pitch/roll), this completely messes up the JS API. So my latest adjustments removed the blind axes by patching the HID reports, and maps Brake/Accel back to Z/Rz. But since the joystick API maps that in the order of `X/Y/Z/Rx/Ry/Rz` its just broken.

I'll have to read the joystick API code to see if it has other ways of deriving the order of axes. For all modern games, the controller mapping works correctly.

I'm not sure which API RetroArch uses, it probably shouldn't use the old legacy JS API if it still does. There's a second problem: HID says the idle point of controller sticks has to be (0,0) but the controller doesn't use that. It uses a range of (0..65535) and (32767,32767) is the idle point. But that's against HID specification and may confuse some games (in particular, it confuses some Wine games). So maybe that's your problem? We already have #57 for that.

@CrashOveride95 So if you want things to become fixed, please ensure your test program uses the same API as the application you're having problems with - otherwise the test is quite meaningless as you can probably see from the text outlined above.

CrashOveride95, 2020-06-22:
[image]
After trying to change input driver in RetroArch from ``udev`` to ``sdl2`` it appears that this is a RetroArch issue
I will ask in their relevant support channels for help there, since this appears to not be an issue with ``xpadneo`` I will be closing this issue.

CrashOveride95, 2020-06-22:
I was able to fix it completely. 
For future people that may have the same issue: changing input driver to ``sdl2`` and downloading the controller configuration files fixes it for me.

kakra, 2020-06-22:
I still understand what the problem was: Your initial post was about wrong axes mapping when using the `js*` device... Did RetroArch have the same problem? Or did input not work at all there?

It RetroArch input did not work at all, that's probably indeed a RetroArch issue. If it just had wrong axes mapping, it's an issue we should try to fix because it would affect all games using the `js*` device (instead of `input*`)

CrashOveride95, 2020-06-22:
I had used jstest as I was unaware of any other test programs ~~it was the first google result for "Linux joystick test program"~~. RetroArch had originally incorrect inputs, so I had assumed that it was an issue with the driver. I was unaware with the situation with how different joystick devices work in Linux, as you explained in your reply.

kakra, 2020-06-22:
Did jstest and RetroArch show the same misbehavior?

CrashOveride95, 2020-06-22:
In RetroArch the right stick would always give incorrect inputs, since I saw it misfunctioning in jstest I assumed it was the same issue.

kakra, 2020-06-22:
Yes, then it is the same issue. With SDL2, you've found a work-around (because SDL2 is not using the js interface but the event interface). I'll re-open this.

kakra, 2020-06-26:
Please run the following command to correct the axes mapping for the `js` interface:

    jscal -u 8,0,1,3,4,2,5,16,17,10,304,305,307,308,310,311,314,315,317,318 /dev/input/js0

Now test all your axes to collect calibration data. You can then use the following command to store the settings permanently:

    sudo jscal-store /dev/input/js0

If the gamepad does not restore the mapping after disconnecting and reconnecting it, i.e., your distribution doesn't ship a proper udev rule for that, you may want to add the this udev rule, then reboot:

    # /etc/udev/rules.d/99-joystick.rules
    KERNEL=="js*", ACTION=="add", RUN+="/usr/bin/jscal-restore %E{DEVNAME}"

From now on, connecting the gamepad should restore the values from `/var/lib/joystick/joystick.state`. If you messed up, simply remove your gamepad from the file and start over.

Please check, if `jscal-restore` is really in `/usr/bin`, otherwise use the correct path: `type -p jscal-restore`.

Does this fix it?

kakra, 2020-06-27:
I tried making this automatic but that messes up the Chrome Gamepad API (and then the problem moved over to Stadia games). I'll leave that in my patch queue for consideration.

CrashOveride95, 2020-06-27:
Running  
``jscal -u 8,0,1,3,4,2,5,16,17,10,304,305,307,308,310,311,314,315,317,318 /dev/input/js0``

and 

```sudo jscal-store /dev/input/js0``` fixed my issue. 

```jstest-gtk``` and RetroArch work perfectly.

I did not need to make a new udev rule, disconnecting and reconnecting

kakra, 2020-06-27:
@CrashOveride95 Thanks for not closing it... ;-) Yes, please leave it open until I fix this with a documentation update or by some other update.

closed 2020-07-11 by kakra

jldelsolare, 2023-09-30:
> Running `jscal -u 8,0,1,3,4,2,5,16,17,10,304,305,307,308,310,311,314,315,317,318 /dev/input/js0`
> 
> and
> 
> `sudo jscal-store /dev/input/js0` fixed my issue.
> 
> `jstest-gtk` and RetroArch work perfectly.
> 
> I did not need to make a new udev rule, disconnecting and reconnecting worked perfectly

I have the following error with my Xbox Series controller when performing the described process:
```shell
jscal -u 8,0,1,3,4,2,5,16,17,10,304,305,307,308,310,311,314,315,317,318 /dev/input/js0
jscal: joystick has 15 buttons and not 10 as specified on command line
``` 
When attempting to add the buttons, the issue persists:
```shell
jscal -u 8,0,1,3,4,2,5,16,17,10,304,305,307,308,310,311,314,315,317,318,319,320,321,322,323 /dev/input/js0
jscal: joystick has 15 buttons and not 10 as specified on command line
``` 
Can I provide more information?
Kubuntu: 22.04

kakra, 2023-09-30:
> When attempting to add the buttons, the issue persists:

You need to replace the `10` by `15` in den command line. But this is not a xpadneo issue because xpadneo only has 10 buttons.