atar-axis/xpadneo #209: DKMS install fails on raspbian 4.19.118

dkumor, 2020-06-28:
## Describe the bug
Running the install script as recommended fails on the Pi 4

## Steps to Reproduce
```
git clone https://github.com/atar-axis/xpadneo.git
cd xpadneo
sudo ./install.sh
```

This gives the following output:
```
* replacing version string if necessary
* copying module into /usr/src
* adding module to DKMS

Creating symlink /var/lib/dkms/hid-xpadneo/0.7/source ->
                 /usr/src/hid-xpadneo-0.7

DKMS: add completed.
* installing module (using DKMS)

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.19.118-v7l+ -C /lib/modules/4.19.118-v7l+/build M=/var/lib/dkms/hid-xpadneo/0.7/build/src modules....(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.118-v7l+ (armv7l)
Consult /var/lib/dkms/hid-xpadneo/0.7/build/make.log for more information.
```

The contents of `/var/lib/dkms/hid-xpadneo/0.7/build/make.log` are:
```
DKMS make.log for hid-xpadneo-0.7 for kernel 4.19.118-v7l+ (armv7l)
Sat 27 Jun 2020 09:18:34 PM EDT
make: Entering directory '/usr/src/linux-headers-4.19.118-v7l+'
  CC [M]  /var/lib/dkms/hid-xpadneo/0.7/build/src/hid-xpadneo.o
/var/lib/dkms/hid-xpadneo/0.7/build/src/hid-xpadneo.c: In function ‘xpadneo_ff_play’:
/var/lib/dkms/hid-xpadneo/0.7/build/src/hid-xpadneo.c:414:3: error: implicit declaration of function ‘hid_notice_once’; did you mean ‘dev_notice_once’? [-Werror=implicit-function-declaration]
   hid_notice_once(hdev, "throttling rumble reprogramming\n");
   ^~~~~~~~~~~~~~~
   dev_notice_once
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:310: /var/lib/dkms/hid-xpadneo/0.7/build/src/hid-xpadneo.o] Error 1
make: *** [Makefile:1522: _module_/var/lib/dkms/hid-xpadneo/0.7/build/src] Error 2
make: Leaving directory '/usr/src/linux-headers-4.19.118-v7l+'

```

## Expected behavior
I expected there to be no error, and the install to succeed

## System information

```console
# uname -a
Linux mediacenter 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
xxd: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor: No such file or directory
```

## Additional context

This is a clean install following instructions from the readme - I would appreciate any guidance on how to get this working!

kakra, 2020-06-28:
It looks like kernel 4.19 is missing the `hid_notice_once` macro... I'll add it.

kakra, 2020-06-28:
Please try with #210

dkumor, 2020-06-29:
#210 worked, thanks!

closed 2020-06-29 by kakra