atar-axis/xpadneo #212: Incorrect Xbox one s wireless controller battery display

Mo0nCake, 2020-07-01:
Incorrect Xbox one s wireless controller battery display. 3 days showed 66% of the battery charge. On day 4 it became 5%. After charging, the battery showed 99% and after a while again 50%.
It was tested on versions 0.7 and 0.7.4.

My system:
Arch linux + KDE Plasma 5.19.2

[image]
[image]
[image]
[image]
[image]

kakra, 2020-07-01:
Do you use standard batteries or the official charge kit?

Mo0nCake, 2020-07-01:
> Do you use standard batteries or the official charge kit?

Standrt batteries, which can be charged.

kakra, 2020-07-01:
I didn't know about `upower --dump`, looks handy to finally fix the battery properly. I'll be looking at it.

Mo0nCake, 2020-07-01:
> I didn't know about `upower --dump`, looks handy to finally fix the battery properly. I'll be looking at it.

Ok. Thank you, I look forward to fixing)

kakra, 2020-07-01:
> I have one more bug: if the gamepad vibrates a lot in the game, then it freezes and you need to restart it. This happens often.

This should be fixed in master, please try again with the git master branch.

BTW: Battery 50% states some unknown state... We have only 4 levels: 99%, 66%, 33% or 5% (which is 0% actually but 5% makes the desktop show "critical"). We cannot poll the battery, the controller tells us. Before we receive the first battery update, we set the battery level to 50%. So right after connecting, it would read 50 but it should update within a few seconds.

But looking at git master, the 50% rule is no longer there. So this may also be fixed in master (which will become v0.8 soon). Could you try master?

Mo0nCake, 2020-07-01:
> This should be fixed in master, please try again with the git master branch.
>
> BTW: Battery 50% states some unknown state... We have only 4 levels: 99%, 66%, 33% or 5% (which is 0% actually but 5% [...]

At the expense of master did not understand. Tell me how to install / update correctly ?! I have xpadneo 0.7.4 now.

kakra, 2020-07-01:
I just looked through the code again: We still initialize with 50% capacity instead of unknown to work around a bug in desktop notifications, and unless the first report was received from the controller we stay there. So it will always jump from 50% to some capacity after a few seconds. It should never jump TO 50%, tho, unless the controller disconnected.

For getting master, run

`git remote update && git reset --hard origin/master` (you need to do this only once)

Then it should show a version number like `v0.7-XX` (with a dash and XX being some higher number).

kakra, 2020-07-01:
If you downloaded the 0.7.4 zip, then use another download link: https://github.com/atar-axis/xpadneo/archive/master.zip

Mo0nCake, 2020-07-01:
> If you downloaded the 0.7.4 zip, then use another download link: https://github.com/atar-axis/xpadneo/archive/master.zip - but that will incorrectly show you version 0.7 despite being much newer.

Don`t install and don`t update. Nothing happens
[image]

kakra, 2020-07-01:
Ah, yes, that bug... See the difference in version number: one has a v, one doesn't. I still need to fix that.

Could you prefix your commands with `LANG=C` as I can't read Cyrillic, e.g., `sudo LANG=C ...`. Your update command shows an error I cannot identify.

Hey, I found my phone could translate photos, it says "this is not a directory" :-D

Mo0nCake, 2020-07-01:
> "this is not a directory"

yes "this is not a directory"

Mo0nCake, 2020-07-01:
> > "this is not a directory"
> 
> yes "this is not a directory"

Please tell me when fix, so that I can install master

Mo0nCake, 2020-07-01:
> > > "this is not a directory"
> > 
> > 
yes "this is not a directory"
Please tell me when fix, so that I can install master

kakra, 2020-07-01:
That's from this line in `lib/installer.sh`:
```bash
git describe --tags --dirty 2>/dev/null || cat "$(dirname "${BASH_SOURCE[0]}/../VERSION")"
```

If you change that line to
```bash
git describe --tags --dirty 2>/dev/null || cat "$(dirname "${BASH_SOURCE[0]}")/../VERSION"
```

it should work...

Mo0nCake, 2020-07-01:
> That's from this line in `lib/installer.sh`:
> 
> ```shell
> git describe --tags --dirty 2>/dev/null || cat "$(dirname "${BASH_SOURCE[0]}/../VERSION")"
> ```
> 
> If you change that line to
> 
> ```shell
> git describe --tags --dirty 2>/dev/null || cat "$(dirname "${BASH_SOURCE[0]}")/../VERSION"
> ```
> 
> it should work...

I replaced this line, but still does not install anything.
[image]

kakra, 2020-07-01:
I cannot fix this currently for the zip downloads. Please pull from git:
```console
cd $HOME && git clone https://github.com/atar-axis/xpadneo.git
```

It will create new directory `xpadneo` and it should work there. You may need to install git first from your distribution.

Mo0nCake, 2020-07-01:
> I cannot fix this currently for the zip downloads. Please pull from git:
> 
> ```
> cd $HOME && git clone https://github.com/atar-axis/xpadneo.git
> ```
> 
> It will create new directory `xpadneo` and it should work there. You may need to install git first from your distribution.

cd $ HOME && git clone https://github.com/atar-axis/xpadneo.git
Updated via sudo ./update.sh and now version 0.7.38. To be honest, I don’t understand anything at all) How can I install the master?)

kakra, 2020-07-01:
It's version `0.7-38` (note the `-` instead of `.`) - that tells you it's 38 development versions above `0.7`, that's the current state of master.

Mo0nCake, 2020-07-01:
> It's version `0.7-38` (note the `-` instead of `.`) - that tells you it's 38 development versions above `0.7`, that's the current state of master.

That is, everything turned out right and I installed master?)

kakra, 2020-07-01:
Yep... When we update the repository you just run `git pull origin` and it will fetch the latest updates without downloading a ZIP. Then you just run the update script again.

Now the rumble problem should be gone.

Mo0nCake, 2020-07-01:
> Yep... When we update the repository you just run `git pull origin` and it will fetch the latest updates without downloading a ZIP. Then you just run the update script again.
> 
> Now the rumble problem should be gone.

Ok, thank you very much. If there is a bug with vibration, I will write)

kakra, 2020-07-01:
Yes, please... The fix wasn't tested by many people yet. Rumble may feel a little different after this update, I'm probably fixing that in v0.8.

About the battery: If it's just showing 50% after connect, that's expected. It should change a few moments later to the correct value.

kakra, 2020-07-01:
@Mo0nCake It looks like you opened a Github account just for reporting this bug.

So: Welcome to the community. I hope your first experience was a good one. Thanks for your contribution, testing, and feedback. We need more people taking this opportunity. :-)

Mo0nCake, 2020-07-01:
> @Mo0nCake It looks like you opened a Github account just for reporting this bug.
> 
> So: Welcome to the community. I hope your first experience was a good one. Thanks for your contribution, testing, and feedback. We need more people taking this opportunity. :-)

You guessed it) Thank you, I will be glad to join the community)

kakra, 2020-07-01:
BTW: You should have trigger rumble now in every game.

Mo0nCake, 2020-07-06:
> BTW: You should have trigger rumble now in every game.
Missing xpadneo module

[image]
[image]

kakra, 2020-07-06:
This is a different issue, you should file a new bug report for that. When doing so, please include the output of `dkms status`.

closed 2020-07-11 by kakra