atar-axis/xpadneo #217: Missing xpadneo module.

Mo0nCake, 2020-07-06:
Missing xpadneo module. If installed, it says that it is, but in fact it is not.
[image]
[image]
[image]
[image]
[image]

Mo0nCake, 2020-07-06:
> Missing xpadneo module. If installed, it says that it is, but in fact it is not.
> [image]
> [image]
> [image]
> [image]
> [image]

kakra, 2020-07-06:
Yeah, something looks seriously broken in your DKMS database. I don't know yet where it comes from. There is a colon in the driver version that doesn't belong there.

Could you `ls -al /var/lib/dkms /var/lib/dkms/hid-xpadneo` please?

Mo0nCake, 2020-07-06:
> Yeah, something looks seriously broken in your DKMS database. I don't know yet where it comes from. There is a colon in the driver version that doesn't belong there.
> 
> Could you `ls -al /var/lib/dkms /var/lib/dkms/hid-xpadneo` please?

[image]

kakra, 2020-07-06:
Please `cat /var/lib/dkms/dkms_dbversion` and run `dkms status | awk -F', ' -- '$1 == "hid-xpadneo" { print $2 }'`

Mo0nCake, 2020-07-06:
> Please `cat /var/lib/dkms/dkms_dbversion` and run `dkms status | awk -F', ' -- '$1 == "hid-xpadneo" { print $2 }'`

[image]

kakra, 2020-07-06:
Please run `sudo mv /var/lib/dkms/hid-xpadneo /var/lib/tmp` and try again. It seems the dkms database is incomplete.

Mo0nCake, 2020-07-06:
> Please run `sudo mv /var/lib/dkms/hid-xpadneo /var/tmp/` and try again. It seems the dkms database is incomplete. This may be an artifact of the previous bug.

Now what?
[image]

kakra, 2020-07-06:
Now try to install. The dkms database seems to be clean.

kakra, 2020-07-06:
BTW: You've been to fast, you should have moved to `/var/tmp/` not `/var/lib/tmp`. If everything works, you can just delete `/var/lib/tmp`.

Mo0nCake, 2020-07-06:
> Now try to install. The dkms database seems to be clean.

[image]

kakra, 2020-07-06:
Please attach the log file from the last line of the screenshot.

Mo0nCake, 2020-07-06:
> Please attach the log file from the last line of the screenshot.

[make.log](https://github.com/atar-axis/xpadneo/files/4879395/make.log)

kakra, 2020-07-06:
I cannot read the error message because its Cyrillic again but I think I know where the problem comes from.

Is the database broken again now?

Mo0nCake, 2020-07-06:
> I cannot read the error message because its Cyrillic again but I think I know where the problem comes from.
> 
> Is the database broken again now?

I completely removed dkms and all references and directories from xpadneo, rebooted the computer, installed dkms again and here:
[image]

kakra, 2020-07-06:
That wasn't what I asked for. We already reset the DKMS status successfully. The question was if the database is broken again now. For why this error occurs in the first place, I'll have to investigate. Give me some time, I need to setup a system which supports DKMS, mine doesn't.

Mo0nCake, 2020-07-06:
> That wasn't what I asked for. We already reset the DKMS status successfully. The question was if the database is broken again now. For why this error occurs in the first place, I'll have to investigate. Give me some time, I need to setup a system which supports DKMS, mine doesn't.

Good. I don’t understand if she is broken now.

Mo0nCake, 2020-07-06:
> That wasn't what I asked for. We already reset the DKMS status successfully. The question was if the database is broken again now. For why this error occurs in the first place, I'll have to investigate. Give me some time, I need to setup a system which supports DKMS, mine doesn't.

Please tell me when it will be possible to install the "fixed" xpadneo?)

kakra, 2020-07-06:
It's ready when it's done. There is no schedule for this. I'll update you here as soon as we made progress or fixed this. Since this is a community project, you're free to opt-in for fixing this yourself and suggesting a patch. Or you just give me some time. Thanks. :-)

kakra, 2020-07-06:
> And I don’t understand why "dirty" is installed along with the regular version

Yeah, that's the bug. Because the install process dirties some files during the install phase, it will switch the version number underneath DKMS - and that does not like it. A proper fix is work in progress.

Mo0nCake, 2020-07-06:
> > And I don’t understand why "dirty" is installed along with the regular version
> 
> Yeah, that's the bug. Because the install process dirties some files during the install phase, it will switch the version number underneath DKMS - and that does not like it. A proper fix is work in progress.

I understand, thanks for the work) I did not want to seem unhappy. I will wait for the fix

kakra, 2020-07-06:
I'll probably publish v0.8 at the beginning or during next week, that will contain all the new features (which are not in the 0.7 versions) and all the bug fixes. I don't want to seem unhappy, too. ;-) You are currently the only person who really tests this and reports back, I really appreciate that. Keep going. Thanks for trying and finding all the bugs.

kakra, 2020-07-06:
Should be fixed now, please retry.

closed 2020-07-06 by kakra

Mo0nCake, 2020-07-06:
> Should be fixed now, please retry.

Oh yes, it is installed and working) Thank you very much)